MiczFlor/RPi-Jukebox-RFID #684: Pause/Stop playback if card is removed

joshwapohlmann, 2019-12-28:
I found that it can be more intuitive to a child, to place the card at let it rest on the box as long as he wishes to listen to the story/song. The expectation is, that the playback stops when the card is removed.

Therefore I would like to have the option to enable this behavior (maybe selectable in the web apps settings page).

joshwapohlmann, 2019-12-28:
related #572

joshwapohlmann, 2019-12-28:
I'd love working on this, solving #683 first would help :)

Erechtheus, 2020-01-21:
I would also like to participate on this issue. Similar to @joshwapohlmann I am not sure how to contribute. As far as I understood the problem is that Reader.py is only called when a RFID-chip is present. So implementing this issue probably requires some kind of watchdog, which regularly  (every second or so) checks if a RFID-card is no longer present and than pauses?

joshwapohlmann, 2020-01-22:
@Erechtheus I'm working on #683 right now and link to the result so we both might get to a mutual understanding what and how to do. :)

ozone89, 2020-02-11:
just leaving my two cents, the [Phidgets 1024 RFID](https://www.phidgets.com/?&prodid=23) reader supports reporting of card presence in their APIs:
```
getTagPresent()

```

This property is true if a compatibile RFID tag is being read by the reader.

- TagPresent will remain true until the tag is out of range and can no longer be read.

```
void onTagLost(self, tag, protocol)
```
Occurs when an RFID tag that was being read is removed from the read range.

UndIch, 2020-03-02:
Thats a great function - but comes with a huge price tag... 60 $ for a RFID reader is quite a lot to ask for...

ozone89, 2020-03-04:
> Thats a great function - but comes with a huge price tag... 60 $ for a RFID reader is quite a lot to ask for...

true, but maybe it's just a software feature in libphidget that could be implemented?

dermexx, 2020-03-12:
a cheap solution for that problem is to use a reed switch that you place close to the rfid reader...
only downside is, that you have to add a magnet to every tag you want to use.

s-martin, 2024-04-19:
Is already implemented.

For V3 see e.g. here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/developers/rfid/basics.md#place_not_swipe-true--false

closed 2024-04-19 by s-martin