MiczFlor/RPi-Jukebox-RFID #685: normalize Audiofiles

kabakakao, 2019-12-29:
It would be nice to "normalize" die imported Audiofiles.
https://packages.debian.org/buster/normalize-audio
So you normally don't have to change the volume.

eg:
`find /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ -name *.mp3 -print0 | xargs -0 normalize`

MiczFlor, 2020-01-04:
Hi @kabakakao 
I agree, and am normalizing and transcoding files on my laptop before placing them on the Phoniebox. But I am not sure that the Phoniebox is the place to do this. Do you? It takes a lot of capacity

kabakakao, 2020-01-04:
I agree that it would take high performance, but i think it is a good place to do that. I think a Button Or Option in the Web Interface would be good. So everyone can decide on his own.

belinea4071, 2020-06-16:
I changed the setting in "/etc/mpd.conf",
**from**: volume_normalization            **"no"**
**to**: volume_normalization            **"yes"**.

_This setting enables on-the-fly normalization volume adjustment. This will
result in the volume of all playing audio to be adjusted so the output has
equal "loudness". This setting is disabled by default._

This did the trick for me.

closed 2020-06-23 by MiczFlor