MiczFlor/RPi-Jukebox-RFID #686: Installation with Spotify support fails

topse, 2019-12-30:
Hi,
I am trying a "clean" Installation on Rasperry Pi Buster, using either the stable or the development version. Within this process the integration of spotify does not work.

While running the one-line-install-script it gives the following error:

`Mopidy` requires Python '>= 3.7' but the running Python is 2.7.16`

After Installation the Web-UI shows Spotify in the menu, but a click on the link gives only a 404: Not Found error.

I tried to  remove the "pylast 2.4" - stuff, but it did not help.

Does anyone have an idea?

Thanks and Best Regards,
          Tobias

Squelsh, 2019-12-30:
Same here. Python 3.7 is installed, but the default python version is 2.7.

mr-miyagi98, 2019-12-31:
Seems to be a problem with a new version of Mopidy:

> Mopidy 3.0 released (December 22, 2019)
We released Mopidy 3.0 yesterday. It no longer supports Python 2.7, but instead requires Python 3.7+. 
https://discourse.mopidy.com/t/mopidy-3-0-released/3646
>

Squelsh, 2020-01-01:
Simply switching my default Python to 3.7 did not do the trick (from the log output of the one-line installer I suppose, that some of the dependencies are relying on Python 2.7).

Squelsh, 2020-01-01:
I switched my default Python to 3.7 and followed the hint from https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/689#issue-544185849.
After that the one-line-installer completed successfully.

But I still have issues: Mopidy IRIS is not reachable and MPD autostart fails. But that is another topic.

IZooRz, 2020-01-01:
How can i change the default version? Thanks for your help in advance

Squelsh, 2020-01-01:
@IZooRz check your` /var/log/syslog`: I found a bunch of errors there about mopidy and iris.
As far as I read, iris does not yet support Python 3.7
See my bugreport here:
https://github.com/jaedb/Iris/issues/474#issue-544370735

MiczFlor, 2020-01-04:
I added the fix from the other thread:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/696
And am open to any suggestions on how to solve the mopidy issue :)

closed 2020-01-04 by MiczFlor

topse, 2020-01-06:
About an idea to fix the mopidy stuff: What about forcing installing the old version?

mikebarkmin, 2020-01-06:
I have removed mopidy-iris and the custom build mopidy-spotify.
Then I installed mopidy-spotify, mopidy-local and mopidy-mpd from the repo. After rescanning my library everything is working except the mopidy-iris stuff (aka spotify-web).