MiczFlor/RPi-Jukebox-RFID #687: RFiD - Alternate Device via Reader.py.experimental

cschwarz007, 2019-12-30:
Dear all,

I am trying to get an RDM6300 to work, following the instructions at https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special.

However, when I try to execute `RegisterDevice.py` I get the following error:

>pi@MusicBox:~/RPi-Jukebox-RFID/scripts $ ./RegisterDevice.py
Traceback (most recent call last):
  File "./RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    return devices
  File "/home/pi/.local/lib/python2.7/site-packages/evdev/__init__.py", line 7, in <module>
    from evdev.device import DeviceInfo, InputDevice, AbsInfo
  File "/home/pi/.local/lib/python2.7/site-packages/evdev/device.py", line 10, in <module>
    from evdev.eventio_async import EventIO
  File "/home/pi/.local/lib/python2.7/site-packages/evdev/eventio_async.py", line 3, in <module>
    import asyncio
  File "/usr/local/lib/python2.7/dist-packages/asyncio/__init__.py", line 9, in <module>
    from . import selectors
  File "/usr/local/lib/python2.7/dist-packages/asyncio/selectors.py", line 39
    "{!r}".format(fileobj)) from None
                               ^
SyntaxError: invalid syntax

What am I doing wrong?

Many thanks for your help in advance!
Christian

ZyanKLee, 2020-01-18:
hello @cschwarz007, which version of this script did you use (develop/master)?
can you please re-test with the latest version from develop branch?

s-martin, 2020-04-13:
This should be fixed now both on master and develop.
I close this issue now, but feel free to reopen it, if you still have the problem.

closed 2020-04-13 by s-martin