MiczFlor/RPi-Jukebox-RFID #688: No Playback via WebInterface / RFID Tag

Squelsh, 2019-12-30:
Hi all!

First things first: Great project! Thanks so much!

Problem: Due to a broken SD card I had to reinstall my box. Now I am struggling with the new version and I just can't get it to play any files. If I use the shell to add songs to mpc and play them it works. But something seems broken when `playout_controls.sh` tries to control mpc...
Nevertheless I can play streams! But no .mp3s.
I switched from devel to master but it made no difference.

I would be thankful for any debugging hints. I have in depth Linux skills. I already set debug for the script but it does not show anything helpful.

Version: 2.0 rc7
Edition: Plus edition (feat. Spotify integration)
Mopidy Server Status: active (running) since Mon 2019-12-30 22:53:22 GMT; 46min ago

Cheers,
Andy

Squelsh, 2019-12-30:
Thats the debug output:

```
#START##### SCRIPT resume_play.sh (2019-12-30.23:51:30) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: savepos
VAR VALUE: 
VAR FOLDER: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: savepos
VAR FOLDER from settings/Latest_Folder_Played: Sing Kinderlieder
  savepos FOLDER: Sing Kinderlieder
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
#END####### SCRIPT resume_play.sh (2019-12-30.23:51:30) ##
remove playing.txt
########### SCRIPT playout_controls.sh (2019-12-30.23:51:30) ##
VAR COMMAND: playlistaddplay
VAR VALUE: Sing Kinderlieder
   playlistaddplay playlist name VALUE: Sing Kinderlieder
   playlistaddplay FOLDER: Sing Kinderlieder
mpc load Sing Kinderlieder
## SCRIPT single_play.sh (2019-12-30.23:51:30) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: single_check
VAR VALUE: 
VAR FOLDER: Sing Kinderlieder
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Sing Kinderlieder/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: single_check
  entering: single_check with value OFF
#START##### SCRIPT resume_play.sh (2019-12-30.23:51:30) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: resume
VAR VALUE: 
VAR FOLDER: Sing Kinderlieder
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Sing Kinderlieder/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: resume
  savepos FOLDER: Sing Kinderlieder
  entering: resume with value OFF
  entering: single with value OFF
#END####### SCRIPT resume_play.sh (2019-12-30.23:51:30) ##
mpc load Sing Kinderlieder && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume -d=Sing Kinderlieder
  echo Sing Kinderlieder > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
  VAR Latest_Folder_Played: Sing Kinderlieder
  # end playout_controls.sh playlistaddplay
########### SCRIPT playout_controls.sh (2019-12-30.23:51:37) ##
VAR COMMAND: playerstop
VAR VALUE: 
#START##### SCRIPT resume_play.sh (2019-12-30.23:51:38) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: savepos
VAR VALUE: 
VAR FOLDER: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: savepos
VAR FOLDER from settings/Latest_Folder_Played: Sing Kinderlieder
  savepos FOLDER: Sing Kinderlieder
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
#END####### SCRIPT resume_play.sh (2019-12-30.23:51:38) ##
remove playing.txt
########### SCRIPT playout_controls.sh (2019-12-30.23:51:38) ##
VAR COMMAND: playlistaddplay
VAR VALUE: Sing Kinderlieder
   playlistaddplay playlist name VALUE: Sing Kinderlieder
   playlistaddplay FOLDER: Sing Kinderlieder
mpc load Sing Kinderlieder
## SCRIPT single_play.sh (2019-12-30.23:51:38) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: single_check
VAR VALUE: 
VAR FOLDER: Sing Kinderlieder
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Sing Kinderlieder/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: single_check
  entering: single_check with value OFF
#START##### SCRIPT resume_play.sh (2019-12-30.23:51:38) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: resume
VAR VALUE: 
VAR FOLDER: Sing Kinderlieder
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Sing Kinderlieder/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: resume
  savepos FOLDER: Sing Kinderlieder
  entering: resume with value OFF
  entering: single with value OFF
#END####### SCRIPT resume_play.sh (2019-12-30.23:51:38) ##
mpc load Sing Kinderlieder && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume -d=Sing Kinderlieder
  echo Sing Kinderlieder > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
  VAR Latest_Folder_Played: Sing Kinderlieder
  # end playout_controls.sh playlistaddplay
########### SCRIPT playout_controls.sh (2019-12-30.23:52:24) ##
VAR COMMAND: volumeup
VAR VALUE: 
pi@phoniebox:~/RPi-Jukebox-RFID/logs $
```

Squelsh, 2019-12-31:
Getting more and more puzzled:
If I use `mpc add FooBar` and then `mpc playlist` I will see that mpc loaded the files from the FooBar directory correctly.
If I use `mpc load FooBar`and then `mpc playlist` I will see nothing.

Why can't mpc read its own playlists?!

```
pi@phoniebox:~/RPi-Jukebox-RFID/playlists $ cat FooBar.m3u 
local:track:FooBar/startupsound.mp3
```

Squelsh, 2020-01-01:
After I changed the playlist_recursive_by_folder.php and use the "classic" way and not the "plusSpotify" way to generate the playlists, MPD works as expected (?).

Squelsh, 2020-01-02:
I think I found the root of the problem:
Since Mopidy 3.x was released, they dropped Python 2.7 support and some required plugins (IRIS and the Playlist parser) are not yet compatible with Python 3.7.
This causes the strange behavior I discovered (only webstreams work, no local files).

You can check if you stop the mopidy daemon and start mopidy on the command line. For me it then prints a bunch of errors when importing the plugins.

samgot1, 2020-01-03:
same issue here

ZyanKLee, 2020-01-05:
does it work for you after downgrading mopidy to the latest 2.3 version?
`sudo apt-get install mopidy=2.3.1-1`

Squelsh, 2020-01-06:
@ZyanKLee Thanks for the hint, already tried that. Fails with tornado:

```
pi@phoniebox:~/mopidy-spotify $ mopidy
INFO     Starting Mopidy 2.3.1
ERROR    Failed to load extension iris: invalid syntax (web.py, line 205)
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/mopidy/ext.py", line 202, in load_extensions
    extension_class = entry_point.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/mopidy_iris/__init__.py", line 5, in <module>
    import tornado.web
  File "/usr/local/lib/python2.7/dist-packages/tornado-6.0.3-py2.7-linux-armv7l.egg/tornado/web.py", line 205
    application: "Application",
               ^
SyntaxError: invalid syntax
INFO     Loading config from builtin defaults
INFO     Loading config from /home/pi/.config/mopidy/mopidy.conf
INFO     Loading config from command line options
ERROR    Validating extension http failed with an exception.
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/mopidy/ext.py", line 272, in validate_extension_data
    data.extension.validate_environment()
  File "/usr/lib/python2.7/dist-packages/mopidy/http/__init__.py", line 34, in validate_environment
    import tornado.web  # noqa
  File "/usr/local/lib/python2.7/dist-packages/tornado-6.0.3-py2.7-linux-armv7l.egg/tornado/web.py", line 205
    application: "Application",
               ^
SyntaxError: invalid syntax

```

No clue, what python does not like about that line...

Squelsh, 2020-01-06:
My mistake: Iris requires tornado <= 5.0
After installing `pip install 'tornado==5.0'` the error is gone.

FloMrFrei, 2020-01-06:
Thank you all for posting problem and solution, including what to type into the shell! This + https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/700 made it work for me on RaspberryPi 2 and Raspbian Buster.
```
sudo apt-get install mopidy=2.3.1-1
sudo python2.7 -m pip install Mopidy==2.3.*
pip install 'tornado==5.0'
```

JoeyRamone, 2020-01-06:
@FloMrFrei so you've ran it prior the oneline Installer?

ZyanKLee, 2020-01-07:
You can even do so afterwards. At least I did and it worked.
Though I ran also the third command with sudo.

FloMrFrei, 2020-01-07:
@JoeyRamone I run those lines after the 1-line installer. I guess otherwise, the installer will install mopidy 3.x over the requried 2.3.1 version.

kingosticks, 2020-01-10:
If you want to prevent apt from upgrading Mopidy you can use `apt-mark hold mopidy` after installing your specific version.

MiczFlor, 2020-01-15:
I am posting the solution (i haven't tested it, though) from @FloMrFrei here in English. Would be great if somebody could create a pull request against the buster install script:

1.) install libspotify-dev
```
wget -q -O - https://apt.mopidy.com/mopidy.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/buster.list
sudo apt update
sudo apt-get install libspotify-dev
```

2.) One-Line-Installer (Buster)

3.) Mopidy downgrade
```
sudo apt-get install mopidy=2.3.1-1
sudo python2.7 -m pip install Mopidy==2.3.*
sudo pip install 'tornado==5.0'
```

s-martin, 2020-02-19:
The PRs #789 and #793 have been merged to master and develop.

Could you please check, if the issue is fixed now?

s-martin, 2023-04-29:
Closing as the info is very outdated.

closed 2023-04-29 by s-martin