MiczFlor/RPi-Jukebox-RFID #690: Issues with Raspbian Buster

Squelsh, 2020-01-01:
I have a bunch of problems with my Buster installation:

- MPD does not start as root:
`Jan  1 13:20:35 phoniebox mpd[984]: Jan 01 13:20 : exception: Failed to set group 0: Operation not permitted`
==> I now run mpd NOT as root but as pi user (which is the official advise) and therefore I moved all mpd files to  `/home/pi/.mpd`
Perhaps you want to change the installer and the docs? Why do you prefer to run mpd as root?

- Setting the volume through the installer script fails:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/1fb92db2228da87c2dc1fd95b3514ffec5789466#commitcomment-36629221

- With my external USB soundcard I can not set a volume lower than 25%. But this is also not possible via alsamixer (?), so not a Jukebox problem.

MiczFlor, 2020-01-04:
Hi @Squelsh 
the fact that mpd is running as root is a legacy issue. the webserver is also running as root :) when Phoniebox was just a bunch of scripts and the Pi they ran on a "one purpose" machine, I felt that the shortcut is justified.
Two years later, I think this should be fixed and no root should be required. But I don't have that time to make these big changes at the moment.
I will take a look at the install script again. It worked for me when I published it on RPi2 and RPi3, so not sure what the issue there is.
Have you tried without the USB card and used the jack, just for error tracing?
All the best, micz

Squelsh, 2020-01-06:
My `/etc/mpd.conf` looks like this and I experienced no problems so far. I had to create the folder `~/.mpd/` and thats all.

```
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
playlist_directory              "/home/pi/RPi-Jukebox-RFID/playlists"
db_file                 "/home/pi/.mpd/tag_cache"
log_file                        "/home/pi/.mpd/mpd.log"
pid_file                        "/home/pi/.mpd/pid"

```

IZooRz, 2020-01-09:
Hi Squelsh,

So this is the fix and Spotify and mpd are working properly? Do I have to add That folder and change the Conf file before installing?

Thanks for your Support

Squelsh, 2020-01-09:
Hi,
this was a missunderstanding on my side. MPD is not used in the Spotify edition, if I am correct.
See https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/683#issuecomment-571129953

But yes, I changed mpd to use the local files and run successfully as "pi" user. It makes no difference, if you create the folder before or after installation. BUT: You have to modify the systemctl config of mpd to load the settings from your local config file.

But as I said: In the Spotify edition mpd is replaced by mopidy. Which I managed to "downgrade" so that it runs with python 2.7 and all required modules.

closed 2020-01-15 by MiczFlor