MiczFlor/RPi-Jukebox-RFID #692: trouble playing locale mp3

felixlindemann, 2020-01-02:
I can play Spotify albums (both  started by RFID / and website), web radio (like bayern from the demo)
but I can't play mp3 stored on the sd-card - both own and demo-files.

I also noticed an error in the chrome-developer tools:
`[Error] Failed to load resource: the server responded with a status of 500 (Internal Server Error) (playlist.php, line 0)` 

I'm running the following setup:
`OS DistributionRaspbian
DescriptionRaspbian GNU/Linux 9.11 (stretch)
Release9.11
Codenamestretch
 Phoniebox Setup
Version2.0 rc7
EditionPlus edition (feat. Spotify integration)
Mopidy Server Status
active (running) since Thu 2020-01-02 00:21:45 UTC; 9min ago`

Squelsh, 2020-01-02:
Sounds like a duplicate of my issue https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/688

I think, the problem is a missing plugin for mopidy which is responsible for parsing the playlists. Perhaps stop the mopidy daemon and start it on the command line manually. This results in a bunch of error messages for me, since mopidy 3.x was released...
But I am running Raspian Buster. Don't know, if you get the latest mopidy for Stretch?

MiczFlor, 2020-01-04:
Hi @Squelsh @felixlindemann 
the spotify integration is an issue - or has become a pressing issue for Phoniebox since the latest mopidy changes. Any experts with time at their hands to help out?
I will close this thread as a duplicate: #688

closed 2020-01-04 by MiczFlor