MiczFlor/RPi-Jukebox-RFID #693: Start song/stream at startup

erhapp, 2020-01-03:
Hello

What is the easiest way to make the player start up with a song/stream playing?

I would like the player to play a radio webstream at startup.

kind regards

Erik

MiczFlor, 2020-01-04:
Hi @erhapp 

i like the idea and try blindly to whip something up (i.e. I am just writing here, not testing anywhere). If it works for you, please share it here, then I can add it to the wiki.

I assume that you have Phoniebox installed and an an RFID connected to the webstream you want to start. In this case, you need to do the following:

1. create a service that will run on startup

In the following file, replace `1234567890` with the ID of your RFID tag

~~~
[Unit]  
Description=Phoniebox Startup Playlist
After=default.target

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/home/piRPi-Jukebox-RFID/scripts/rfid_trigger_play.sh -i=1234567890

[Install]
WantedBy=multi-user.target
~~~

Save the file at this location:
~~~
/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-startup-playlist.service.buster-default.sample
~~~

2. Install and activate the service

The following lines will copy the file to the right place to run on startup:
~~~
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-startup-playlist.service.buster-default.sample /etc/systemd/system
/etc/systemd/system/phoniebox-startup-playlist.service
sudo chown root:root /etc/systemd/system/phoniebox-startup-playlist.service
sudo chmod 644 /etc/systemd/system/phoniebox-startup-playlist.service
sudo systemctl enable phoniebox-startup-playlist
~~~
The reboot, tell me what happens.

All the best, micz

erhapp, 2020-01-04:
Micz

Perfect. The solution works great.

This addition greatly improves the WAF (Wife Acceptance Factor) of the kitchen radio I'm building. ;-)

Thank you very much!

Erik

erhapp, 2020-01-04:
It's appears that I was a little bit to eager to get this working. The music isn't playing after all.

It did work the first two times that I rebooted. Now I realize that MPD was playing the stream when I gave the reboot command. Maybe MPD resumed playing after reboot?

Anyways, I need to look a little bit further into this solution.

erhapp, 2020-01-06:
While testing the new script I noticed that the startup sound wasn't playing either.

Adding 'After=network.target iptables.service firewalld.service rfid-reader.service' to the startup service solved this issue. (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/645)

Adding the same change to the playlist service made the service work as well.

MiczFlor, 2020-01-15:
Hi @erhapp 
could you post your service script here, so we have a working solution (not just my "wild guess" :)

erhapp, 2020-01-16:
@MiczFlor : Sure. Thanks for the help.

`[Unit]  
Description=Phoniebox Startup Playlist
After=network.target iptables.service firewalld.service rfid-reader.service

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh -i=21211137166240

[Install]
WantedBy=multi-user.target`