MiczFlor/RPi-Jukebox-RFID #699: Streaming radiostations fails without message

Barzille, 2020-01-04:
Hi,
local media files are playing fine.
But when I try to add a radio station like with the following urls:

http://streams.rsh.de/kids/mp3-192/streams.rsh.de/play.m3u
or https://www.ndr.de/resources/metadaten/audio_ssl/m3u/n-joy.m3u

It plays simply nothing.

MPD log shows only

Jan 04 21:58 : client: [1004] opened from [::1]:38456
Jan 04 21:58 : client: [1004] closed
Jan 04 21:59 : client: [1005] opened from [::1]:38458
Jan 04 21:59 : client: [1005] closed

I am currently on a PI without WiFi and only connected via cable until the new device with WiFi arrives. Could this be the issue? Otherwise how can I get more infos about what will be the error?

I am on 2.0 rc7

Cheers Maik

kabakakao, 2020-01-08:
Hi Maik,
you have to use the Link within the m3u file:
1) http://rsh.hoerradar.de/rsh-kids-mp3-hq?sABC=5r163398%230%23921n87353926q9249ps0q4n84q43ro98%23fgernzf.efu.qr&amsparams=playerid:streams.rsh.de
2) https://ndr-njoy-live.sslcast.addradio.de/ndr/njoy/live/mp3/128/stream.mp3

hope, this helps ;)

Regards,
Stephan

Barzille, 2020-01-10:
Yes, perfect. Thank you 🙏

closed 2020-01-10 by Barzille