MiczFlor/RPi-Jukebox-RFID #703: keinWiFi nach install script

Moppel2k, 2020-01-07:
Hallo Zusammen
Ich musste am Wochenende unsere Phoniebox neu aufsetzen (haben einen neuen Router bekommen).

Nun habe ich das Problem wenn ich das One line install l script für Raspbian Buster ausführe habe ich anschließend Probleme mich mit dem Wälan zu verbinden.
Die angegebenen SSID ist korrekt das Passwort auch!
Die IP Adresse ist im Router an den Raspbarry gebunden und sollte auch kein Problem sein!

Wenn ich das Script durchlaufen lasse und die W-Lan Konfiguration weg lassen verbindet es sich nach dem Neustart ohne Probleme! 

ich komme aber nicht auf die Seite der Phoniebox unter der IP erreiche ich sie nicht mit dem Laptop?!?

könnt ihr mir da weiterhelfen?

JoeyRamone, 2020-01-07:
@Moppel2k 
Wie sieht es aus, wenn du nach dem raspbian Image direkt die wpa_supplicant.conf und ssh auf den pi kopierst. Kannst du dann über WiFi zugreifen?

ddbkva, 2020-01-07:
oder wenn der pi am monitor angeschlossen ist  sudo raspi-config da kann man auch das wlan einstellen dann braucht man es nicht mehr im script zu machen.

Moppel2k, 2020-01-09:
Sorry erstmal das ich mich erst jetzt melde!
Hatte viel zu tun in den letzten tagen!

Der fehler lag bei mir!!! (Wer hätte es gedacht)

Ich muss beim neuen Router die "Länderkennung" anders als beim alten auf CH stellen statt auf DE!

Trotzdem danke für eure Unterstützung!

Jetzt habe ich nur noch das Problem das ich wenn ich die IP vom Browser aus aufrufe, egal ob von der Box oder aus dem Netz, ich nicht auf die Seite komme!

Dazu muss ich aber erst mal suchen ob es dafür schon eine Lösung irgendwo gibt!

Gruss
moppel2k

MiczFlor, 2020-01-15:
Hi @Moppel2k 
Ich habe deinen Hinweis im Wiki noch eingebaut:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#configure-the-wifi
das ist so ein typisches Problem, bei dem man wahnsinnig werden kann :)

closed 2020-01-15 by MiczFlor