MiczFlor/RPi-Jukebox-RFID #705: Keyboard emulation with USB arcade buttons

mistersunshine2020, 2020-01-08:
Hi, 

Firstly, thanks for the amazing work here. I've been playing around for the first time with Raspberry Pi in order to create an RFID music player with local files and housed in a lego case. It's going well - been a steep learning curve for me to start basically from scratch, but I have a system working. 

One of the things I was hoping to add (in spirit of the solder less approach) was adding some arcade buttons to control play/pause, next and previous tracks. I figured it would be easy enough to get the Zero Delay USB Encoder board commonly used for arcade devices, only attach three buttons, and emulate a key stroke when they were pressed. 

It is not that simple - for me at least. I thought from the Logitech V20 example I could do something similar, but I'm struggling to know where to begin. 

Any ideas, or should I attach the buttons directly to the GPIO (and if so, can I directly connect rather than using a breadboard/capacitor?

It seems a shame not to be able to take advantage of the USB controller, but every search I've done hasn't come up with any solution, so thought I'd reach out. 

Best,

Tom

IZooRz, 2020-01-09:
Good Morning, I just finished adding arcarde buttons. In the docs is a file for that topic. I directly connected the Buttons with the gpio pins on the Raspberry and it works.

mistersunshine2020, 2020-01-12:
Hi there, thanks for the feedback. I went the GPIO route and everything was working well. 

BUT then I added a HiFiBerry lite which I've managed to set up as the audio output - everything works, except the previous and next buttons have stopped working. Play/Pause continues to work fine, so is there a potential conflict with the installation of the HiFiBerry? I'm only using play/pause on GPIO 27, previous on GPIO 22, and next on GPIO 23, and removed all other button command from the gpis-buttons script.

As stated, this worked all fine before the installation of the HiFiBerry Lite, and I don't think there's a conflict with any of the sound cards functions and the GPIO selections above. 

Any clues? 

Best,

Tom

MiczFlor, 2020-01-15:
Hi @mistersunshine2020 @IZooRz 
I think the file in the docs you mention is also here?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons
And, yes, the GPIO conflicts can create frustration. More often than not, this is the reason for posts "Phoniebox stopped working!". There is little to do than try. Remember the Phoniebox slogan: everyone builds an original :)

mistersunshine2020, 2020-01-16:
> Hi @mistersunshine2020 @IZooRz
> I think the file in the docs you mention is also here?
> https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons
> And, yes, the GPIO conflicts can [...]

Thanks - I wondered initially if I'd soldered the additional GPIO bins incorrectly, and then completely destroyed the HAT when trying to resolder it ...

That said, without it, the buttons all work fine as intended, so I am going to work on the assumption that it was a GPIO conflict, and see if I can install a USB sound card to prevent any potential conflicts in the future. It's all working so well at the moment and don't want to poke the bear.

s-martin, 2020-03-07:
I’m closing this as the issue seems to be fixed.
Please feel free to reopen this issue, if the problem persists.

closed 2020-03-07 by s-martin