MiczFlor/RPi-Jukebox-RFID #709: Irgendwie läuft hier gar nichts...

GMZ87, 2020-01-09:
Hi @ all,

habe mich dazu durchgerungen dieses tolle Projekt zu starten und muss sagen das ich vorne und hinten nicht klar komme.
Ich bekomme die Jukebox einfach nicht installiert. Dazu habe ich jeweils diese One-Line-Install Commands für Stretch und auch für Buster genutzt. Buster und Stretch ebenfalls mit unterschiedlichen Datum.
Leider habe ich immer wieder Fehler bei der Installation.

Bei der letzten Installation unter Buster Lite vom 10.07.2019:

Do you want to start the installation? [Y/n] y
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libspotify-dev
./buster-install-default.sh: line 494: git: command not found
./buster-install-default.sh: line 497: cd: RPi-Jukebox-RFID: No such file or directory
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/smb.conf.buster-default.sample': No such file or directory
chown: cannot access '/etc/samba/smb.conf': No such file or directory
chmod: cannot access '/etc/samba/smb.conf': No such file or directory
sed: can't read /etc/samba/smb.conf: No such file or directory
sudo: smbpasswd: command not found
... und so weiter

Ich habe ca. 7 vergebliche Versuche hinter mir. Die SD wird jeweils frisch aufgesetzt und nur per wpa_supplicant.conf und "ssh-Datei" modifieziert. Da ich kein Bildschirm dran hängen möchte, arbeite ich hier nur über ssh. 
Hat jemand einen Tipp für mich woran das liegen kann? Bin im Linux absolut unbewandert und arbeite immer nur Tutorials ab. Was bis dato immer gelang. Aber hier hapert es schon am ersten schritt...
Eventuell noch Interessant:

wlan0     IEEE 802.11bgn  ESSID:"WLAN-FN5MED"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.462 GHz  Access Point: 7C:7D:3D:1D:52:00
          Bit Rate:72.2 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=100/100  Signal level=83/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.108  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::69ca:64dd:454f:ec55  prefixlen 64  scopeid 0x20<link>
        inet6 2003:c2:874d:8c44:16ef:4e8b:899e:e119  prefixlen 64  scopeid 0x0<g  

--2020-01-09 20:33:40--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... failed: Connection refused.
Connecting to localhost (localhost)|127.0.0.1|:80... failed: Connection refused.

UndIch, 2020-01-11:
Schau mal in #648 - klingt ein bisschen ähnlich wie mein Problem - eventuell hilft es dir auch.

ZyanKLee, 2020-01-11:
Hallo @GMZ87,
probier es doch bitte noch mal. Seit dem 10.07. hat sich eine Menge geändert und ich gehe davon aus, dass dieses Problem zumindest schon längst behoben ist. Meine letzte Installation war Anfang Januar 2020 und mit einem kleinen Workaround (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/688#issuecomment-571129900) funktionierte danach dann alles.

GMZ87, 2020-01-11:
vielen Dank für eure Tips. Es scheint jetzt zu laufen... 
Nach dem gefühlten 20ten Versuch habe ich einfach mal den Update und Upgrade befehlt weggelassen. dies hat jetzt mit der hier angebotenen "Stretch" Version dann auch vernünftig installiert.

WebApp startet. Man kann diese auch Problemlos bedienen und auch der Scanner funktioniert und erkennt die Karten bzw. Spielt einen angegebenen Ordner nach dem einscannen der Karte ab. 
Fast zu schön um Wahr zu sein kommt dann nun die nächste Herausforderung. Kein Ton! 

Ich habe hier einen alten Pi Modell B+ auf den ich nun per Hifiberry Miniamp 2 kleine Lautsprecher anschließen wollte. Den Miniamp habe ich (vorerst) direkt auf dem GPIO Port gesteckt und in der Config.txt die Zeile #dtparam=audio=on und dtoverlay=hifiberry-dac. 
Beim Start den Pi kommt das leichte knacken was teils beschrieben wird und dann ist Ruhe. Mit viel Fantasie würde ich behaupten das die Lautsprecher leicht rauschen. Per Web App Lautstärke verändern bringt leider nichts.

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Müsste an der Stelle nicht"Subdevices: 1/1" stehen?

Die Veränderung der Lautstärke wird aber angenommen:

pi@raspberrypi:~ $ ./RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup
OK MPD 0.19.0
OK
pi@raspberrypi:~ $ . RPi-Jukebox-RFID/scripts/playout_controls.sh -c=setvolume -v=80
OK MPD 0.19.0
OK
pi@raspberrypi:~ $

Vermute so langsam das der Hifiberry Defekt ist.....

MFG

UndIch, 2020-01-11:
Was passiert wenn du die  speaker-Test Funktion laufen lässt?

GMZ87, 2020-01-11:
pi@raspberrypi:~ $ speaker-test -D hifiberry -c 2

speaker-test 1.1.3

Playback device is hifiberry
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 128 to 131072
Period size range from 64 to 65536
Using max buffer size 131072
Periods = 4
was set period_size = 32768
was set buffer_size = 131072
 0 - Front Left
 1 - Front Right
Time per period = 2.769736
 0 - Front Left
 1 - Front Right
Time per period = 5.462485
 0 - Front Left
 1 - Front Right
Time per period = 5.462463
 0 - Front Left

und so weiter....
Wie beendet man das eigentlich? :)

UndIch, 2020-01-11:
Und die spannende Frage - hast du was gehört?

GMZ87, 2020-01-12:
Leider hatte ich hier immer noch kein Ton.

Ich habe nun noch einige Änderungen vorgenommen. 
Umstieg auf Raspi 2
Neue SD Karte 
Buttons deaktiviert
So läuft es nun vorerst wunderbar. Ich glaube hauptsächlich lag es aber an den Buttons, da noch keine Anschlossen sind. Habe derzeit auch noch keine, kommen aus China ;) Mit dem PI 2 läuft das komplette System bei weitem besser und schneller. Von daher fliegt der Pi 1B+ wieder in die Versenkung.
Bin mal gespannt wie das hier weiter geht. Mache mich jetzt erstmal an das Gehäuse, wo morgen die ersten Teile zu im Laser geschnitten werden. Das soll ein Cube werden mit 13,5x13,5x13,5 und Touch Sensoren als Schalter... Leider nicht die Original Größe der Toniebox aber dafür mit 2 alten Bose Lautsprechern. 
Lieben Dank an Alle

MiczFlor, 2020-01-15:
Hallo @GMZ87 
habe gerade erst the langen thread gelesen und bin froh, dass es läuft - und mache den thread zu.
Nicht vergessen: Bilder posten!
Beste Grüße, micz

closed 2020-01-15 by MiczFlor