MiczFlor/RPi-Jukebox-RFID #711: Suggestion: Automated Testing

ZyanKLee, 2020-01-11:
Hello @MiczFlor,

I follow and use your project for some time now and I'm a great fan. I do have the impression though it could be beneficial to implement some kind of automated testing to find obvious bugs early in the process and try to maintain a stable master (perhaps even develop) branch at all times.

There are various tools that may assist in such an endavour - starting from simple linting tools for various script languages to unit-testing php and bash scripts.

Setting this up would take some development time in the beginning, but the benefit in the long run would be easier code reviews as you do not need to take care about obvious errors anymore and can concentrate on the actual features and solutions that are queued up for merging.

What are your thoughts about this?

MiczFlor, 2020-01-15:
Hi @ZyanKLee 
Phoniebox has become such a complex beast :) depending on peoples hardware (soundcard, rotary dial, RFID reader) often the GPIO conflicts are errors. And Phoniebox has become a very community based code base, which results in many different styles of writing and level of tech insight. 
Currently, I find myself learning other peoples code before testing and finally mergind with master.
Automated testing would be an improvement for me, but code contributors who send great tweaks and fixes, might shy away from writing tests.
What would be a solution to provide both:

* A good testing environment to merge faster to master (I should TM that last part of the sentence :)
* Easy entry point for code contributions

Thanks and looking forward to hear your thoughts.

s-martin, 2020-01-15:
A somehow simple first step could be to add maybe a Python linter to GitHub Actions, which would report possible issues after each commit.

This could be extended step by step.

ZyanKLee, 2020-01-15:
I aggree, small steps - one at a time - will be best. We should not forget that introducing linters will definitely produce a bunch of errors with existing code right from the start.

ZyanKLee, 2020-01-15:
This might be a starting point as first step: https://github.com/actions/starter-workflows/blob/master/ci/python-app.yml

s-martin, 2020-01-15:
I added a first, simple lint Action to my fork, see https://github.com/s-martin/RPi-Jukebox-RFID/blob/master/.github/workflows/pythonpackage.yml

Found already some issues e.g. in RegisterDevice.py nd others, because I configured it only for Python 3:

> 
> 2020-01-15T21:46:59.8331155Z ./scripts/RegisterDevice.py:9:35: E999 SyntaxError: invalid syntax
> 2020-01-15T21:46:59.8332125Z print "Choose the reader from list"
> 2020-01-15T21:46:59.8336777Z                                   ^
> 2020-01-15T21:46:59.8337132Z ./scripts/MFRC522.py:290:20: E999 SyntaxError: invalid syntax
> 2020-01-15T21:46:59.8337504Z       print "Size: " + str(backData[0])
> 2020-01-15T21:46:59.8337786Z                    ^
> 2020-01-15T21:46:59.8338094Z ./scripts/daemon_rfid_reader.py:12:14: E999 SyntaxError: invalid syntax
> 2020-01-15T21:46:59.8338382Z print dir_path
> 2020-01-15T21:46:59.8338672Z              ^
> 2020-01-15T21:46:59.8339579Z ./scripts/python-phoniebox/Phoniebox.py:150:9: F821 undefined name 'self'
> 2020-01-15T21:46:59.8340375Z         self.mpd_connect_timeout()
> 2020-01-15T21:46:59.8340670Z         ^
> 2020-01-15T21:46:59.8346868Z ./scripts/python-phoniebox/Phoniebox.py:151:22: F821 undefined name 'self'
> 2020-01-15T21:46:59.8347301Z         mpd_status = self.client.status()
> 2020-01-15T21:46:59.8347668Z                      ^
> 2020-01-15T21:46:59.8348231Z ./scripts/python-phoniebox/Phoniebox.py:152:9: F821 undefined name 'self'
> 2020-01-15T21:46:59.8354353Z         self.set_mpd_playmode(self.lastplayedID)
> 2020-01-15T21:46:59.8354504Z         ^
> 2020-01-15T21:46:59.8355493Z ./scripts/python-phoniebox/Phoniebox.py:152:31: F821 undefined name 'self'
> 2020-01-15T21:46:59.8358977Z         self.set_mpd_playmode(self.lastplayedID)
> 2020-01-15T21:46:59.8359261Z                               ^
> 2020-01-15T21:46:59.8359639Z ./scripts/python-phoniebox/Phoniebox.py:154:9: F821 undefined name 'self'
> 2020-01-15T21:46:59.8359948Z         self.client.play(mpd_status['song'])
> 2020-01-15T21:46:59.8360052Z         ^
> 2020-01-15T21:46:59.8360778Z ./scripts/python-phoniebox/PhonieboxConfigChanger.py:72:46: E999 SyntaxError: invalid syntax
> 2020-01-15T21:46:59.8360925Z             raise configparser.NoSectionError, e
> 2020-01-15T21:46:59.8361373Z                                              ^
> 2020-01-15T21:46:59.8361850Z ./scripts/python-phoniebox/Reader.py:60:34: E999 SyntaxError: invalid syntax
> 2020-01-15T21:46:59.8361985Z             print "Card detected."

So is it planned to migrate all scripts to Python 3?

I guess there has to be a common understanding first, which linter to use and which rules to follow.

s-martin, 2020-01-16:
After reading comments in #721 I assume RegisterDevice.py should be Python3 already?

s-martin, 2020-05-31:
The following GitHub Actions exist:
https://github.com/MiczFlor/RPi-Jukebox-RFID/actions

* Python linter flake8 and some unit tests
* Test of installation script on a Docker image

In my repo I added a branch to add php unit tests, which needs some polishing:
https://github.com/s-martin/RPi-Jukebox-RFID/tree/add-php-action?files=1

s-martin, 2022-12-24:
Many Actions exist, so closing now.

closed 2022-12-24 by s-martin