MiczFlor/RPi-Jukebox-RFID #715: Question: Suggestions for test environment

s-martin, 2020-01-12:
Hi,

 I’m looking for recommendations how to set up a test/dev environment. Maybe there are some tutorials/howtos available, I didn’t find yet.

Currently I only have a productive Phoniebox, which is used daily by my kids and I don’t want to use it for experimenting. Probably I could buy another Pi, but so far I’m looking for other options.

Has anybody already used maybe a VM or WSL and has some recommendations? 
Thanks in advance

MiczFlor, 2020-01-15:
@s-martin 
my thoughts exactly. I tried once to set up 
https://webnist.de/wie-man-raspberry-pi-desktop-unter-virtualbox-einrichtet/
https://www.osboxes.org/raspbian/
but with no success.
If you find something, please share!

s-martin, 2020-01-16:
I'll look into it and share it here.

I just thought maybe it's possible to create a docker image with Raspian for some tests and playing around, which could also be used for automatic testing (e.g. #711) on a build server.

s-martin, 2020-01-19:
#731

ZyanKLee, 2020-01-20:
> #731

This requires a spare RPi, though. And it has some missing/broken parts (iwconfig not available, systemd not working for example). Though they might be fixable.

ZyanKLee, 2020-01-20:
On the other hand - everything in the repository should be independend of the processors architecture type. So modifying the Dockerfile to support debian:buster with amd64 architecture should not be a big problem, or would it @MiczFlor ?

ZyanKLee, 2020-01-20:
created #738 with changes for amd64 systems

s-martin, 2020-01-31:
Possible tool to fake events from devices (e.g. RFID reader): https://github.com/martinpitt/umockdev 

Also mentioned in #747

Groovylein, 2020-03-10:
> Possible tool to fake events from devices (e.g. RFID reader): https://github.com/martinpitt/umockdev

Since I am having some problems installing the RPi-Jukebox-RFID (already mentioned: iwconfig not available, systemd not working for example) I will try this umockdev

topas-rec, 2020-06-09:
I just wanted to start testing for #179.
[Someone suggested](https://raspberrytips.com/run-raspberry-in-virtual-machine/) a Raspberry Pi Desktop image for use in a VM.

_I use virt-manager because it possible to use bluetooth in this machine. (Didn't worked properly in virtualbox.) GNOME boxes was unable to set the VM location on disk and virt-manager is a perfect tool in my eyes._
I found that Phoniebox depends on netcat (nc) which was not installed in the Desktop image by default. After installing netcat and installing Phoniebox I was able to get a working web interface with sound output (even the startup sound)

Side note: Why don't we install netcat in the install script?

[PS: virt-manager is deprecated (but might still be developed) and will be replaced by cockpit.](https://www.techotopia.com/index.php/Creating_KVM_Virtual_Machines_on_RHEL_using_Cockpit_and_virt-manager)

topas-rec, 2020-06-09:
> I just wanted to start testing for #179.
> [Someone suggested](https://raspberrytips.com/run-raspberry-in-virtual-machine/) a Raspberry Pi Desktop image for use in a VM.
> 
> That is much easier handling than a docker container. (Think of snapshots - The reason why I wanted to use a VM)
> 
> _I use [virt-manager](https://virt-manager.org/) because it possible to use bluetooth in this machine. (Didn't worked properly in virtualbox.) GNOME boxes was unable to set the VM location on disk and virt-manager is a perfect tool in my eyes._
> I found that Phoniebox depends on netcat (nc) which was not installed in the Desktop image by default. After installing netcat and installing Phoniebox I was able to get a working web interface with sound output (even the startup sound).
> 
> Side note: Why don't we install netcat in the install script though it is a dependency?
> 
> [PS: virt-manager is deprecated (but might still be developed) and will be replaced by cockpit.](https://www.techotopia.com/index.php/Creating_KVM_Virtual_Machines_on_RHEL_using_Cockpit_and_virt-manager)

Since USB devices can be send to the VM one can test with USB RFID Reader.
GPIO button do of course not work

chilaspt, 2020-06-17:
Hi Everyone!

I'm new here, but I have spent about 30 hours of the last 2 days working on this, 
I have managed to setup this project on a VM in my home lab - esx based and I think I have it working now. ( using an HP Gen 8 microserver )

I'm using a KKMoon RFID reader, and an ASUS bluetooth dongle, both passedthrough to my VM to get audio working. - the speaker is an Onkyo smart speaker with Google home, but I am using it as a plain bluetooth speaker.

 I am using the development build without spotify, not sure if that helps.

I am also trying to set it up on an old HP T620 thin client whilst our pi doesn't get here.
(the plan is to have the VM for testing purposes and to have pi as "production").

Happy to try and share config files if required, or to share some of the issues I have had and everything I tried to fix them. (I'm far from an expert on these issues, but am good at searching the web and trying fixes...) the possibility of taking a snapshot of the VM helps a great deal with trial and error.

s-martin, 2020-06-18:
Hi @chilaspt, great to hear that you got a working test environment.

If you could post a howto or steps how to do it here in this issue, we would add it to the wiki later.