MiczFlor/RPi-Jukebox-RFID #716: Fighting with correct folder structure with spotify

veloxidSchweiz, 2020-01-12:
Hey,
I am really fighting hard with the folder structures and do not really understand where are the correct places for the music, etc.
I am using +Spotify edition
My structure is the following:
Local Music: /home/pi/Music

I really do not understand who the script inc.processAddNewStream.php should work in case of adding a spotify song....
it creates a folder 
`.$Audio_Folders_Path."/".$post['streamFolderName']`
i.e.
`/home/pi/Music/NewSpotifyAlbum`
but then tries to write the spotify.txt file to
`$filestream = $conf['shared_abs']."/audiofolders/".$post['streamFolderName']."/".$filestream;`
i.e.
`/home/pi/Music/audiofolders//NewSpotifyAlbum/spotify.txt`

What should be the correct behaviour....
I would love to help to improve the code there as I think this part of the code seems to be a little  error-prone. Defining the correspoding path variables at the beginnging and only use these would probably help here.
After change that code I was able to play music from spotify.  But i am not sure if i did it in the way it was expected...

I do also have problems to actually play the local music. How can i debug that best?

Cheers,
Felix

veloxidSchweiz, 2020-01-21:
Any one who could help me? Still not sure how to proceed. Which informaiton could be helpful?

ZyanKLee, 2020-01-22:
not sure if this helps: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Architecture_FoldersFiles

music files should be in `/home/pi/RPi-Jukebox-RFID/shared/audiofolders` and subfolders of it.

closed 2020-12-15 by s-martin