MiczFlor/RPi-Jukebox-RFID #717: Send custom command via RFID

hyperbit00, 2020-01-12:
Hey guys,

is there a way to send a custom commad via nfc?

I'd like to disable a service via rfid.

Would be nice if its possible to send a "

> service XYZ stop/start"

 command via RFID

hyperbit00, 2020-01-14:
Solved
Open rfid_trigger_play.sh Line 274 and add
$STOPXYSERVICE)
udo service XYSERVICE stop
;;
$STARTXYSERVICE)
sudo service XYSERVICE start
;;

MiczFlor, 2020-01-15:
Hi @hyperbit00 
genauso lässt sich das erweitern. Welchen Service fährst du hoch und runter? Vielleicht ist es sinnvoll das in die code base aufzunehmen?
Für WiFi hatten wir das schon mit reingenommen:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/fb3e3f63c54d17f92ad6510d2320ddba6f920c05/scripts/rfid_trigger_play.sh#L241
Da wurde dann der Service nicht im gleichen Shell sondern woanders angelegt.

closed 2020-01-15 by MiczFlor

hyperbit00, 2020-01-15:
Es handelt sich hierbei um das Buttons LED Service Script von Olli splittscheid.de 

> > https://github.com/splitti/phoniebox_led_control

und hier hat der Author mir direkt den entscheidenden Hinweis gegeben [http://splittscheid.de/selfmade-phoniebox/#comment-537](url)

that-coding-human, 2020-12-13:
I was looking for this exact same functionality for deactivating/activating the LED service using a RFID card (e.g. for some kind of "night mode")
Having more flexibility would be nice. I would like to achieve two different scenarios using a rfid card:
- activating a "night mode", which lowers the max volume
- shuts of the LEDs in my arcade buttons
- sets a shutdown timer to e.g. 30 minutes

A second RFID card should show some system information on my oled display, like:
- current IP adress
- WLAN ssid
- system name 
etc. you name it.

Some of the ideas people have are quiet specific. So having a possibility to execute a script by linking the call with a RFID card, would provide maximum flexibility.
Are there any concerns to not implement a "call custom script" function call?

s-martin, 2020-12-13:
I don't see any reasons why not do this.

It just needs somebody to come up with a flexible concept and implement it in a sustainable and extendable way.

Please open a new enhancement issue or PR, if you want to discuss and provide a proposal.

If you want to discuss some vague ideas, you could open a new thread under Discussions Ideas.

Please don't continue the discussion in a closed issue, these comments usually get lost, because most people don't read in closed issues.

that-coding-human, 2020-12-13:
I came here from #913 and missed the "closed" status. (Maybe I shouldn't be on github late at night :-) ) Sorry for the confusion. 
#913 seems appropriate to followup on this, as it is already flagged as enhancement, as it seems to be exactly what I am thinking about, and it has not much content yet. No need for issue sprawl. I will think about an implementation and put my input about a proposal in there.

s-martin, 2020-12-13:
> I came here from #913 and missed the "closed" status. (Maybe I shouldn't be on github late at night :-) ) Sorry for the confusion. 

No problem, i just want to make sure discussions don't get lost.