MiczFlor/RPi-Jukebox-RFID #718: Phonibox Installation klappt nicht

Ghostsl84, 2020-01-12:
Guten Abend,

ich versuche seit mehreren Stunden eine 2 Phoniebox aufzusetzen doch leider vergebens.
Ich nutzte einen Rapsberry Pi 3b+ mit Rasbian Buster desktop (habe es auch schon mit der Lite versucht).
Egal welchen link ich nutze zum Installieren der Phonibox (Buster, Stretch) ich bekomme es nicht zum laufen.
Der Raspberry ist an einem Kabel angeschlossen also kein Wlan.
Ich komme also nicht auf die Weboberfläche und "nameserver" hatte ich auch schon probiert und dies klappte leider auch nicht.
Ich führe auch mal meine Installationsverlauf auf und hoffe das Ihr mir weiterhelfen könnt.

START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes
# between 15 minutes to half an hour, depending on
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [261 kB]
Fetched 13.3 MB in 8s (1,583 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-headers-4.9.0-6-all' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-all-armhf' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi2' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common-rt' for regex 'linux-headers-4.9'
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by glob 'php7.0-cgi'
E: Couldn't find any package by regex 'php7.0-cgi'
E: Unable to locate package php7.0-fpm
E: Couldn't find any package by glob 'php7.0-fpm'
E: Couldn't find any package by regex 'php7.0-fpm'
Cloning into 'RPi-Jukebox-RFID'...
remote: Enumerating objects: 5600, done.
remote: Total 5600 (delta 0), reused 0 (delta 0), pack-reused 5600
Receiving objects: 100% (5600/5600), 16.48 MiB | 8.19 MiB/s, done.
Resolving deltas: 100% (3715/3715), done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting evdev==0.7.0 (from -r requirements.txt (line 7))
  Downloading https://files.pythonhosted.org/packages/67/15/eac376f3e1fc1960a54439c21459b2582e68340001aff83b4ace9e5bd110/evdev-0.7.0.tar.gz
Collecting spi-py from git+git://github.com/lthiery/SPI-Py.git#egg=spi-py (from -r requirements.txt (line 8))
  Cloning git://github.com/lthiery/SPI-Py.git to /tmp/pip-install-6ubahp/spi-py
Collecting youtube_dl (from -r requirements.txt (line 9))
  Downloading https://files.pythonhosted.org/packages/ec/4c/4aa3a86dc43623b9d5a4b0df2915908413703adc1899965c4b2e00bb8a4d/youtube_dl-2020.1.1-py2.py3-none-any.whl (1.8MB)
    100% |████████████████████████████████| 1.8MB 158kB/s
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 10)) (3.4)
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 12)) (0.7.0)
Collecting pi-rc522 (from -r requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/84/f2/e3d02257949e9caa9bff26044e0185e743ff45d3e0e099a93880e10bf718/pi-rc522-2.2.1.tar.gz
[... 55 lines cut ...]
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] y
Choose the reader from list
0 HXGCoLtd Keyboard
Device Number: 0

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/

vielen Dank
Ghostsl

Ghostsl84, 2020-01-13:
Guten Abend,

ich habe es doch geschaft, mit einer alten Rasbian Stretch von dieser Seite die Phoniebox zum laufen zu bekommen.
Ich kann auch Karten registrieren und Musik hochladen, jedoch wird keine Karte abgespielt.
Kann mir da jemand weiterhelfen?

Vielen Dank
Ghostsl

FloMrFrei, 2020-01-14:
Hi Ghostsl84

bei mir hat es letztlich mit einem Rapsi2 und Raspian Buster geklappt (Phoniebox-Spotify). Lösungen sind hier:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/700 
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/688 

In aller kürze:
1.) libspotify-dev installieren (sonst scheitert die Installation - egal ob Spotify oder nicht - wenn ich das Auto-Install-Script richtig verstehe)
```
wget -q -O - https://apt.mopidy.com/mopidy.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/buster.list
sudo apt update
sudo apt-get install libspotify-dev
```

2.) One-Line-Installer (Buster)

3.) Mopidy downgrade
```
sudo apt-get install mopidy=2.3.1-1
sudo python2.7 -m pip install Mopidy==2.3.*
sudo pip install 'tornado==5.0'
```

Viel Erfolg!

MiczFlor, 2020-01-15:
Hi @FloMrFrei 
kannst du eine Pull request mit deinen Änderungen zum Install script für Buster machen? Ich bin gerade "on the road" und habe keine Testumgebung (ein VirtualBox VM für Raspbian wäre cool...).

@Ghostsl84 
Ich hatte das manchmal: wenn man mit Kabel verbunden ist, läuft apt-get nicht. Ich habe das nicht weiter verfolgt. WiFi ging immer. Ich hatte das Gefühl das ist ein Raspbian Problem.
Bei deiner Stretch Installation: geht di Web app?

Ghostsl84, 2020-01-17:
Guten Abend,

sorry das es so lange gedauert hat, war viel unterwegs.
Ja die Web App funktioniert, ich kann auch Daten Hochladen und die Karte einem Album zuweisen, aber sobald ich die Karte über den RFID Scanner ziehe pipst es, aber der Player startet nicht.
Ich kann das Album auch nicht über die Web App manuell starten.
Früher musste man die Bibliotheke aktualisieren, aber diesen Button finde ich nicht.

Gruß
Ghostsl

splitti, 2020-01-18:
Bei dir gibt es Python Probleme. Der neue Installer muss einmal von Micz freigegeben werden, dann wird alles funktionieren.
Ich vermute du hast dir Classic Version? Dort gibt es keinen Folder Scan.

Ghostsl84, 2020-01-18:
Guten Morgen,

genau das Fehlt.
Aber mit der Buster Installation wie SIe von FloMrFrei beschrieben wurde klappt alles super.
Mal noch eine frage, auch bei dieser Installation bekomme ich meinen Bluetooth Lautsprecher nicht in die gänge.
Der Lautsprecher wurde wie immer gepairt, connect und trust alles klappt wunderbar also die Verbindung zwischen Raspberry und meinem Lautsprecher.
Aber er spielt mir auch auf dieser Box keinen Sound.
Habe auch alles mit Bluealsa probiert ohne erfolg, es kommt kein Ton aus dem Lautsprecher.

habe diese Aleitung befolgt und damals schon alles probiert aber ohne erfolg:

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/284

wenn ich diesen Befehl nutze:
aplay -D btspeaker /usr/share/sounds/alsa/Front_Center.wav

erhalte ich folgenden Fehler:
ALSA lib conf.c:1891:(_snd_config_load_with_include) :25:1:Unexpected }
ALSA lib conf.c:3650:(config_file_open) /etc/asound.conf may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3572:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:4026:(snd_config_update_r) hooks failed, removing configuration
aplay: main:828: audio open error: Invalid argument

vielleicht habt ihr eine Lösung dafür.

Vielen Dank
Ghostsl

MiczFlor, 2020-01-18:
Hi @Ghostsl84 
probier noch mal einen neuen Install. @splitti hat sich die letzte Nacht um die Ohren geschlagen, um das hinzubekommen. Bei ihm läuft's. Ich habe es committed, aber noch nicht getestet.

Ghostsl84, 2020-01-18:
Hi,

vielen Dank, hab es ausporbiert und läuft.
Aber jetzt hat man bei Verbinden eins Bluetooth Lautsprechers folgendes:

[bluetooth]# pair F4:4E:FD:AB:0C:35
Attempting to pair with F4:4E:FD:AB:0C:35
[CHG] Device F4:4E:FD:AB:0C:35 Connected: yes
Request confirmation
[agent] Confirm passkey 271642 (yes/no): yes
[CHG] Device F4:4E:FD:AB:0C:35 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device F4:4E:FD:AB:0C:35 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device F4:4E:FD:AB:0C:35 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device F4:4E:FD:AB:0C:35 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device F4:4E:FD:AB:0C:35 ServicesResolved: yes
[CHG] Device F4:4E:FD:AB:0C:35 Paired: yes
Pairing successful
[CHG] Device F4:4E:FD:AB:0C:35 ServicesResolved: no
[CHG] Device F4:4E:FD:AB:0C:35 Connected: no
[bluetooth]# trust F4:4E:FD:AB:0C:35
[CHG] Device F4:4E:FD:AB:0C:35 Trusted: yes
Changing F4:4E:FD:AB:0C:35 trust succeeded
[bluetooth]# connect F4:4E:FD:AB:0C:35
Attempting to connect to F4:4E:FD:AB:0C:35
Failed to connect: org.bluez.Error.Failed
[bluetooth]# connect F4:4E:FD:AB:0C:35
Attempting to connect to F4:4E:FD:AB:0C:35
Failed to connect: org.bluez.Error.Failed

über Kabel klappt aber alles wunderbar

Muss ich jetzt etwas umstellen?
Und wie Kann man eine Titel weiterkarte erstellen

Sorry fragen über fragen, aber ich verstehe das nicht so richtig.

Vielen Dank

ZyanKLee, 2020-01-18:
hallo @Ghostsl84 in deinem ersten Beitrag wegen den Bluetooth-Lautsprechern hast du die Ausgabe von aplay zitiert. Darin erscheint folgende Meldung: '/etc/asound.conf may be old or corrupted: consider to remove or fix it'
Was steht denn in dieser Datei?

Ghostsl84, 2020-01-18:
HI

in der Datei steht folgendes:

pcm.btspeaker {

type plug

slave.pcm {

    type bluealsa

    interface "hci0"

    device "F4:4E:FD:AB:0C:35"

    profile "a2dp"

    delay -20000

}

hint {

    show on

    description "SoundCore mini"

}

}

MiczFlor, 2020-01-22:
Hi @Ghostsl84 
kannst du ein neues Ticket aufmachen und das hier schließen? Am besten mit `Bluetooth` im Titel, damit sich vielleicht andere mit einklinken können, die damit Erfahrung haben.
Danke, micz

Ghostsl84, 2020-01-23:
Mache ich 

Danke für die Hilfe

closed 2020-01-23 by Ghostsl84