MiczFlor/RPi-Jukebox-RFID #721: Test and improve latest install script for Buster

MiczFlor, 2020-01-15:
Hi everybody,
Please use the one line install script from the below page that says "For latest updates, try the version in the `develop` branch":
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#raspbian-buster

Please post your results and bug fixes and feedback in this thread. I want to take the discussion from these posts to this new thread to make it easier to manage change:

* https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/718
* https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/700
* https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/688

splitti, 2020-01-15:
I've used today the normal installer. After the installation, i made these steps:
sudo apt-get install mopidy=2.3.1-1
sudo python2.7 -m pip install Mopidy==2.3.*
pip install 'tornado==5.0'

They are working fine to me. I will try a brand new installation in the next days! But a problem is, the RegisterDevice.py is not the python3 version, it is the old version of the Master repository in python2 Style.

patrickweigelt, 2020-01-15:
Hi @splitti You are right. The line `python3 RegisterDevice.py` should be `python2 RegisterDevice.py` like in the install script on the master branch. I must have overlooked that when solving the conflicts between my suggestions and the install script on the develop branch.

Actually also IRIS should work. It does on my boxes. Unfortunately I didn't have the time to run the script again today. I hope I will manage to try a new installation from scratch tomorrow evening

patrickweigelt, 2020-01-15:
I created a pull request for what I mentioned above. However I was wondering if we would maybe need a second version or another branch for the buster installer script. One that makes the old mopidy 2.3.1-1 with IRIS run and uses all files as they are on the master branch at the moment and one that prepares for the use of mopidy 3 once all libraries are migrated (including IRIS). I am suggesting this because I have the feeling that I changed back things that others prepared for being able to use mopidy 3 in the future.

splitti, 2020-01-15:
Wait, it must be tested with a fresh Installation. I will make an Installation tomorrow... Maybe the RegisterDevice.py was drin the Master repo. I am not Sure, because in the dev Environment it looks fine.

beng81, 2020-01-15:
In the last step I get the following error:

Have you connected your USB Reader? [Y/n] Y
  File "RegisterDevice.py", line 9
    print "Choose the reader from list"
                                      ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Choose the reader from list")?
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

DONE. Let the sounds begin.

splitti, 2020-01-16:
@MiczFlor why is the nee RegisterDevice.py not in the dev repo? It is python2 Script, Not the nee one.
You could Change to python2 in the Installer or Change the File to the new one.

splitti, 2020-01-16:
There are Phyton Pip Errors before the RegisterReader. I tried also also git clone --branch develop in the Script. In this Case, the Script took the python3 Version. But there are also errors. It tried the python2 Version, also errors because of missing Python endev. I am really confused of that. Spotify isn't working for me with a fresh Installation, i tried the Develop Installer with cloning master and another time with Develop Branch, both time a fresh installed SD Card. I have to Invest time in the Installer and of course, Python...

MiczFlor, 2020-01-16:
Hi @splitti @patrickweigelt @beng81 
I just merged the pull request for register device on python2 with develop.
Thanks for the input. I feel motivated and a bit pushed :) in a good way :)

splitti, 2020-01-16:
That is not the solution, the problem is, the mopidy Installation does not work. There are dependencies in the setup.py for the following pip Installations. And so on, the RegisterDevice does not work. Python3 for Register Device is correct.

I will send you some new code, maybe this fixes all problems...

splitti, 2020-01-16:
So, there are different files to be fixed to python3-compatibility. RegisterDevice is working for me, the setup.py need to be executed be python2. The RFid Service needs python3 compatibility. The service itself need a refresh to python3. I think there is a lot of work because of python3! These steps could work, please add the last Requests to the develop branch! Please notice, that i clone the develop branch in the installer this time!

MiczFlor, 2020-01-17:
Hi @splitti 
thanks for your input (and, obviously, your great blog posts a few months back :)
I merged all to masters. There were a few conflicts to resolve. I haven't tested it yet, but I believe it should work as you intended.
I got the cloning in the install script. Not sure if this is the best way or if we should use the develop install to install the master code base. What is your intention to switch to develop?

splitti, 2020-01-17:
This is Just dir loading the Files python3 ready. It is more a package, but i edited the Files in the Browser, so there are different Pull requests. The File should bei updated finally together to master Branch, and the Clone should bei changed to master also for the master Branch.

patrickweigelt, 2020-01-17:
Hi @splitti Thanks a lot for your effort. I think this is bringing it much forward. My take on it or how I understood @MiczFlor was to first make the old installer work with the old py files to have a working version including spotify and IRIS and to be able to push it to the master branch as soon as possible and then work on the updates of the python scripts and do some testing on the develop branch. I thought I had a working version including IRIS (is running on my two boxes now) but I didn't manage to try another fresh install so far to confirm the installer was working properly. Sorry for that. The way we're approaching it now is nice because it moves things forward much more quickly. I just fear there is not much I can contribute at this stage any more as both Raspbian and python are not really my expertise. Is there something I should test. Now that the develop install script also clones the develop repository there is probably a lot new stuff. Is is all working? Is IRIS running? Cheers

splitti, 2020-01-17:
Hmm, for that Problem, i Just can make a wget of the 3 files and Take the master Branch. You are right, maybe we get other Code making Problems... I have no Idea to optimize my Dev Environment this time. Please wait up to sunday, i have a Lot of Work this Weekend...

splitti, 2020-01-17:
@MiczFlor could you apply the new pull request, i make a fresh installation to test it...

sglock, 2020-01-17:
Hi, thanks for the work you put into the project. Can't get the RFID Reader to register. Tried both one line installers but every time i try to run `RegisterDevice.py` same error.

> File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 9
    print "Choose the reader from list"
                                      ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Choose the reader from list")?

Tried to clean up the code but no luck. Maybe you have a tip.

splitti, 2020-01-17:
@sglock 
In which time did you Install? And which Branch, Master or Develop

sglock, 2020-01-17:
@splitti installed about 2 hours ago. First I tried the develop and then the master branch. 
Have to admit I didn't set up from scratch. Used my already installed system.

RealTatron, 2020-01-17:
Moin,
I installed a fresh 2019-09-26-raspbian-buster-lite image on my raspberry pi 4 today, 1 hour ago.
After updating with
```
sudo apt update
sudo apt full-upgrade
sudo reboot
```
I started the Phoniebox master install (no Spotify selected) with the following result:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libspotify-dev
./buster-install-default.sh: line 494: git: command not found
./buster-install-default.sh: line 497: cd: RPi-Jukebox-RFID: No such file or directory
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/smb.conf.buster-default.sample': No such file or directory
chown: cannot access '/etc/samba/smb.conf': No such file or directory
chmod: cannot access '/etc/samba/smb.conf': No such file or directory
sed: can't read /etc/samba/smb.conf: No such file or directory
sudo: smbpasswd: command not found
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/lighttpd.conf.buster-default.sample': No such file or directory
chown: cannot access '/etc/lighttpd/lighttpd.conf': No such file or directory
chmod: cannot access '/etc/lighttpd/lighttpd.conf': No such file or directory
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/15-fastcgi-php.conf.buster-default.sample': No such file or directory
chown: cannot access '/etc/lighttpd/conf-available/15-fastcgi-php.conf': No such file or directory
chmod: cannot access '/etc/lighttpd/conf-available/15-fastcgi-php.conf': No such file or directory
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/php.ini.buster-default.sample': No such file or directory
chown: cannot access '/etc/php/7.3/cgi/php.ini': No such file or directory
chmod: cannot access '/etc/php/7.3/cgi/php.ini': No such file or directory
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/sudoers.buster-default.sample': No such file or directory
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/settings/rfid_trigger_play.conf.sample': No such file or directory
./buster-install-default.sh: line 555: /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name: No such file or directory
./buster-install-default.sh: line 556: /home/pi/RPi-Jukebox-RFID/settings/Audio_Folders_Path: No such file or directory
./buster-install-default.sh: line 557: /home/pi/RPi-Jukebox-RFID/settings/Audio_Volume_Change_Step: No such file or directory
./buster-install-default.sh: line 558: /home/pi/RPi-Jukebox-RFID/settings/Max_Volume_Limit: No such file or directory
./buster-install-default.sh: line 559: /home/pi/RPi-Jukebox-RFID/settings/Idle_Time_Before_Shutdown: No such file or directory
./buster-install-default.sh: line 560: /home/pi/RPi-Jukebox-RFID/settings/Second_Swipe: No such file or directory
./buster-install-default.sh: line 561: /home/pi/RPi-Jukebox-RFID/settings/Playlists_Folders_Path: No such file or directory
./buster-install-default.sh: line 562: /home/pi/RPi-Jukebox-RFID/settings/ShowCover: No such file or directory
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/htdocs/config.php.sample': No such file or directory
sudo: lighttpd-enable-mod: command not found
sudo: lighttpd-enable-mod: command not found
Failed to restart lighttpd.service: Unit lighttpd.service not found.
Failed to restart php7.3-fpm.service: Unit php7.3-fpm.service not found.
cp: cannot stat '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.py.sample': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py': No such file or directory
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/*.sh': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/*.sh': No such file or directory
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/*.py': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/*.py': No such file or directory
### Deleting older versions of service daemons. This might throw errors, ignore them
Failed to disable unit: Unit file idle-watchdog.service does not exist.
Failed to disable unit: Unit file rfid-reader.service does not exist.
Failed to disable unit: Unit file startup-sound.service does not exist.
Failed to disable unit: Unit file gpio-buttons.service does not exist.
rm: cannot remove '/etc/systemd/system/rfid-reader.service': No such file or directory
rm: cannot remove '/etc/systemd/system/startup-sound.service': No such file or directory
rm: cannot remove '/etc/systemd/system/gpio-buttons.service': No such file or directory
rm: cannot remove '/etc/systemd/system/idle-watchdog.service': No such file or directory
[... 50 lines cut ...]
#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n]
./buster-install-default.sh: line 833: cd: /home/pi/RPi-Jukebox-RFID/scripts/: No such file or directory
python2: can't open file 'RegisterDevice.py': [Errno 2] No such file or directory
[... cut ...]
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
Reboot is needed to activate all settings
Would you like to reboot now? [Y/n]
```
If I do a develop install followed by the master install I get a working system without working RFID Reader, because of the same error like user sglock.

Tell me if I should do some more testing with a fresh image.

Regards,
Tatron

RealTatron, 2020-01-17:
RFID error in both install scripts:
```
Have you connected your USB Reader? [Y/n]
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, categorize, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
```

ZyanKLee, 2020-01-17:
@RealTatron had this issue once myself. the first error tells you what went wrong:
`./buster-install-default.sh: line 494: git: command not found`

just install git before running the script again:
`sudo apt-get update && sudo apt-get -y install git`

splitti, 2020-01-17:
All done in the past hours, the Onelineinstaller is working fine now... The most problems are because of incompatibility between python2 and python3.
The installation works including Iris, Mopidy and also the python2 environment for "old" scripts. Please make it as soon as possible to master: #730 

This Pull requests are needed. But they are working ONLY in master branch, it you want to test it, please relace the files in master imager: 
RegisterDevice.py MiczFlor #707 
phoniebox-rfid-reader.service.stretch-default.sample MiczFlor #729

s-martin, 2020-01-18:
@ZyanKLee Would it make sense to also add git to the packages to be installed first?

sglock, 2020-01-18:
@splitti Just finished a complete fresh install with master branch. Registering RFID reader failed

> pi@phoniepi:~ $ python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, categorize, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
>
Using the 
> /Reader.py.experimental
>
throws the old error

> pi@phoniepi:~ $ cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.experimental /home/pi/RPi-Jukebox-RFID/scripts/Reader.py
pi@phoniepi:~ $ python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 61
    print "Card detected."
                         ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Card detected.")?
>
Hope that helps finding the errors.

splitti, 2020-01-18:
Use python2.7 to call the python-File. It was working for me with that installation. Maybe it must be 2.7, the most changes were transported manually.

splitti, 2020-01-18:
This is working for me, please check:

```
cd /home/pi/RPi-Jukebox-RFID/scripts/
python2 RegisterDevice.py
```

sglock, 2020-01-18:
@splitti Works! 
Thx for your help!

ZyanKLee, 2020-01-18:
> @ZyanKLee Would it make sense to also add git to the packages to be installed first?

@s-martin  definitely

splitti, 2020-01-18:
Git should bei installed, but in there were Errors in eine cases. With the new Installer should it work.

JoeyRamone, 2020-01-19:
@splitti @MiczFlor 
I've checked the current master onelineinstaller with a Raspi 4 now (fresh installation) and everything seems to work fine. I've tested it before with an "re-installation", but this ended in multiple errors. 
thank you guys, for all the time you've already spent for this project.

s-martin, 2020-01-19:
Ideas for improvement:
* If the classic version is used, is the installation of libspotify really necessary or should this be moved to the spotify section?
* the installation directory `/home/pi` could be substituted with a variable
* are there reasons (besides finding existing installations) that `/home/pi` is always the installation directory?
* dont the remaining python packages like `python-pip` need to be also python3?

fatalerror1981, 2020-01-21:
This failed with below message during the buster master onelineinstaller 

E: Version '2.3.1-1' for 'mopidy' was not found

this is after a fresh raspian buster lite installation

fatalerror1981, 2020-01-21:
+ fresh buster lite and buster develop onlineinstaller fails also with similiar messages:

Building dependency tree
Reading state information... Done
E: Version '2.3.1-1' for 'mopidy' was not found
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Mopidy==2.3.*
Could not find a version that satisfies the requirement Mopidy==2.3.* (from versions: )
No matching distribution found for Mopidy==2.3.*

MiczFlor, 2020-01-22:
Hi @fatalerror1981 
did the install fail or give warnings?
And what hardware are you using?

Vanierica, 2020-01-24:
@MiczFlor I've the same Problem and posted my output in #741 few days ago. Maybe this helps. I'm using Pi 3 Model B.

ZyanKLee, 2020-01-24:
>  E: Version '2.3.1-1' for 'mopidy' was not found

that might be caused by conversion to python3 as mopidy2 is not compatible with python3:
` Requires: Python >= 2.7, < 3 `
https://pypi.org/project/Mopidy/2.3.1/

fifox, 2020-01-25:
Hi,
I'm facing to exact the same problem. 
Unfortunately, i have backup only RPi-Jukebox-RFID folder and I need the box running until next friday. 

Question: How can I install an old "rock stable" version i.e. without python3?
If I remember right, the version from december 2019 was working fine.

Shaquille23, 2020-01-26:
Did a fresh installation with the master branch and spotify enabled on Raspberry 4 with Buster Lite.

`Failed to enable unit: Unit file mopidy.service does not exist.

Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
cp: cannot create regular file '/etc/mopidy/mopidy.conf': No such file or directory
cp: cannot create regular file '/root/.config/mopidy/mopidy.conf': No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /root/.config/mopidy/mopidy.conf: No such file or directory
sed: can't read /root/.config/mopidy/mopidy.conf: No such file or directory
sed: can't read /root/.config/mopidy/mopidy.conf: No such file or`

The Webinterface is working, but http://192.168.xx.xxx:6680/iris shows a 404 Error.

marsuu, 2020-01-26:
> > E: Version '2.3.1-1' for 'mopidy' was not found
> 
> that might be caused by conversion to python3 as mopidy2 is not compatible with python3:
> `Requires: Python >= 2.7, < 3`
> https://pypi.org/project/Mopidy/2.3.1/

Also facing the same issues (master and develop online installer) after upgrading buster and reinstallation. With fresh installation mopidy will not be installed with the online installer.
Perhaps Mopidy 2.3.1-1 no longer contained within the buster mopidy apt repository?

ZyanKLee, 2020-01-26:
Would you be willing to test my current version of the installer?

If so, please start from a fresh buster lite image with following commands:

```
cd /home/pi
wget https://raw.githubusercontent.com/ZyanKLee/RPi-Jukebox-RFID/stretch-dockerfiles/scripts/installscripts/buster-install-default.sh
GIT_BRANCH=develop bash buster-install-default.sh
```

Shaquille23, 2020-01-26:
> Would you be willing to test my current version of the installer?
>
> If so, please start from a fresh buster lite image with following commands:
>
> ```
> cd /home/pi
> wget [...]

```
E: Version '2.3.1-1' for 'mopidy' was not found
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Mopidy==2.3.*
  Downloading https://files.pythonhosted.org/packages/5f/d5/867cb8500d6c1dce1bcc4b9ef7eb467148fcf801e15b47c39d55bcac01aa/Mopidy-2.3.1-py2.py3-none-any.whl (215kB)
    100% |████████████████████████████████| 225kB 1.1MB/s
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from Mopidy==2.3.*) (40.8.0)
Collecting Pykka>=1.1 (from Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/d7/1d/acaefaa25356b4cbc027eef320616b2f82406ea74516318b70c81b513d59/Pykka-2.0.2-py2.py3-none-any.whl
Collecting requests>=2.0 (from Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl (57kB)
    100% |████████████████████████████████| 61kB 2.7MB/s
Collecting tornado<6,>=4.4 (from Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/e6/78/6e7b5af12c12bdf38ca9bfe863fcaf53dc10430a312d0324e76c1e5ca426/tornado-5.1.1.tar.gz (516kB)
    100% |████████████████████████████████| 522kB 643kB/s
Collecting idna<2.9,>=2.5 (from requests>=2.0->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl (58kB)
    100% |████████████████████████████████| 61kB 2.1MB/s
Collecting chardet<3.1.0,>=3.0.2 (from requests>=2.0->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 1.6MB/s
Collecting certifi>=2017.4.17 (from requests>=2.0->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/b9/63/df50cac98ea0d5b006c55a399c3bf1db9da7b5a24de7890bc9cfd5dd9e99/certifi-2019.11.28-py2.py3-none-any.whl (156kB)
    100% |████████████████████████████████| 163kB 1.5MB/s
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests>=2.0->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/e8/74/6e4f91745020f967d09332bb2b8b9b10090957334692eb88ea4afe91b77f/urllib3-1.25.8-py2.py3-none-any.whl (125kB)
    100% |████████████████████████████████| 133kB 1.8MB/s
Collecting backports_abc>=0.4 (from tornado<6,>=4.4->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/7d/56/6f3ac1b816d0cd8994e83d0c4e55bc64567532f7dc543378bd87f81cebc7/backports_abc-0.5-py2.py3-none-any.whl
Collecting futures (from tornado<6,>=4.4->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/d8/a6/f46ae3f1da0cd4361c344888f59ec2f5785e69c872e175a748ef6071cdb5/futures-3.3.0-py2-none-any.whl
Collecting singledispatch (from tornado<6,>=4.4->Mopidy==2.3.*)
  Downloading https://files.pythonhosted.org/packages/c5/10/369f50bcd4621b263927b0a1519987a04383d4a98fb10438042ad410cf88/singledispatch-3.4.0.3-py2.py3-none-any.whl
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=4.4->Mopidy==2.3.*) (1.12.0)
Building wheels for collected packages: tornado
  Running setup.py bdist_wheel for tornado ... done
  Stored in directory: /root/.cache/pip/wheels/6d/e1/ce/f4ee2fa420cc6b940123c64992b81047816d0a9fad6b879325
Successfully built tornado
Installing collected packages: Pykka, idna, chardet, certifi, urllib3, requests, backports-abc, futures, singledispatch, tornado, Mopidy
Successfully installed Mopidy-2.3.1 Pykka-2.0.2 backports-abc-0.5 certifi-2019.11.28 chardet-3.0.4 futures-3.3.0 idna-2.8 requests-2.22.0 singledispatch-3.4.0.3 tornado-5.1.1 urllib3-1.25.8
Reading package lists... Done
Building dependency tree
Reading state information... Done
libspotify12 is already the newest version (12.1.103-0mopidy1).
libspotify12 set to manually installed.
Suggested packages:
  python-ply-doc python-spotify-doc
The following NEW packages will be installed:
  python-cffi python-ply python-pycparser python-spotify
0 upgraded, 4 newly installed, 0 to remove and 1 not upgraded.
Need to get 319 kB of archives.
After this operation, 1,833 kB of additional disk space will be used.
Get:2 https://apt.mopidy.com buster/contrib armhf python-spotify armhf 2.1.3-0mopidy1 [95.2 kB]
Get:1 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf python-ply all 3.11-3 [65.1 kB]
Get:3 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf python-pycparser all 2.19-1 [75.7 kB]
Get:4 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf python-cffi all 1.12.2-1 [83.2 kB]
Fetched 319 kB in 1s (456 kB/s)
Selecting previously unselected package python-ply.
(Reading database ... 98688 files and directories currently installed.)
Preparing to unpack .../python-ply_3.11-3_all.deb ...
Unpacking python-ply (3.11-3) ...
Selecting previously unselected package python-pycparser.
[... 303 lines cut ...]
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
cp: cannot create regular file '/etc/mopidy/mopidy.conf': No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
[... cut ...]
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
rm: cannot remove '/home/pi/RPi-Jukebox-RFID/settings/global.conf': No such file or directory

#
# INSTALLATION FINISHED
#
#####################################################
```

The webinterface works, but http://192.168.xx.xx:6680/iris gives a 404 error.

felixlindemann, 2020-01-26:
Thank you for this script...
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/721#issuecomment-578498842

unfortunately it didn't work for me as well. same as @Shaquille23 reported.
for me it seems as if mopidy is not installed.

ZyanKLee, 2020-01-26:
they did some baaad thing with deleting 2.3.x releases

marsuu, 2020-01-27:
Thanks for the provided install script  @ZyanKLee :
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/721#issuecomment-578498842

works good for me, after downgrading to 2.3.1-1 from apt-cache (luckily available due to previous faulty upgrade) with small dirty (useless for fresh installs) fix in your install script:

```
#sudo apt-get install --yes mopidy=2.3.1-1
sudo dpkg -i /var/cache/apt/archives/mopidy_2.3.1-1_all.deb
sudo apt-mark mopidy
```

splitti, 2020-01-27:
Sorry, i leaved the project a few days alone and i am currently 200% at my RL because life is sometimes an a**hole. 

But i am a little bit confused about the installer. The last time i had a ready one for the Spotify Version, also with working Iris, Reader Setup (python3) and Reader Service (python 2.7). The problem seems to be the classic version and missing packets. I think firstly we should clarify the needed packages for the version Mopidy + Iris and the other dependencies, and on the other hand also for the Classic Version. I agree with you, that we should go up to python 3. What is the current problem getting the installer working? Are the old releases of Mopidy really deleted?

I get the three new Pi's to test in the next days and i will try to code a completly new Installer. We need a good overview which packages are really needed! Did someone invest some time in finding the needed packages?

ZyanKLee, 2020-01-27:
Not yet. I mostly concentrated on getting those three installers back on track, when suddenly those buster mopidy 2.3 packages disappeared.

Perhaps it would be best to ditch the apt repo and go for the pip repos instead? But in that case the requirements.txt should be used for all python deps. Requirements.txt files can include others, btw. (Add a line with `-r global-req.txt` for example)

ZyanKLee, 2020-01-27:
That way we could try to keep as much of the dependencies in a central file, add stretch/buster specifics in two others and add more for Spotify or gmusic stuff and then we can collect those into 'collections' like "buster w/ Spotify and gmusic" or "stretch w/ otherStuff" in which almost only includes would be.

Then we only need to keep the stuff up to date in a single file at most.

splitti, 2020-01-27:
Sounds like a really fresh Installer, based on Module steps for OS and so on

s-martin, 2020-01-28:
Is there a reason why we use mopidity only for Spotify version and not for everything?

splitti, 2020-01-28:
For local Music is the mpd Service much faster, starting and playing... Mopidy needs an Internet Connection. In my opinion, this is a good way to have more Performance and to react on use cases.

s-martin, 2020-01-28:
> For local Music is the mpd Service much faster, starting and playing... Mopidy needs an Internet Connection. In my opinion, this is a good way to have more Performance and to react on use cases.

Ok, understood.

ZyanKLee, 2020-01-28:
In the other hand it introduces a lot more complexity with a second software that solves mostly the same problem.

ZyanKLee, 2020-01-28:
A completely different idea would be to generate whole images.

splitti, 2020-01-28:
Yep, in my opinion also a good way. Here we can be sure, that there is no Installation problem.
Please Tell me, what problems came after my Installer? I think one was the Classic Version, another that Mopidy cancled supporting Version 2.3.x?

s-martin, 2020-01-28:
I think clearly separating dependencies (and switching to pip) for each variant is a good idea. Each variant should install all dependencies necessary to run alone.

Maybe we can add tests so each python script is started at least and doesn't throw exceptions. The  docker images provided by @ZyanKLee probably would help for such a task.

splitti, 2020-01-28:
For a couple of weeks at the beginning of everything goes bad with the installert, i said in my opinion it is a incompatibility beteween the different python versions. Just calling the file the right way and the installer was working. Ok, for now, if 2.3.x is gone, that was only a short workaround.

I am not sure, if docker is the best way for testing installations. I am building a box just for testing, with LEDs, Display and all i need. I also buyed a set of rotary knobs for testing (i don't use them in my boxes). In my opinion, docker is nice for development, if the environment is running in Raspbian Buster. And i am a Docker Fan! But sometimes there are Hardware problems caused by simple dependencies. Thats why i want a simple test box for me and for developing. 
Please go on with docker, maybe my thinking is wrong at this place, it is just my opinion.

I look forward to make a "clean" installation step by step without installer. I will make a documentation and based on that, a new installer could be written. And we need that for Classic, Spotify and maybe Google Music. Sounds like a lot of fun, work and to less time ;-)

ZyanKLee, 2020-01-28:
You are correct in thinking docker won't solve all testing scenarios. Especially hardware related stuff can't be tested within docker containers. Good thing you are building a box dedicated for that.
But there are things that can be tested on docker and as lot faster than on a raspberry pi.

splitti, 2020-01-28:
True, i love docker. Maybe on Weekend i start the Project.

kuesimon, 2020-01-28:
Hi Guys. First of all thanks for this great Project and all the effort you put into it!

I tested the Install Script for stretch (+Spotify) and both official Installer for Buster (master and Develop).

I can confirm the behaviour described by @Shaquille23.

As the "Missing mopidy" topic ist currently overlaying the Python 3 Integration i believe there will be No "quick fix" in the next hours/days - am i right (Just asking because of overthinking my Weekend plans...😉)

Thanks for all your effort.

notapirate, 2020-02-02:
Created pull request #771 to install mopidy-2.2.2-1 from main repository to fix the issue "E: Version '2.3.1-1' for 'mopidy' was not found"

[2 more replies]

closed 2020-03-07 by ZyanKLee