MiczFlor/RPi-Jukebox-RFID #734: Web App not running?

Erechtheus, 2020-01-18:
Hey folks! I successfully installed this project (no error messages). However, I am unable to access the web-app through my browser. Neither directly on the raspberry-pi nor on my laptop in the same network. Any idea how to check if the web-service is running? Or to run the web-service manually?

splitti, 2020-01-18:
`sudo service lighttpd status`

Erechtheus, 2020-01-19:
Thanks! This command returns:
`sudo service lighttpd status`
`Unit lighttpd.service could not be found.`

Any idea on how to proceed? Can I manually start the web-service somehow? The logs folder is empty and all other services seem to be running

splitti, 2020-01-19:
That ist the Problem, try a reinstall...

stefantruehl, 2020-01-19:
hi there, I had the same issue yesterday when I was installing master branch on my brand new RPi4 + latest buster image; webserver didn't start and I hat no time to investigate. today I tried the one line installer for the dev-branch and it worked. I still seem to have one issue the rfid-reader not sending the red code to the application but at least the webserver is now configured correctly and running. I will investigate further. 
Hope this helps

maxpousre, 2020-01-19:
I have had problems loading the webapp as well in the past 24 hours. Although the lighttpd.service is active and running. Any other services that I could check?

MiczFlor, 2020-01-19:
@maxpousre 
does it work on the Pi itself, if you open a browser and type `localhost`?

maxpousre, 2020-01-19:
I am running headless on a raspberry pi zero. Do you recommend installing a desktop to verify?

splitti, 2020-01-19:
Try
`wgt URL`

maxpousre, 2020-01-19:
500 response when connecting to the the phoniebox ip address
`HTTP request sent, awaiting response... 500 Internal Server Error
2020-01-19 18:18:36 ERROR 500: Internal Server Error.`

splitti, 2020-01-19:
First: Installation with or without Spotify?

maxpousre, 2020-01-19:
without

splitti, 2020-01-19:
Damn, i have an Idea why. Need more SD cards for Testing. Micz, i will make tomorrow evening a new Pull Request.

Erechtheus, 2020-01-19:
Okay, I just reinstalled the software and it runs now. The only thing I changed was installing the following packages:
`sudo apt-get install at lighttpd`
but actually I don't believe that this solved the issue :)

closed 2020-01-21 by Erechtheus