MiczFlor/RPi-Jukebox-RFID #737: Spotify Premium sponsorship for MiczFlor

domu83, 2020-01-20:
Hi all,

as discussed in another thread, I truly belive that for all Spotify+ Phoniebox users it would be benefical to enable @MiczFlor with Spotify. However, as @MiczFlor is the creator and major contributor of this great project, I don't think he should pay this on his own, as he has no intentions do use Spotify for privat reasons (at least not yet ;)).
Therefor I suggest the community to contribute to a sponsorship for @MiczFlor.
I order to collect the required 99€ for the annual premium account I have created a paypal moneypool. As I never did something similar, this was for me the easyest way to get it done. If there are other suggestions do so please feel free. Here is the moneypool: https://paypal.me/pools/c/8lQ7tMIYad

downside of the moneypool as far as I have seen it, I cannot direct the money directly to @MiczFlor so it goes through my account first. However I hope this is somehow trustful. I have already put 30€ into the bucket to start over.
Looking forward for some contributions.

thanks!!!
Dominik

MiczFlor, 2020-01-21:
Hi @domu83 
thanks for the collection :) I have a paypal account: micz.flor@web.de 
I guess the time is right to do what I don't really believe in: support a streaming service for music which results in huge amounts of cabondioxide due to the costs of hosting and more so: streaming. (only half kidding...). However, I strongly believe that the world is a better place with more Phonieboxes in the making (only half kidding, again, I am stunned by the positive energy I sometimes get based on the families experience in building and using their Phonieboxes).
Via donations, I get some money through "buy me a coffee" and paypal. Thanks to those who have been kind in the past to support my work. It's - obviously - not about hourly rates, it does mean a lot to me, thought (just as [posting designs of Phonieboxes](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/639) ).
See also here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/428

domu83, 2020-02-29:
The Paypal Moneypool has ended on Friday. 18€ have been contributed. @MiczFlor I have just send you a spotify e-voucher of €60.
unfortunately we did not manage to get you the full fund for a yearly premium account

topas-rec, 2021-01-30:
20 € für Spotify Premium sind im Topf für 2021.