MiczFlor/RPi-Jukebox-RFID #739: Touch Sensoren Anschluss und Inbetriebnahme

GMZ87, 2020-01-20:
Hi @ all.

wie schon in meinem alten Thread erwähnt krampfe ich mich durch dieses tolle Projekt. Mitlerweile sieht das alles auch schon sehr gut aus. Nur leider bekomme ich meine Touch Sensoren nicht zum laufen. 

https://www.reichelt.de/entwicklerboards-kapazitiver-beruehrungssensor-debo-touch-p253985.html?PROVID=2788&gclid=EAIaIQobChMIk9OF3d2R5wIVEcDeCh0AWgRzEAkYAiABEgI_N_D_BwE&&r=1
Das sind die Teile. 
Die habe ich mit 3,3v und GND versorgt und den I/O Kanal an einem GPIO Port angeschlossen. 
Danach die Config an den entsprechenden Gpio angepasst und nach neustart getestet. Keine Reaktion.

Nun bei der Bedienung des Sensors wird per LED signalisiert das es einen Kontakt gab. An dem I/O Anschluss liegt dauerhaft die Eingangsspannung von 3,3v an. Sowie der Sensor betätigt wird, fällt die Spannung auf Null. Der Schalter funktioniert also erstmal wie er soll.
Ich denke aber das der Pi eigentlich im Normalzustand auf ein 0 Signal liegt und nur bei Tastendruck die volle Spannung kriegen sollte. Sprich High und Low Signal.
Liege ich da mit der Vermutung richtig? und wenn ja wie kann ich meine Sensoren trotzdem nutzen. Sprich das der Pi nicht auf ein High Signal sondern auf ein Low wartet?

LG Micha

ZyanKLee, 2020-01-20:
schau mal in die gpio-buttons.py - dort kannst du jeden schalter von pulldown auf pullup umstellen: https://github.com/dertobes/RPi-Jukebox-RFID/blob/develop/misc/sampleconfigs/gpio-buttons.py.sample#L52-L58

GMZ87, 2020-01-20:
ja das stand alles auf Pullup, habe jetzt einfach mal PullDown draus gemacht. Bringt aber auch nix.

Dazu ist mir jetzt noch aufgefallen das in der Webapp die GPIO Buttons Deaktiviert sind. Aktivieren lassen sie sich nicht. Der schaltet zwar auf Aktiviert, aber sowie man wieder auf das Einstellungsfenster geht sind sie direkt wieder auf deaktiviert.

#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5102A
# due to the assignment of GPIO 19 and 21 to a buttons

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

def def_recordstart():
    check_call("./scripts/playout_controls.sh -c=recordstart", shell=True)

def def_recordstop():
    check_call("./scripts/playout_controls.sh -c=recordstop", shell=True)

def def_recordplaylatest():
    check_call("./scripts/playout_controls.sh -c=recordplaylatest", shell=True)

#shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
volU = Button(16,pull_down=True,hold_time=0.3)
volD = Button(13,pull_down=True,hold_time=0.3)
next = Button(26,pull_down=True)
prev = Button(6,pull_down=True)
halt = Button(12,pull_down=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
volU.when_held = def_volU
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
volD.when_held = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt
#reco.when_pressed = def_recordstart
#reco.when_released = def_recordstop
#play.when_pressed = def_recordplaylatest

pause()

ZyanKLee, 2020-01-20:
Ist der Button auf gpio26 gesteckt?

GMZ87, 2020-01-20:
Es sind insgesamt 5 Buttons verbaut.
Diese werden über Pin 17 (3,3v) mit Spannung versorgt und an Pin 9 mit GND.
Jeder Button hat ein I/O Anschluss an dem das geschaltete Signal ausgegeben wird. Sprich auf den I/O ´s liegen dauerhaft 3,3v an und sowie der Schalter betätigt wird, wird die Verbindung gekappt und die Spannung fällt an die I/O Anschluss auf null. Die einzelnen I/O Anschlüsse sind an den jeweiligen GPIO´s wie in der Config angeschlossen.

GMZ87, 2020-01-22:
So also die Tasten funktionieren anscheinend jetzt. Hatte nochmals neu installiert und neu konfiguriert. 
Bis auf VolU und VolD reagieren die Tasten und neuerdings läuft jetzt der RFID Reader nicht mehr. Selbes Problem wie bei Maltheluda https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/747

closed 2020-01-24 by GMZ87