MiczFlor/RPi-Jukebox-RFID #741: Gmusic installation on Raspbian Buster

Vanierica, 2020-01-20:
Hi together,

I´m really new into Raspbian etc. 
So maybe I´m just too dumb.

Currently I want to setup the RFID Jukebox with Google Play Music integration.
At first I tried to install it Raspbian Stretch. But with Stretch I´ve the problem that the rfid scanner is crashing every time when I installed Mopidy-Gmusic.

So I decided to try it with Raspbian Buster
I used the following Raspbian Image "2019-09-26-raspbian-buster-full.img"

After first start and configuration of raspbian I installed the RFID Jukebox with the One-Line installer for Buster

After that I installed Mopidy Gmusic with following command and edit the mopidy.conf with my Google Music settings:
sudo pip install mopidy-gmusic

I get no errors or something else, but Gmusic is not working and it seems that mopidy knows nothing about Gmusic. 
Also Mopidy.log file shows nothing about gmusic. 

Do I have to do something else?

MiczFlor, 2020-01-21:
Hi @Vanierica 
thanks for your feedback. Here is the "howto", did you follow these steps?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Enable-Google-Play-Music-GMusic
Would be good to get feedback if and how you made it work - once it does :)

ZyanKLee, 2020-01-21:
I suspect gmusic is being installed on python2 with that command?

ZyanKLee, 2020-01-21:
@vanierica did you use the certain from master or develop branch?

Vanierica, 2020-01-21:
@MiczFlor I followed this "howto". On Raspian Stretch i found out that Mopidy-Gmusic needs minimum pyasn1 Version 0.4.6. so I updated to this Version and Gmusic was functional. But in the Stretch environment my RFID Scanner is not working. I´m using the recommended from Neuftech. I´ve done a plain installation with the One-Line installer with Spotify and the RFID Scanner is not working.

With Buster my RFID Scanner is working but Mopidy / Mopidy-Iris not. Here I´ve also done a plain Installation with the Master (not delvelop) One-Line installer. Nothing else. It seems that the mopidy / mopidy-iris service is not running.

Thanks

Vanierica, 2020-01-21:
This is the output of the On-Line Installer at the end. There are some mopidy errors. 
But first I installed with the master One-Line Installer and then i tried the develop one-liner without flashing everything. This is the output from the second installation.

Created symlink /etc/systemd/system/multi-user.target.wants/phoniebox-rotary-encoder.service → /etc/systemd/system/phoniebox-rotary-encoder.service.
Synchronizing state of mpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable mpd
Failed to enable unit: Unit file mopidy.service does not exist.
Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
  en_GB.UTF-8... done
Generation complete.
mkdir: cannot create directory ‘/home/pi/.config’: File exists
mkdir: cannot create directory ‘/home/pi/.config/mopidy’: File exists
cp: cannot create regular file '/etc/mopidy/mopidy.conf': No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
sed: can't read /etc/mopidy/mopidy.conf: No such file or directory
mpd error: Connection refused
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
rm: cannot remove '/home/pi/RPi-Jukebox-RFID/settings/global.conf': No such file or directory

Vanierica, 2020-01-21:
Update: I get the same error messages at the end of the installation when I use the One-Line Installer on a fresh raspbian buster. This time I used the develop one.

ZyanKLee, 2020-01-22:
Can you please post the complete output (beginning with the start of the installation i.e. after the questions) to paste.ubuntu.com ? Most probably the error is somewhere near the start

Vanierica, 2020-01-22:
The output are both from the develop version

ZyanKLee, 2020-01-22:
I notice

```
The following packages have unmet dependencies: vlc-bin : Depends: libvlc-bin (= 3.0.8-0+deb10u1+rpt1) but 3.0.8-0+deb10u1+rpt7 is to be installed vlc-plugin-skins2 : Depends: vlc-plugin-qt (= 3.0.8-0+deb10u1+rpt7) but 3.0.8-0+deb10u1+rpt1 is to be installed E: Broken packages
```

s-martin, 2020-01-22:
> 
> 
> I notice this error message as the first error in the buster log:
> 
> ```
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  vlc-bin : Depends: libvlc-bin (= 3.0.8-0+deb10u1+rpt1) but 3.0.8-0+deb10u1+rpt7 is to be installed
>  vlc-plugin-skins2 : Depends: vlc-plugin-qt (= 3.0.8-0+deb10u1+rpt7) but 3.0.8-0+deb10u1+rpt1 is to be installed
> E: Broken packages
> ```
> 
> all following errors are most probably caused by that one.
> 
> The command leading to that error is most probably in line 479:
> 
> ```
> sudo apt-get --yes upgrade
> ```
> 
> I don't see where this would be leading to this issue, though. At least the buster minimal image does not include vlc.

OP was using Raspian *FULL*.

Vanierica, 2020-01-24:
Is there something I can do / test?

Vanierica, 2020-01-25:
If I understand it right, Google Music changed the authentication mechanism which is currently implemented in Mopidy 3. Also Mopidy-Gmusic needs phython 3,7 in the newer Version. 

So with the current release its not possible to authenticate with google music if I understand it right.

Are there any plans to use Mopidy 3 and Python 3.7? 
Otherwise I would think about to switch to Spotify.

ZyanKLee, 2020-01-25:
we are currently working on porting the whole project to python3 - which is the source for most of the trouble people are currently having with their phonieboxes.

s-martin, 2020-02-23:
So we are in the process of upgrading to Mopidy 3, see #803.

There's a PR #804, which should be merged soon.
Once this PR is merged you could try the script `buster-install-mopidy3.sh` in the `develop` branch to see if this is now working with gmusic.

s-martin, 2020-02-24:
PR #804 has been merged. Please try Mopidy 3 by using
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

s-martin, 2020-03-09:
So on `develop` Mopidy 3 is now used.

This version can be installed by using:
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh`

@Vanierica, if Gmusic integration is needed, you could open a PR against `develop` with the necessary changes or we try to assist you with that.

s-martin, 2020-04-12:
This should work now on `develop` out of the box, because Mopidy 3 is used there.

@Vanierica could you test it?

closed 2020-04-14 by MiczFlor