MiczFlor/RPi-Jukebox-RFID #745: Question for my understanding: mopidy vs mpd

ZyanKLee, 2020-01-21:
I'm just at the task of cleaning up the mopidy3 install script and stumbled upon the following question:

mpd and mopidy are both effectively mp daemons, right?
They both support the same protocol and clients for controlling?
mopidy has just been chosen because it implements spotify, but mpd does not?

is there a reason not to switch completely to mopidy3 even with not spotify installed?

MiczFlor, 2020-01-22:
Hi @ZyanKLee 
good point. And I don't have the answer. For local files with the `mpd` solution, new files will automatically  be indexed. To this day, I don't have a Spotify integration myself, so I am not sure how `mopidy` does it.
In the Spotify FAQ, this might give you a pointer to potential issues?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ
MP3 (local music) handling has completely changed for the +Spotify Edition. The tracks need to be indexed by Mopidy & Mopidy-Spotify (this is the part of Phoniebox, which gives you spotify support) and the created M3U files do have another structure than normal M3U files.
You have to scan you library once (if you hadn't yet) and everytime you upload new local tracks to your Phoniebox.
You can scan local files under "Folders & Files" (Top Navigation).
For the future we try to integrate an automation for that.

ZyanKLee, 2020-02-02:
/close

closed 2020-02-02 by ZyanKLee