MiczFlor/RPi-Jukebox-RFID #747: RFID Reader funktioniert nicht Neuftech EM4100

maltheluda, 2020-01-22:
Hallo zusammen,
ein wahnsinnig gutes Projekt wurde hier auf die Beine gestellt. Dafür erst mal Hut ab! Richtig gut!
Leider habe ich ein Problem:

Trotz mehrmaliger Neuinstallation per One-Line-Installation (egal ob mit Spotify oder ohne) wird mein RFID Reader von Neuftech EM4100 nicht erkannt. Bzw. das System erkennt ihn. In der WebApp steht der RFID Reader allerdings als "disabled" und er wird bei klick auf "Switch ON" auch nicht aktivert.

Habe schon einiges Ausprobiert, was hier im Forum steht. Auch der MDP Server läuft nicht. Ich kann also auch via WebApp keine Dateien abspielen.

Mein Setup:
Raspberry 3 B+
Neuftech EM4100
Raspbian GNU/Linux 9.3 (stretch)

Ich freue mich über jede Hilfe!

[image]
[image]

ZyanKLee, 2020-01-22:
I had issues with my rfid reader, too. What worked, though, was using the `Reader.py.experimental` instead:

```
cd /home/pi/RPi-Jukebox-RFID/scripts/
mv Reader.py Reader.py.original
mv Reader.py.experimental Reader.py
```

Give it a test-run - if it won't work either, please report back.

---

Ich hatte auch Probleme mit meinem RFID Reader. Die Lösung war bei mir, die `Reader.py.experimental` zu nutzen:

```
cd /home/pi/RPi-Jukebox-RFID/scripts/
mv Reader.py Reader.py.original
mv Reader.py.experimental Reader.py
```

Probier es mal aus. Wenn es nicht funktioniert gib wieder Bescheid.

maltheluda, 2020-01-22:
Danke @ZyanKLee für die schnelle Antwort!
Allerdings hat sich weder an der Fehlermeldung noch an dem Problem etwas geändert.

Im Web Interface kann ich nach wie vor den RFID Reader nicht von disabled auf enabled setzen.

---

Thanks for your quick response!
But it did not work.

It is still the same problem including the same error messages. On the web interfaces it‘s still not possible to switch the RFID Reader from disabled to enabled.

ZyanKLee, 2020-01-22:
did you reboot? forgot to mention that step

maltheluda, 2020-01-22:
yes I did. No changes

Vanierica, 2020-01-22:
I've the same Problem. On my first install on sunday last Week everything was fine with stretch Master Brand. Now everytime the rfid Reader is disabled.

s-martin, 2020-01-22:
Can you post the result of `sudo pip list` or `sudo pip3 list`?

I suspect that currently the installscript doesn't install evdev for Python 3.

maltheluda, 2020-01-22:
result of `sudo pip list`

`automationhat (0.0.4)
blinker (1.3)
blinkt (0.1.1)
buttonshim (0.0.2)
Cap1xxx (0.1.3)
chardet (2.3.0)
click (6.6)
colorama (0.3.7)
colorzero (1.1)
cryptography (1.7.1)
dnspython (1.15.0)
drumhat (0.1.0)
enum34 (1.1.6)
envirophat (0.0.6)
evdev (0.7.0)
ExplorerHAT (0.4.2)
Flask (0.12.1)
fourletterphat (0.1.0)
gpiozero (1.5.0)
idna (2.2)
ipaddress (1.0.17)
itsdangerous (0.24)
Jinja2 (2.8)
keyring (10.1)
keyrings.alt (1.3)
lxkeymap (0.1)
MarkupSafe (0.23)
mcpi (0.1.1)
microdotphat (0.1.3)
mote (0.0.3)
motephat (0.0.2)
mutagen (1.36)
numpy (1.12.1)
oauthlib (2.0.1)
pantilthat (0.0.5)
phatbeat (0.1.0)
pi-rc522 (2.2.1)
pianohat (0.1.0)
picamera (1.13)
picraft (1.0)
piglow (1.2.4)
pigpio (1.38)
Pillow (4.0.0)
pip (9.0.1)
pyasn1 (0.1.9)
pycrypto (2.6.1)
pygame (1.9.3)
pygobject (3.22.0)
pyinotify (0.9.6)
PyJWT (1.4.2)
pyOpenSSL (16.2.0)
pyserial (3.2.1)
pyxdg (0.25)
rainbowhat (0.0.2)
requests (2.12.4)
requests-oauthlib (0.7.0)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
scrollphat (0.0.7)
scrollphathd (1.2.0)
SecretStorage (2.3.1)
sense-emu (1.0)
sense-hat (2.2.0)
setuptools (33.1.1)
simplejson (3.10.0)
six (1.10.0)
skywriter (0.0.7)
sn3218 (1.2.7)
SPI-Py (1.0)
spidev (3.3)
touchphat (0.0.1)
twython (3.4.0)
unicornhathd (0.0.3)
urllib3 (1.19.1)
Werkzeug (0.11.15)
wheel (0.29.0)`

result of `sudo pip3 list`
`automationhat (0.0.4)
blinker (1.3)
blinkt (0.1.1)
buttonshim (0.0.2)
Cap1xxx (0.1.3)
chardet (2.3.0)
click (6.6)
colorama (0.3.7)
colorzero (1.1)
cryptography (1.7.1)
drumhat (0.1.0)
envirophat (0.0.6)
ExplorerHAT (0.4.2)
Flask (0.12.1)
fourletterphat (0.1.0)
gpiozero (1.5.0)
idna (2.2)
itsdangerous (0.24)
jedi (0.10.2)
Jinja2 (2.8)
keyring (10.1)
keyrings.alt (1.3)
MarkupSafe (0.23)
mcpi (0.1.1)
microdotphat (0.1.3)
mote (0.0.3)
motephat (0.0.2)
numpy (1.12.1)
oauthlib (2.0.1)
pantilthat (0.0.5)
pgzero (1.2)
phatbeat (0.1.0)
pianohat (0.1.0)
picamera (1.13)
picraft (1.0)
piglow (1.2.4)
pigpio (1.38)
Pillow (4.0.0)
pip (9.0.1)
pyasn1 (0.1.9)
pycrypto (2.6.1)
pygame (1.9.3)
pygobject (3.22.0)
pyinotify (0.9.6)
PyJWT (1.4.2)
pyOpenSSL (16.2.0)
pyserial (3.2.1)
python-apt (1.1.0b5)
pyxdg (0.25)
rainbowhat (0.0.2)
requests (2.12.4)
requests-oauthlib (0.7.0)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
scrollphat (0.0.7)
scrollphathd (1.2.0)
SecretStorage (2.3.1)
sense-emu (1.0)
sense-hat (2.2.0)
setuptools (33.1.1)
simplejson (3.10.0)
six (1.10.0)
skywriter (0.0.7)
sn3218 (1.2.7)
spidev (3.3)
thonny (2.1.16)
touchphat (0.0.1)
twython (3.4.0)
unicornhathd (0.0.3)
urllib3 (1.19.1)
Werkzeug (0.11.15)
wheel (0.29.0)`

you are right, @s-martin  there is no evedev for Python 3

s-martin, 2020-01-22:
Please execute `sudo pip install "evdev == 0.7.0"` and try again.

s-martin, 2020-01-22:
And also
```
sudo pip3 install git+git://github.com/lthiery/SPI-Py.git#egg=spi-py
sudo pip3 install pi-rc522

```

maltheluda, 2020-01-22:
I did so. No changes on that problem.

s-martin, 2020-01-22:
You still have the traceback error from `daemon_rfid_reader.py`?

maltheluda, 2020-01-22:
`systemctl status `phoniebox-rfid-reader.service`

give this as a feedback:
`● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)`
No traceback error anymore, right? ;-)

s-martin, 2020-01-22:
Start the service with `systemctl start phoniebox-rfid-reader.service`

maltheluda, 2020-01-22:
After authentication I got this:

`Failed to start phoniebox-rfid-reader.service: Unit phoniebox-rfid-reader.service is not loaded properly: Invalid argument.
See system logs and 'systemctl status phoniebox-rfid-reader.service' for details`

and the result of `systemctl status phoniebox-rfid-reader.service` is again this:
`● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)`

maltheluda, 2020-01-22:
So, after I did all these tries with @s-martin (thanks a lot dude) I did a reboot.
This is the result:

`Failed to start phoniebox-rfid-reader.service: Unit phoniebox-rfid-reader.service is not loaded properly: Invalid argument.
See system logs and 'systemctl status phoniebox-rfid-reader.service' for details`

These are the known details:
`● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)`

s-martin, 2020-01-22:
Did you try

```
cd /home/pi/RPi-Jukebox-RFID/scripts/
python3 RegisterDevice.py
```

according to [configure](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#register-your-usb-device-for-the-phoniebox)?

maltheluda, 2020-01-22:
yes, I did. and `HXGCoLtd` is registered.

And the device works on OS. It beeps when it turns on, beeps when swiping card over it and writes the ID of the card into a textfile after swiping.

But RFID Reader is still disabled on webapp and does not work
[image]

ZyanKLee, 2020-01-22:
can you please paste the content of `/etc/systemd/system/phoniebox-rfid-reader.service`

maltheluda, 2020-01-22:
the content of `/etc/systemd/system/phoniebox-rfid-reader.service`:

`[Unit]  
Description=Phoniebox RFID-Reader Service
After=network.target iptables.service firewalld.service

[Service]
User=pi
Group=pi
Restart=always
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

[Install]
WantedBy=multi-user.target`

result of:
`cd /home/pi/RPi-Jukebox-RFID && sudo -u pi -H python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py`

is just:
`/home/pi/RPi-Jukebox-RFID/scripts`

for minutes ... if I am pressing strg+c to startover, the result is:
`^CTraceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 29, in <module>
    cardid = reader.reader.readCard()
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 57, in readCard
    r, w, x = select([self.dev], [], [])
KeyboardInterrupt`

how does it sound for you?

ZyanKLee, 2020-01-22:
sounds great. Now everything works as it should?

maltheluda, 2020-01-22:
nope. sadly not.
MPD Server is working well.

But RFID does not work and shown as disabled either. I have took a photo of the 2 lines of code which are shown while booting:

`2.371148]` system[1]: [/etc/systemd/system/phoniebox-rfud-reader.service:10] Executable path is not absolute, ignoring: python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
2.389123 system[1]: phoniebox-rifd.reader.service: Service lacks both ExecStart= and ExecStop= setting. Refusing.`

still status of `systemctl status phoniebox-rfid-reader.service:`
`● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)`

s-martin, 2020-01-23:
The issue is probably the ExecStart command `ExecStart=python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py` (from develop).

MiczFlor, 2020-01-23:
If you want to try this with python2 - and let us know, if it makes a difference, you don't need to reinstall. You need to change two lines in two scripts. the one line that @s-martin posted above 

The line in the file `/etc/systemd/system/phoniebox-rfid-reader.service` the line from
~~~
ExecStart=python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
~~~
to 
~~~
ExecStart=python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
~~~

and the first line in the file `/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py`
from 
~~~
#!/usr/bin/env python3
~~~
to
~~~
#!/usr/bin/env python2
~~~

maltheluda, 2020-01-23:
thanks @s-martin and @MiczFlor for your answers.
I just edited the files as you said, but no improvement anyway.

You guys think it is more a software issue than a hardware issue? Because I just bought 3 different RFID Reader models to test them.

s-martin, 2020-01-23:
I suspect it's a software issue, because Python version was recently harmonized and I didn't really test it on stretch.

maltheluda, 2020-01-23:
I just crashed the whoole system. I flashed the SD card again. 
Let‘s start at zero. - I will use the one-line-install to install phoniebox again and will report

maltheluda, 2020-01-23:
After installing stretch and phoniebox via one-line again I had exactly the same iusses.
Now I  am installing buster to check if this could work.

Other idea: Could upload somebody a kind of mirrored-stretch OS with enabled rfid in webappliation to wetransfer or something? may this is the easiest way? 🤔

maltheluda, 2020-01-23:
okay. I think this will be a neverending story. on `buster` is so much trouble while installation reg. RFID. Sorry for german code, I switched on the german version.

`If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] y
./buster-install-default.sh: Zeile 865: cd: /home/pi/RPi-Jukebox-RFID/scripts/: Datei oder Verzeichnis nicht gefunden
python2: can't open file 'RegisterDevice.py': [Errno 2] No such file or directory
chown: Zugriff auf '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt' nicht möglich: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt' nicht möglich: Datei oder Verzeichnis nicht gefunden`

s-martin, 2020-01-24:
I should have a new Pi to test this weekend, so I hope I can investigate a little more.

SwipeCircus, 2020-01-24:
I am freaking out, guys! IT WORKS!

What have I done?
I just replaced the `rfid-reader.service` with this code ([from your version 1.1.5](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/Version-1.1.5-dynamicRFIDreader/misc/rfid-reader.service))
`[Unit]  
Description=RFID-Reader Service
After=network.target iptables.service firewalld.service mpd.service

[Service]
User=pi
Group=pi
Restart=always
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

[Install]
WantedBy=multi-user.target`

maltheluda, 2020-01-24:
I am freaking out, guys! IT WORKS!

What have I done?
I just replaced the `rfid-reader.service` with this code ([from your version 1.1.5](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/Version-1.1.5-dynamicRFIDreader/misc/rfid-reader.service))

```
[Unit]  
Description=RFID-Reader Service
After=network.target iptables.service firewalld.service mpd.service

[Service]
User=pi
Group=pi
Restart=always
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

[Install]
WantedBy=multi-user.target
```

Vanierica, 2020-01-24:
Can confirm. Its Working!

GMZ87, 2020-01-24:
Bestätige, bei mir auch ;)

maltheluda, 2020-01-24:
Is everything working fine @Vanierica and @GMZ87 ? Because I do have some problems with the MPD / MPC but I think thats another topic :)

GMZ87, 2020-01-24:
Bei mir läuft der Ton nicht. Der Startsound ertönt beim Hochfahren und danach ist ruhe. Songs werden laut der Webapp abgespielt aber kein laut zu hören. Auch per Speakertest kommt nun kein Ton mehr. Hat ja aber nichts mit dem RFID zu tun. Irgendwas läuft zurzeit in den Installationen verkehrt. Fehler ist bei Buster sowie Stretch.

Vanierica, 2020-01-24:
@GMZ87 Blöde Frage, aber hast du am Raspberry die Tonausgabe von HDMI auf Analog gestellt? Bei mir funktioniert der Ton einwandfrei.

@for me everything is working except Google Music. I´m struggling with google authentication. But thats another topic

Diablo-Rojo, 2020-01-25:
@maltheluda Good morning. I have the same problem with my current installation (the previous three are working fine). Would you mind to explain where the changes you mentioned need to be done? (I am actually a prototype-noob to all of this)

GMZ87, 2020-01-25:
@Vanierica 
Neu Aufgesetzte SD Karte (Raspi 2 mit Hifiberry Miniamp) -> 
Config.txt angepasst -> 
asound.conf angepasst -> 
aplay -l zeigt Gerät richtig an -> 
Speakertest läuft und gibt sound aus ->
Installation RPI-Jukebox-RFID (Buster oder Stretch) erfolgreich ->
Jukebox startet ->
Startsound ertönt ->
Webapp ist online ->
Musik kann gestartet werden, aber kein Ton zu hören (Lautstärke auf 80%) ->
Speakertest läuft, aber ebenfalls kein Ton mehr zu hören.

Mehr mache ich hier nicht. Sowie die Jukebox installiert ist gibt es kein ton mehr aus der Box.
Sorry für Offtopic und vielen Dank für eure Hilfe

maltheluda, 2020-01-26:
@Diablo-Rojo go to ‘RPi-Jukebox-RFID/misc/sampleconfigs/‘ , open the file ‘phoniebox-rfid-reader.service‘ and edit the content to:

`[Unit]  
Description=RFID-Reader Service
After=network.target iptables.service firewalld.service mpd.service

[Service]
User=pi
Group=pi
Restart=always
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

[Install]
WantedBy=multi-user.target`

save it. and typ this command into the terminal:

`sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-rfid-reader.service.stretch-default.sample /etc/systemd/system/phoniebox-rfid-reader.service`

After that. Reboot your system.

What the content change is doing: 
The rfid-reader will be running on python2 not python3. 
This small changes has worked for my issue perfectly. 

BTW: I am actually on the same noob-level as you are ;-)

Diablo-Rojo, 2020-01-26:
@maltheluda Worked perfectly. Not all heroes wear capes. ;-) Thank  you very much.

Moppel2k, 2020-01-26:
Bei mir klappt es leider nicht :-(
ich nutze einen RC522(den ich mit acht adern angeschlossen habe) der auch mal funktioniert hat nur leider musste ich die box neu aufsetzen!
Bei der Installation! bin ich nach den wie Hier https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special beschrieben vorgegangen! Zusätzlich hab ich das oben beschriebene Verfahren durchlaufen aber ich kann den RFID reader nicht im Einstellungsfenster in der Weboberfläche aktivieren, nach dem neuladen der seite ist er wieder Disabled!

ich bin langsam am vereinzeln ;-(

Chairon22, 2020-01-26:
Super, hat auch bei mir funktioniert👍👍👍. Ich war schon nahe der Verzweiflung

rubynubee, 2020-01-27:
@maltheluda 
Thanks! Even I run the box on raspbian strech, replacing rfid-reader.service with the code from version 1.1.5 was the solution.

@MiczFlor 
I wonder what changed, last time I set up the box with the same hardware, it worked flawless from the beginning. Except the GIPIO Buttons, but that seems to be another problem – I didn't manage, till now, but maybe a hardware issue. 
Anyway, the kids love the box, and they don't mind using RFID Chips to control it. So many thanks for all your work and effort you put in this project.
Cheers!

MiczFlor, 2020-01-27:
Hi @rubynubee 
it's been a long chain of events. Mopidy for Spotify integration changed. That triggered a general eneregy in the community to move everything to Python3. And that broke a few things. And they haven't been fully fixed.
@s-martin @ZyanKLee do you think I should merge `develop` to `master` at this stage? I apologise for my absence in proper testing and having opinions and walking through new code line by line. It's been a busy few weeks since new year and I haven't found a weekend to dedicate to Phoniebox yet.

ZyanKLee, 2020-01-27:
Hey @MiczFlor, there is no need to apologize, this is a hobby for all of us after all. So RL is always more important.
Having opinions and checking the coffee thoroughly also is not a crime but badly needed. Please challenge my ideas and suggestions so we can learn from each other.

Regarding the better from develop to master: not yet.
I want to get it fixed properly to finally have a stable master again.

hoodvisions, 2020-01-28:
> @Diablo-Rojo go to `RPi-Jukebox-RFID/misc/sampleconfigs/` , open the file `phoniebox-rfid-reader.service` and edit the content to:
>
> ```
> [Unit]
> Description=RFID-Reader Service
> After=network.target [...]

I tried that but unfortunately it didn't help. I have the same RFID Reader, using Stretch and due to being a complete noob I used the one line install script. I built the box 2 years ago and it worked. This time it doesn't.

The code you posted is definitely in /etc/systemd/system/phoniebox-rfid-reader.service
It doesn't have to be added to any other files as well, right?

When I type sudo systemctl status phoniebox-rfid-reader  I get 

`Jan 28 13:15:13 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Jan 28 13:15:13 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Jan 28 13:15:13 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Jan 28 13:15:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service hold-off time over, scheduling restart.
Jan 28 13:15:14 raspberrypi systemd[1]: Stopped RFID-Reader Service.
Jan 28 13:15:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Jan 28 13:15:14 raspberrypi systemd[1]: Failed to start RFID-Reader Service.
Jan 28 13:15:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Jan 28 13:15:14 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.`

I started from scratch, this time w/o the one line install script but I end up with the same results. Should I just buy another RFID Reader to fix this quickly? I wanted to build this for my daugther with Spotify integrated and I wouldn't mind to go the quick route via amazon. 
Thanks in advance for any help!

s-martin, 2020-01-28:
Did you try `sudo systemctl stop phoniebox-rfid-reader` and `sudo systemctl start phoniebox-rfid-reader`?

What says `sudo systemctl status phoniebox-rfid-reader` after the restart of the service?

hoodvisions, 2020-01-28:
Unfortunately it remains the same after a reboot and restart of the service. But it is pretty likely my RFID Reader, right? Maybe if I just get the EXACT same model as the people who got this working here?

erbenheim, 2020-01-29:
Did you change something on the on line installer since sunday? My Installation and the change of the phoniebox-rfid-reader.service worked on sunday and my new installation today has the Same Problem like hoodvisions.

s-martin, 2020-01-30:
I have found a tool, which is supposed to record hardware events and later replay them for integration tests, see https://github.com/martinpitt/umockdev

This could be helpful to create tests, which could be used to improve changes.

s-martin, 2020-01-31:
@MiczFlor do you think it would make sense to revert the download link on the start page to an older version, so people could use Phoniebox for the moment until we get this issue fixed? And maybe tell people to not install Spotify edition for the moment, because of the mopidity package issue?

I used rc5 as Classic before Christmas which worked like a charm.

Or does anybody has a fix already?

Alwiz79, 2020-01-31:
@s-martin could give me a hint about how to install "rc5"?
I urgently need to get ready with another box on this weekend and cannot wait for a fix...an older classic version would be enough for me :-)

Alwiz79, 2020-01-31:
can anyone give a quick hint on to install an older version of the phoniebox?
I tried to edit the one line installation script like this
`cd; rm stretch-install-*; wget https://github.com/MiczFlor/RPi-Jukebox-RFID**/blob/v1.1.7**/scripts/installscripts/stretch-install-default.sh; chmod +x stretch-install-default.sh; ./stretch-install-default.sh`

unfortunately this did not work :-(

githubtee, 2020-01-31:
hi all,
please see the working solution from me in #761 
this solution might be temporary until developer find the real fix for this problem...

Alwiz79, 2020-01-31:
@githubtee I don't know how follow your hint :-(
the link in #761 is not working

githubtee, 2020-01-31:
sorry, there was an mistake in the link. i have edited and fixed it

Alwiz79, 2020-01-31:
@githubtee still no success here.,.i made a clean one-line-stretch installation.
Than simply exchanged the `daemon_rfid_reader.py` from 1.1.8 from your link.
still not working, RFID Reader is displayed as disabled, and cannot be activated

s-martin, 2020-01-31:
What happens when you execute `/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py`?

Alwiz79, 2020-01-31:
this comes as an output, and then nothing more
`/home/pi/RPi-Jukebox-RFID/scripts`

s-martin, 2020-01-31:
Sorry this should be one line (or one command):

`/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py`

[45 more replies]

closed 2020-02-04 by MiczFlor