MiczFlor/RPi-Jukebox-RFID #749: Check, if correct Python packages are installed for Python 3

s-martin, 2020-01-22:
Error in #747  could be caused by missing python packages, because the script doesn't install evdev (and maybe others) for Python 3.

I don't know, if this snippet in `buster-install-default.sh`
```
# use python3.7 as default
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 1
```

really is sufficient for `pip` and maybe other tools.

s-martin, 2020-01-22:
Unfortunately I don't have a Pi available at this time to test it, new Pi should arrive on friday.

maltheluda, 2020-01-22:
`update-alternatives: error: alternative path /usr/bin/python3.7 doesn't exist`

s-martin, 2020-01-22:
Please check in /usr/bin which python executable exist. I suspect `python3.7.1`

maltheluda, 2020-01-22:
sorry, I am not that deep intp raspberry. in `/usr/bin` is just one folder, a alias of `X11`

s-martin, 2020-01-22:
Please post the result of `ls -ls /usr/bin/python*`

maltheluda, 2020-01-22:
`  0 lrwxrwxrwx 1 root root       9 Jan 24  2017 /usr/bin/python -> python2.7
   0 lrwxrwxrwx 1 root root       9 Jan 24  2017 /usr/bin/python2 -> python2.7
3092 -rwxr-xr-x 1 root root 3162224 Nov 24  2017 /usr/bin/python2.7
   0 lrwxrwxrwx 1 root root      36 Nov 24  2017 /usr/bin/python2.7-config -> arm-linux-gnueabihf-python2.7-config
   0 lrwxrwxrwx 1 root root      16 Jan 24  2017 /usr/bin/python2-config -> python2.7-config
   0 lrwxrwxrwx 1 root root       9 Jan 20  2017 /usr/bin/python3 -> python3.5
3884 -rwxr-xr-x 2 root root 3976264 Jan 19  2017 /usr/bin/python3.5
   0 lrwxrwxrwx 1 root root      36 Jan 19  2017 /usr/bin/python3.5-config -> arm-linux-gnueabihf-python3.5-config
3884 -rwxr-xr-x 2 root root 3976264 Jan 19  2017 /usr/bin/python3.5m
   0 lrwxrwxrwx 1 root root      37 Jan 19  2017 /usr/bin/python3.5m-config -> arm-linux-gnueabihf-python3.5m-config
   0 lrwxrwxrwx 1 root root      16 Jan 20  2017 /usr/bin/python3-config -> python3.5-config
   0 lrwxrwxrwx 1 root root      10 Jan 20  2017 /usr/bin/python3m -> python3.5m
   0 lrwxrwxrwx 1 root root      17 Jan 20  2017 /usr/bin/python3m-config -> python3.5m-config
   0 lrwxrwxrwx 1 root root      16 Jan 24  2017 /usr/bin/python-config -> python2.7-config`

s-martin, 2020-01-22:
I don't understand why you don't have Python 3.7 installed, it's the default package on buster for python 3.

Please try
```
sudo apt-get --yes --allow-downgrades --allow-remove-essential --allow-change-held-packages install apt-transport-https samba samba-common-bin python3 python-dev python-pip gcc lighttpd php7.3-common php7.3-cgi php7.3 php7.3-fpm at mpd mpc mpg123 git ffmpeg python-mutagen python3-gpiozero resolvconf spi-tools python-spidev python3-spidev
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 1

```

and check the python version again.

ZyanKLee, 2020-01-22:
@s-martin if I remember correctly @maltheluda used the stretch variant

maltheluda, 2020-01-22:
I installed the linked version of raspberry. its stretch as @ZyanKLee mentioned.

after running your commands, this is the result:
`update-alternatives: error: alternative path /usr/bin/python3.7 doesn't exist`

seams there is no python 3.7 anywhere ... 🤔

s-martin, 2020-01-22:
> 
> 
> @s-martin if I remember correctly @maltheluda used the stretch variant

Ahh.

Then everything I tried was more or less useless.

ZyanKLee, 2020-01-22:
not really - we were made aware that we did on one hand port some of the scripts to python3, but never took care that the python3 on stretch is actually really catered for in the installer scripts :-D

s-martin, 2020-01-22:
So you could try `sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.5 1`

maltheluda, 2020-01-22:
`update-alternatives: using /usr/bin/python3.5 to provide /usr/bin/python (python) in auto mode`

s-martin, 2020-01-22:
Now you could try #747 again :-)

s-martin, 2020-02-01:
#750 and #751 fixed this and added python packages for 2 and 3

closed 2020-02-01 by s-martin