MiczFlor/RPi-Jukebox-RFID #753: Ansible based installer

tcse10, 2020-01-25:
Hi,
I have noticed, there are many issues with the installation scripts. I bereave using can deliver a better user experience for differing platforms.  
I can start a ansible based installer, but I cannot maintain and support it on my own. Are there folks out there experienced with ansible and willing to contribute?

ZyanKLee, 2020-01-25:
hey @tcse10, I had the same thought some days ago, but ultimately the trouble is not with the way of installing dependencies, but the versions of those dependencies and their (in-)compatibilities between several of those versions.
In the end it is @MiczFlor's project and he should make such fundamental decisions.

MoJo2600, 2020-06-19:
Wouldn't it be easier to create ready to use images that everybody can download and directly flash on the SD-Card? The setup routine would have to be changed. I'm thinking about a worklfow like this:

- Flash SD-Card and boot
- Phoniebox creates a Wifi Hotspot
- When you connect to the hotspot, you're redirected to a configuration page where you can set the Wifi credentials and set up your reader, etc.

This would lower the barrier for people not used to the Linux shell.