MiczFlor/RPi-Jukebox-RFID #757: Create cover.jpg if image is in mp3 file

thedipling, 2020-01-29:
## Feature Description

with FFMPEG the system creates a cover.jpg out of a mp3-file.

in general "ffmpeg -i file.mp3 file.jpg" would do the trick

### What functionality would you like to see in your phoniebox?

to activate this automatically or manually

### How do you envision the feature to work from a users perspective?

Show an option to create the cover.jpgs

### Further information that might help

N/A

ZyanKLee, 2020-01-30:
If bulk option is implemented (i.e. for a folder and all its subfolders) user should be asked if already existing cover.jpg's should be overwritten, renamed or skipped.

s-martin, 2023-12-28:
Implemented in future3/develop for **version 3.x** since PR #2177 was merged.

This will not be implemented in version 2.x anymore.

closed 2023-12-28 by s-martin