MiczFlor/RPi-Jukebox-RFID #758: Option for GPIO for RPi 1/2/.../zero

thedipling, 2020-01-29:
## Feature Description

Since I used a RPi 1B for the installation of Phoniebox an I had a lot of trouble to get the gpio-service getting up and running. I would request an option to set GPIOs depending on the hardware and/or needs

### What functionality would you like to see in your phoniebox?
1. During the installation you can choose your raspberry and the appropriate GPIOs will be activated
2. In the WebUI you can see the GPIOs active
3. There is an option in the WebUI to change and/or activate GPIOs an users demands

### How do you envision the feature to work from a users perspective?

a chapter in the Options

### Further information that might help

N/A

s-martin, 2020-01-30:
I think providing GPIO settings in the web UI could be a good idea. Should be in some kind of advanced settings section.

For the moment you can edit the file `/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py` to adjust the GPIO settings to your device. 

For more info see also here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons
But be aware that you don't need any pullup resistors anymore. @MiczFlor I think we should update that page.

Kyokuhin-b, 2021-01-09:
Hi, i can't find the gpio-buttons.py

Where is placed in the last update? I can't find it on "code" also.

greetings

Frederic

thedipling, 2021-01-09:
From which version did you do your update?
as far as I see the 2.2 changed a lot - so you will find that settings in 

/settings/gpio_settings.ini

and in /scripts/playout_controls.sh 

you will find the script itself.