MiczFlor/RPi-Jukebox-RFID #759: Clean reinstallation or uninstall

dhanessar, 2020-01-29:
Hallo,
Ich hatte die Jukebox installiert und sie hat wundebar funktioniert.
Meine Tochter hat sie gerne benutzt.
Wirklich ein tolles Projekt!

Dann habe ich mich leider mit dem Spotify Konto auf einem Gerät angemeldet und musste das Passwort ändern, da ich es vergessen hatte.
Da ich nicht geschafft habe, das Passwort auf der Jukebox zu ändern habe ich die Installation noch einmal durchgeführt. Nun funktioniert leider einiges nicht mehr.
Der RFID-Reader wird am Ende der Installation angezeigt und ich wähle ihn aus. Trotzdem kann ich in den Einstellungen der Weboberfläche nicht aktivieren. 
Außerdem funktioniert die Verbindung zu Spotify nicht mehr.

Um die SD-Karte rauszuholen, um das auch Raspian Buster neu aufzusetzen müsste ich die ganze Box auseinanderreißen. Deswegen würde ich gerne RPi-Jukebox-RFID komplett deinstallieren und neu installieren oder irgendwie anders eine saubere Neuinstallation durchführen.
Wie kann ich dieses bewerkstelligen?

Beim Installationsvorgang erscheinen am Ende einige Meldunge, dass manche Vorgänge nicht durchgeführt werden konnten (siehe [Installationlog.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4131006/Installationlog.txt)). Deswegen denke ich, dass ich die vorige Version zuerst löschen muss, um eine frische Installation durchzuführen.

Gruß, Daniel

s-martin, 2020-01-31:
Das Logfile sieht eigentlich OK aus.
Die Installation des RFID Readers ist leider derzeit fehlerhaft und noch nicht gefixt.

Das wird derzeit in #747 abgehandelt, leider gibt es noch keinen Fix. 

Ich würde die Box nicht auseinander reißen, sondern noch abwarten, bis wieder eine funktionsfähiger Installer da ist (siehe #747).

s-martin, 2020-02-01:
Der one line installer für stable sollte jetzt wieder funktionieren.

dhanessar, 2020-02-11:
Okay, vielen Dank für den Hinweis!
Dann liegt gar nicht an meinem System. Dann werde ich noch ein wenig warten.

dhanessar, 2020-02-11:
Okay, danke, der RFIS-Reader funtioniert nun wieder.
Aber die Phoniebox spielt keinen Sound weder Startup, noch lokale mp3s noch spotify.
Nach der Installation ist die Lautstärkeanzeige im Webinterface ausgegraut und lässt sich nich einstellen.
Es sieht so aus, als wenn der mpd und der mopidy service nicht laufen.
Nach dem ich mpd mit
`systemctl enable mpd
systemctl start mpd`
starte, kann ich den Lautstärkewert verändern, aber totzdem wird nichts abgespielt.
Mopidy kann ich maNuell nur mit 'mopidy' starten aber nicht als service.
Sieht nach ähnlichen Schwierigkeit wie in #783 aus. Ich werde das im Auge behalten. Ich bin aber für jeden Hinweis dankbar
Raspberry Pi: 3b
Raspian: Buster
Phoniebox Version: 2.0 rc7

Okay, thank you. The RFID reader ist working again but the phonibox doesn't play any sound, not startup sound nor local mp3s nor spotify.
After the one line installation the volume gauge in the web interface is greyed out and can't be adjusted.
Looks like the mpd and mopidy services are not running.
After starting mpd via
`systemctl enable mpd
systemctl start mpd`
I can adjust the volume value. But still there isn#t any sound.
I can start mopidy manually but as service.
Looks like similar problems lik in issue #783. I will watch it closely but an ideas are welcome!
[image]
[image]

s-martin, 2020-02-12:
Sounds like this is the same  issue like #783.

I created a PR #789, which maybe fixes this.

You don’t need to enable mpd, because in Spotify edition mopidy is used.

s-martin, 2020-02-17:
We have merged PR #789 into `master`.

Could you check, if you do a new installation, if this fixes your issue?

**ATTENTION**: Removing the directory `/home/pi/RPi-jukebox-RFID` deletes **all** your settings and also your **music files**!

s-martin, 2020-02-19:
It seems that mopidy-iris dropped support for python 2 and mopidy2 after version 3.43.0. PR #793 was merged to `master` and should fix this.

Please test.

dhanessar, 2020-02-21:
Thank you, for the notification.
I did test and tried a new installation.
At least the volume gauge is working again. A volume level is displayed and I can change it.
But there is still no sound, neither start up sound nor local files. When I click on the spotify menu the spotify menu does not apear. 404: Not Found is displayed. Before the menu apeared but couldn't connect to Spotify. Hence, I cann't play spotify songs either.

You can see the results of tail -n 500 /var/log/syslog: and sudo systemctl status mopidy here:
[https://pastebin.com/LkxRc6NQ](url)

s-martin, 2020-02-22:
In your log there seems to be a problem with mopidy iris:
`Feb 21 23:10:49 phoniebox mopidy[3855]: ERROR    [MainThread] mopidy.ext Failed to load extension iris: No module named 'handlers`

Could be a problem with the iris version 3.43 we use: https://github.com/jaedb/Iris/issues/474

We have to check, if we need a newer iris version, but then we would need python 3, which is currently in progress #796

s-martin, 2020-02-24:
PR #804 has been merged. Please try Mopidy 3 by using
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

dhanessar, 2020-02-25:
Now, the start up sound an local files are played!
Thank you!
But I still cannot to connect to the spotify interface. Clicking on the spotify menu leads to "404; Not Found" with the browser address http://phoniebox:6680/iris/

s-martin, 2020-02-26:
Can you post the results of ` tail -n 500 /var/log/syslog`?

dhanessar, 2020-02-26:
Yes I can. The result of 
`tail -n 500 /var/log/syslog`
is

```
Feb 26 21:51:31 phoniebox rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="327" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Feb 26 21:51:31 phoniebox systemd[1]: logrotate.service: Succeeded.
Feb 26 21:51:31 phoniebox systemd[1]: Started Rotate log files.
Feb 26 21:51:31 phoniebox mopidy[463]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Feb 26 21:51:31 phoniebox systemd[1]: phpsessionclean.service: Succeeded.
Feb 26 21:51:31 phoniebox systemd[1]: Started Clean php session files.
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 4 Spotify playlists
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 26 21:51:32 phoniebox mopidy[463]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 26 21:51:32 phoniebox lighttpd[848]: echo: write error: Broken pipe
Feb 26 21:51:36 phoniebox systemd[1]: man-db.service: Succeeded.
Feb 26 21:51:36 phoniebox systemd[1]: Started Daily man-db regeneration.
Feb 26 21:51:36 phoniebox mopidy[463]: INFO     [MpdSession-14] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38106
Feb 26 21:51:37 phoniebox systemd[1]: systemd-hostnamed.service: Succeeded.
Feb 26 21:51:37 phoniebox mopidy[463]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38110
Feb 26 21:51:38 phoniebox systemd[1]: apt-daily.service: Succeeded.
Feb 26 21:51:38 phoniebox systemd[1]: Started Daily apt download activities.
Feb 26 21:51:38 phoniebox systemd[1]: Starting Daily apt upgrade and clean activities...
Feb 26 21:51:41 phoniebox systemd[1]: apt-daily-upgrade.service: Succeeded.
Feb 26 21:51:41 phoniebox systemd[1]: Started Daily apt upgrade and clean activities.
Feb 26 21:51:42 phoniebox mopidy[463]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38114
Feb 26 21:51:47 phoniebox mopidy[463]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38118
Feb 26 21:51:52 phoniebox mopidy[463]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38122
Feb 26 21:51:57 phoniebox mopidy[463]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38126
Feb 26 21:52:02 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Feb 26 21:52:02 phoniebox mopidy[463]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38130
Feb 26 21:52:08 phoniebox mopidy[463]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38134
Feb 26 21:52:13 phoniebox mopidy[463]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38138
Feb 26 21:52:18 phoniebox mopidy[463]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38142
Feb 26 21:52:20 phoniebox systemd[1]: Started Session c3 of user pi.
Feb 26 21:52:23 phoniebox mopidy[463]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38146

```

s-martin, 2020-02-27:
Unfortunately the log file was just restarted, so in this part there's nothing unusual.

Could you reboot, try to use spotify and then post the output of `tail -n 500 /var/log/syslog` again? Thanks.

dhanessar, 2020-02-27:
Thanks for your patience!

Now I rebooted the box, did try to use spotify several time by clicking on the spotify menu.
And this is the result of `tail -n 500 /var/log/syslog`
```
Feb 27 22:54:19 phoniebox rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="344" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Feb 27 22:54:19 phoniebox systemd[1]: logrotate.service: Succeeded.
Feb 27 22:54:19 phoniebox systemd[1]: Started Rotate log files.
Feb 27 22:54:19 phoniebox systemd[1]: phpsessionclean.service: Succeeded.
Feb 27 22:54:19 phoniebox systemd[1]: Started Clean php session files.
Feb 27 22:54:23 phoniebox mopidy[462]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52216
Feb 27 22:54:24 phoniebox systemd[1]: man-db.service: Succeeded.
Feb 27 22:54:24 phoniebox systemd[1]: Started Daily man-db regeneration.
Feb 27 22:54:26 phoniebox systemd[1]: apt-daily.service: Succeeded.
Feb 27 22:54:26 phoniebox systemd[1]: Started Daily apt download activities.
Feb 27 22:54:26 phoniebox systemd[1]: Starting Daily apt upgrade and clean activities...
Feb 27 22:54:28 phoniebox mopidy[462]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52220
Feb 27 22:54:30 phoniebox systemd[1]: apt-daily-upgrade.service: Succeeded.
Feb 27 22:54:30 phoniebox systemd[1]: Started Daily apt upgrade and clean activities.
Feb 27 22:54:34 phoniebox mopidy[462]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52224
Feb 27 22:54:39 phoniebox mopidy[462]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52228
Feb 27 22:54:40 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Feb 27 22:54:44 phoniebox mopidy[462]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52232
Feb 27 22:54:49 phoniebox mopidy[462]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52236
Feb 27 22:54:54 phoniebox mopidy[462]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52240
Feb 27 22:55:00 phoniebox mopidy[462]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52244
Feb 27 22:55:05 phoniebox mopidy[462]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52248
Feb 27 22:55:10 phoniebox mopidy[462]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52252
Feb 27 22:55:15 phoniebox mopidy[462]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52256
Feb 27 22:55:20 phoniebox mopidy[462]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52260
Feb 27 22:55:24 phoniebox mopidy[462]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52264
Feb 27 22:55:24 phoniebox mopidy[462]: INFO     [MpdSession-33] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52268
Feb 27 22:55:29 phoniebox mopidy[462]: INFO     [MpdSession-34] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52272
Feb 27 22:55:34 phoniebox mopidy[462]: INFO     [MpdSession-35] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52276
Feb 27 22:55:40 phoniebox mopidy[462]: INFO     [MpdSession-36] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52280
Feb 27 22:55:45 phoniebox mopidy[462]: INFO     [MpdSession-37] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52284
Feb 27 22:55:50 phoniebox mopidy[462]: INFO     [MpdSession-38] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52288
Feb 27 22:55:55 phoniebox mopidy[462]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52292
Feb 27 22:56:00 phoniebox mopidy[462]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52296
Feb 27 22:56:06 phoniebox mopidy[462]: INFO     [MpdSession-41] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52300
Feb 27 22:56:11 phoniebox mopidy[462]: INFO     [MpdSession-42] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52304
Feb 27 22:56:16 phoniebox mopidy[462]: INFO     [MpdSession-43] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52308
```
I am afraid it doesn't look so different.

s-martin, 2020-02-28:
Ok, what's the output of `tail -n 500 /var/log/mopidy/mopidy.log`?

dhanessar, 2020-02-28:
That's quite short!
```
tail -n 500 /var/log/mopidy/mopidy.log

tail: cannot open '/var/log/mopidy/mopidy.log' for reading: No such file or directory
```

s-martin, 2020-02-28:
Can you try `journalctl -u mopidy`?

dhanessar, 2020-02-28:
Then some errors apear in the output;
```
journalctl -u mopidy

-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Fri 2020-02-28 22:14:56 CET. --
Feb 28 22:12:24 phoniebox systemd[1]: Starting Mopidy music server...
Feb 28 22:12:24 phoniebox systemd[1]: Started Mopidy music server.
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.1
Feb 28 22:12:38 phoniebox mopidy[467]: ERROR    [MainThread] mopidy.ext Failed to load extension iris: No module named 'handlers'
Feb 28 22:12:38 phoniebox mopidy[467]: Traceback (most recent call last):
Feb 28 22:12:38 phoniebox mopidy[467]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/ext.py", line 209, in load_extensions
Feb 28 22:12:38 phoniebox mopidy[467]:     extension_class = entry_point.resolve()
Feb 28 22:12:38 phoniebox mopidy[467]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
Feb 28 22:12:38 phoniebox mopidy[467]:     module = __import__(self.module_name, fromlist=['__name__'], level=0)
Feb 28 22:12:38 phoniebox mopidy[467]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/__init__.py", line 7, in <module>
Feb 28 22:12:38 phoniebox mopidy[467]:     import handlers
Feb 28 22:12:38 phoniebox mopidy[467]: ModuleNotFoundError: No module named 'handlers'
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Feb 28 22:12:38 phoniebox mopidy[467]: INFO     [MainThread] mopidy.config Loading config from command line options
Feb 28 22:12:38 phoniebox mopidy[467]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Feb 28 22:12:39 phoniebox mopidy[467]: INFO     [MainThread] mopidy.__main__ Enabled extensions: m3u, file, mpd, local, softwaremixer, spotify, http, stream, local-images
Feb 28 22:12:39 phoniebox mopidy[467]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Feb 28 22:12:40 phoniebox mopidy[467]: ERROR    [MainThread] mopidy.__main__ Extension local-images failed during setup. This might have left the registry in a bad state.
Feb 28 22:12:40 phoniebox mopidy[467]: Traceback (most recent call last):
Feb 28 22:12:40 phoniebox mopidy[467]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/__main__.py", line 129, in main
Feb 28 22:12:40 phoniebox mopidy[467]:     extension.setup(registry)
Feb 28 22:12:40 phoniebox mopidy[467]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_local_images/__init__.py", line 31, in setup
Feb 28 22:12:40 phoniebox mopidy[467]:     from .library import ImageLibrary
Feb 28 22:12:40 phoniebox mopidy[467]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_local_images/library.py", line 52, in <module>
Feb 28 22:12:40 phoniebox mopidy[467]:     class ImageLibrary(local.Library):
Feb 28 22:12:40 phoniebox mopidy[467]: AttributeError: module 'mopidy.local' has no attribute 'Library'
Feb 28 22:12:41 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Feb 28 22:12:41 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Feb 28 22:12:41 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Feb 28 22:12:41 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Feb 28 22:12:41 phoniebox mopidy[467]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Feb 28 22:12:42 phoniebox mopidy[467]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 7baikc8mqeawsn4t579esi6j3
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 4 Spotify playlists
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 28 22:12:43 phoniebox mopidy[467]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 28 22:12:46 phoniebox mopidy[467]: INFO     [MpdSession-14] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36030
Feb 28 22:12:46 phoniebox mopidy[467]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36034
Feb 28 22:12:47 phoniebox mopidy[467]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36038
Feb 28 22:12:47 phoniebox mopidy[467]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36042
[... cut ...]
Feb 28 22:14:40 phoniebox mopidy[467]: INFO     [MpdSession-37] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36122
Feb 28 22:14:46 phoniebox mopidy[467]: INFO     [MpdSession-38] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36126
Feb 28 22:14:51 phoniebox mopidy[467]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36130
Feb 28 22:14:56 phoniebox mopidy[467]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36134
lines 48-70/70 (END)
```

s-martin, 2020-02-28:
What does `python3 -m pip show Mopidy-Iris` return?

dhanessar, 2020-02-28:
It does return this:
''''
python3 -m pip show Mopidy-Iris

Name: Mopidy-Iris
Version: 3.43.0
Summary: A fully-functional Mopidy web client encompassing Spotify and many other backends
Home-page: https://github.com/jaedb/iris
Author: James Barnsley
Author-email: james@barnsley.nz
License: Apache License, Version 2.0
Location: /usr/local/lib/python3.7/dist-packages
Requires: setuptools, requests, Mopidy, Mopidy-Local-Images, ConfigObj
Required-by:
''''

s-martin, 2020-02-29:
Please use `python3 -m pip install --upgrade Mopidy-Iris`, so the Iris package upgrades to the latest version.

dhanessar, 2020-02-29:
I did. And it download some new files. But I still get the error message 404: NotFound

`python3 -m pip show Mopidy-Iris` returns now
````
Name: Mopidy-Iris
Version: 3.45.1
Summary: Fully-featured Mopidy frontend client
Home-page: https://github.com/jaedb/iris
Author: James Barnsley
Author-email: james@barnsley.nz
License: Apache License, Version 2.0
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: Mopidy, Pykka, setuptools
Required-by:
````

dhanessar, 2020-02-29:
`journalctl -u mopidy` returns now
````
-- Logs begin at Sat 2020-02-29 09:58:35 CET, end at Sat 2020-02-29 10:04:32 CET. --
Feb 29 09:58:40 phoniebox systemd[1]: Starting Mopidy music server...
Feb 29 09:58:40 phoniebox systemd[1]: Started Mopidy music server.
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.1
Feb 29 09:58:54 phoniebox mopidy[464]: ERROR    [MainThread] mopidy.ext Failed to load extension iris: No module named 'handlers'
Feb 29 09:58:54 phoniebox mopidy[464]: Traceback (most recent call last):
Feb 29 09:58:54 phoniebox mopidy[464]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/ext.py", line 209, in load_extensions
Feb 29 09:58:54 phoniebox mopidy[464]:     extension_class = entry_point.resolve()
Feb 29 09:58:54 phoniebox mopidy[464]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
Feb 29 09:58:54 phoniebox mopidy[464]:     module = __import__(self.module_name, fromlist=['__name__'], level=0)
Feb 29 09:58:54 phoniebox mopidy[464]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/__init__.py", line 7, in <module>
Feb 29 09:58:54 phoniebox mopidy[464]:     import handlers
Feb 29 09:58:54 phoniebox mopidy[464]: ModuleNotFoundError: No module named 'handlers'
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.config Loading config from command line options
Feb 29 09:58:54 phoniebox mopidy[464]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.__main__ Enabled extensions: http, spotify, local, mpd, stream, local-images, softwaremixer, m3u, file
Feb 29 09:58:54 phoniebox mopidy[464]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Feb 29 09:58:56 phoniebox mopidy[464]: ERROR    [MainThread] mopidy.__main__ Extension local-images failed during setup. This might have left the registry in a bad state.
Feb 29 09:58:56 phoniebox mopidy[464]: Traceback (most recent call last):
Feb 29 09:58:56 phoniebox mopidy[464]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/__main__.py", line 129, in main
Feb 29 09:58:56 phoniebox mopidy[464]:     extension.setup(registry)
Feb 29 09:58:56 phoniebox mopidy[464]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_local_images/__init__.py", line 31, in setup
Feb 29 09:58:56 phoniebox mopidy[464]:     from .library import ImageLibrary
Feb 29 09:58:56 phoniebox mopidy[464]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_local_images/library.py", line 52, in <module>
Feb 29 09:58:56 phoniebox mopidy[464]:     class ImageLibrary(local.Library):
Feb 29 09:58:56 phoniebox mopidy[464]: AttributeError: module 'mopidy.local' has no attribute 'Library'
Feb 29 09:58:56 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Feb 29 09:58:56 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Feb 29 09:58:56 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Feb 29 09:58:57 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Feb 29 09:58:57 phoniebox mopidy[464]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Feb 29 09:58:58 phoniebox mopidy[464]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 7baikc8mqeawsn4t579esi6j3
Feb 29 09:58:58 phoniebox mopidy[464]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 4 Spotify playlists
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 29 09:58:59 phoniebox mopidy[464]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 29 09:59:07 phoniebox mopidy[464]: INFO     [MpdSession-14] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55316
Feb 29 09:59:43 phoniebox mopidy[464]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55320
Feb 29 09:59:47 phoniebox mopidy[464]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55324
Feb 29 09:59:47 phoniebox mopidy[464]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55328
[... cut ...]
Feb 29 10:04:16 phoniebox mopidy[464]: INFO     [MpdSession-69] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55536
Feb 29 10:04:21 phoniebox mopidy[464]: INFO     [MpdSession-70] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55540
Feb 29 10:04:27 phoniebox mopidy[464]: INFO     [MpdSession-71] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55544
Feb 29 10:04:32 phoniebox mopidy[464]: INFO     [MpdSession-72] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55548
````

s-martin, 2020-02-29:
Maybe you need to restart mopidy with `sudo systemctl restart mopidy` or just reboot your pi.

dhanessar, 2020-02-29:
I did reboot my pi after updating moidy-iris. Now I also tried `sudo systemctl restart mopidy` but it makes no difference. Still "404: Not Found"

s-martin, 2020-03-01:
I try to check again with the Iris maintainers https://github.com/jaedb/Iris/issues/474#issuecomment-593135821

kingosticks, 2020-03-01:
If you are using the Mopidy service then you need to install/upgrade stuff with pip using `sudo python3 -m pip install --upgrade Mopidy-Iris`. 

Without `sudo` it will install the package in the user's home directory (/home/pi/.local/lib/python3.7/site-packages). But the Mopidy service cannot see those files. If you were to run `sudo mopidyctl deps` to display your versions I imagine it will show the version used by the service (installed at /usr/local/lib/python3.7/dist-packages/mopidy_iris/) is old.

s-martin, 2020-03-01:
Thanks @kingosticks!

@dhanessar, sorry I missed the `sudo`.

dhanessar, 2020-03-01:
It is working again!!! :-)
Thank you for your help!
My Daughter will be happy!

dhanessar, 2020-03-01:
I have the same phenomen like issue #799. But that is a minor problem which doesn't bother me when using the RFID cards.
I will keep an eye on issue #799.

s-martin, 2020-03-07:
Seems to be resolved, I close this now.
If you still experience this issue feel free to reopen this ticket.

closed 2020-03-07 by s-martin