MiczFlor/RPi-Jukebox-RFID #760: :bulb: Multiple RFID Readers

Franzformator, 2020-01-30:
## Feature Description
Multiple RFID Readers for parallel Card and "Figure" function. 
### What functionality would you like to see in your phoniebox?
Control the Phoniebox by placing a “Figure” on the top or **alternatively** by placing a card in a separate “Card-Slot”. Both functions should have a separate RFID Reader to ensure a good wireless reception. 
### How do you envision the feature to work from a users perspective?
It should be possible to control the Phoniebox with RFID Figures and Cards. Set a Figure on one reader would start play. Removing the figure will stop playing. Same function for Cards.
If a figure and a card is on a reader simultaneously the last placed RFID chip should be followed. 
### Further information that might help
In the moment I’m using a Phoniebox with a Neuftech RFID reader and it only plays if a RFID chip is on the reader (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/62)
This feature is elementary for me!
I already have a card slot and a position for figures on my Phoniebox. For now, I have realized this with two RFID coils on one reader. This works under most conditions. But sometimes the RFID chip in a card is to bad, so no card is detected.
It would be nice to get the possibility to connect multiple readers (2+). If it is not possible with the Neuftech reader, maybe it ist possible with RC522

s-martin, 2020-01-30:
I know some people used the Neuftech reader and soldered additional spirals from a cheap reader in parallel to the Neuftech spiral.

s-martin, 2020-01-30:
Looking at the reader code, this could be possible.
But I don't know, if it's possible two differentiate 2 reader devices.

s-martin, 2020-01-31:
Does somebody have two readers and can check how list devices works?

s-martin, 2020-02-02:
#665 is somehow related although it's a different use case.

Wallnusskaese, 2020-02-15:
I have thought of this feature from a different angle. I have some bluetooth headphones which I pair over RFID (just glued an chip on them). 
I think it would be neat, if they had their own spot to hold onto. 
According to [https://www.raspberrypi.org/forums/viewtopic.php?t=195628](url), from the hardware side this could be done easily at least for the RC522 I'm using. From what i red there, they just show up at /dev/spidev0.0 and /dev/spidev0.1. 

I did not dig into all of this too much though, because my use case just don't justifies the time i could be listening to audiobooks with my little ones otherwise :D.

closed 2020-07-16 by MiczFlor