MiczFlor/RPi-Jukebox-RFID #761: rfid reader does not work

deheingit, 2020-01-30:
Hi all,

i installed phoniebox with the single line script and was asked for the RFID Reader (USB) which was detected. 

I also re-ran the configuration:
```
pi@MusicBoxJonah:~/RPi-Jukebox-RFID/scripts $ python2 RegisterDevice.py
Choose the reader from list
(0, 'HXGCoLtd Keyboard')
Device Number: 0
```

But when I look in the webinterface the RFID reader is diabled. If i click enable it turns green but after refreshing the page its disabed again.

Any idea what to do?
Thanks so much!

Abakaba, 2020-01-30:
Hi,

I have exactly the same problem.

I did a clean install for buster using the single line script. I am using the Neuftech USB card reader.

Best,
Cornelius

githubtee, 2020-01-30:
Hi,
exactly the same here (HXGCoLtd Keyboard) - also the same behaviour like in issue #747 

The changes from python3->python2 in `daemon_rfid_reader.py` and `phoniebox-rfid-reader.service` are already done.

`sudo systemctl status phoniebox-rfid-reader`

results in

`pi@Phoniebox:~ $ sudo systemctl status phoniebox-rfid-reader
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-01-30 22:38:22 UTC; 18s ago
  Process: 772 ExecStart=/usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 772 (code=exited, status=1/FAILURE)

Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Service hold-off time over, scheduling restart.
Jan 30 22:38:22 Phoniebox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Jan 30 22:38:22 Phoniebox systemd[1]: Failed to start Phoniebox RFID-Reader Service.
Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Jan 30 22:38:22 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
`

Any idea what is buggy?
Thx

cedivan0718, 2020-01-31:
Hello, I am also having the same issue. The web interface is not recognizing my rfid card reader and when i turn it on the page refreshes and the rfid reader remains as disabled

s-martin, 2020-01-31:
Did you try https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/747#issuecomment-579223322 ??

I think it's the same issue as #747 Unfortunately this hasn't been been fixed for everyone.

cedivan0718, 2020-01-31:
Yes i went through the entire thread and have tried things like changing the script for the rfid reader to python 2 instead of 3 but nothing has worked

s-martin, 2020-01-31:
> Yes i went through the entire thread and have tried things like changing the script for the rfid reader to python 2 instead of 3 but nothing has worked

Did you also try manually restarting the service mentioned in that specific comment I linked?

githubtee, 2020-01-31:
> Yes i went through the entire thread and have tried things like changing the script for the rfid reader to python 2 instead of 3 but nothing has worked

same here

s-martin, 2020-01-31:
@MiczFlor as this issue is probably the same as #747 I suggest to track it in issue #747 and close this issue.

deheingit, 2020-01-31:
Yes that makes sense. Sorry for double posting and thanks for your replies

MiczFlor, 2020-01-31:
Hi @s-martin 
would be good to have an english issue thread as default for the same problems, rather than German, no?

githubtee, 2020-01-31:
hi all!
i could fix this reader problem by replacing the `daemon_rfid_reader.py` with the file from release 1.1.8
now the rfid-reader service works and i can register rfid cards in the web gui.
since i am not a python coder so i cannot explain the real solution behind this.

@MiczFlor 
can you use this information to find a solution for the last release `daemon_rfid_reader.py`?

githubtee, 2020-01-31:
i tested it with 2018-03-13-raspbian-stretch as mentioned in the installation instruction.

s-martin, 2020-01-31:
Could you try out something?

Replace `cardid = reader.readCard()` with `cardid = reader.reader.readCard()` in the file `daemon_rfid_reader.py`.

Does it still work?

githubtee, 2020-01-31:
> Could you try out something?
> 
> Replace `cardid = reader.readCard()` with `cardid = reader.reader.readCard()` in the file `daemon_rfid_reader.py`.
> 
> Does it still work?

i already tried it before and unfortunately it didn't work.

s-martin, 2020-01-31:
Probably fixed, when #767 is merged.

s-martin, 2020-02-01:
Fix in #767 was merged. 

Could everyone still having this issue, check if a installation described in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch works now again?

deheingit, 2020-02-01:
@s-martin: Thank you. 
Sorry for the noob question: Does "merged" mean the fix is included in the stable version? ;)
Also I used buster before, does it make any difference? So should the one-line installer work again?

Thanks again

s-martin, 2020-02-01:
Yes, the fix is now included in the stable version.

One line installer should work now again.

cedivan0718, 2020-02-01:
Hello, Should i delete the old installation and try the one line script for stretch again? By the way I appreciate all the help!

s-martin, 2020-02-01:
> Hello, Should i delete the old installation and try the one line script for stretch again? By the way I appreciate all the help!

Yes, but be aware, if you delete the old installation you lose **all** your settings.

cedivan0718, 2020-02-01:
Just reinstalled the whole thing and now the rfid reader works! Scans just as it’s supposed to but now i have a problem on phoniebox itself. I registered an rfid card with an album from spotify but now when i scan the card only the “thumbnail” of the album shows up instead of actually playing the album itself, any clue as to what this problem could be?

s-martin, 2020-02-01:
> Just reinstalled the whole thing and now the rfid reader works! Scans just as it’s supposed to but now i have a problem on phoniebox itself. I registered an rfid card with an album from spotify but [...]

Good to hear that the reader works again.
The not working streaming could be related to #766

deheingit, 2020-02-01:
werks - thanks you!

closed 2020-02-01 by deheingit