MiczFlor/RPi-Jukebox-RFID #765: :bug: [mpd not starting at boot using buster]

Alwiz79, 2020-01-31:
So I upgraded my Raspbian Stretch to Buster.
After that I one-line-installed the latest Phoniebox, keeping my old setting.
Everything wokred fine.

Now I see a severe issue:

It seems as if mpd server is not  starting from boot. At least no music will be played from the web frontend, also the volume setting in "inactive".
As soon as I ssh to the box and start mpd manually
`sudo mpd`
everything is back to normal and working.

Is there a known issue with buster? Is a like an "autostart config file" that can be edited and added mpd?

s-martin, 2020-01-31:
Maybe related to #767

s-martin, 2020-02-01:
@Alwiz79 did you check, if this is also fixed now?

kuesimon, 2020-02-01:
Hi. I Tested it today with the master branch and do have the same issue as Alwiz79. So this is not fixed yet.

s-martin, 2020-02-02:
Could you reboot and if it's not working post the output of `tail -n 1000 /var/log/syslog`?

s-martin, 2020-02-19:
@Alwiz79, @kuesimon: Some major issues were fixed in the last two weeks.

Could you do a new installation from `master` and check, if this fixes your issue?

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

 **ATTENTION**: Removing the directory /home/pi/RPi-jukebox-RFID deletes **all your settings** and also your **music files**!

kuesimon, 2020-02-27:
@s-martin Sry for my late reply. My intention at that time was to install the Spotify version and this was done with errors. So tested the none spotify version where i had the described error. Now Spotify version is working except my bug report (#810 ).

I did not try to reinstalll the non.spotify version but i try to quickly do it when i open my box to fix #810 iand will post my findings for the mpd issue then.

Regards

s-martin, 2020-04-10:
Removed from 2.0 milestone as maybe not relevant anymore.

closed 2020-05-22 by s-martin