MiczFlor/RPi-Jukebox-RFID #766: :bug: [Webstreams not working]

Alwiz79, 2020-01-31:
After updating my Stretch installation to Buster and one-line-installing the latest phoniebox my before working webstreams are not working anymore.

When I ssh to the box and enter 
`mpg123 -@ http://stream.spreeradio.de/spree-80er/mp3-128/kontakt/play.m3u`
everything is working fine.

But it won't play using the web frontend or swipe cards.
Any idea?

s-martin, 2020-01-31:
Maybe realated to #767

s-martin, 2020-02-01:
@Alwiz79 did you check, if this is also fixed now?

cedivan0718, 2020-02-01:
When im on the spotify website via phoniebox tab and try to manually add songs on to the queue instead of scan it says “mopidy:failed to add some tracks” don’t know if this could be what is wrong but yeah still only shows album cover when swiping the rfid

kuesimon, 2020-02-02:
I'm a little bit unsure if my comment is really fittig to this bug as it'spotify related...

I installed both - buster non-spotify and buster+spotify on my box. As mentioned in #765 the non-spotify is working now after typing in sudo mpd.

For the spotify version there is still only the thumbnail of the playlist and no sound - even with the current master install script.

s-martin, 2020-02-02:
Unfortunately Spotify version is broken right now as there are some package version issue.

Alwiz79, 2020-02-03:
I just found out that linking the actual stream adress (ending with *.mp3) which is stored within the m3u rather then the m3u is working.
So for me, this issue is closed now.

closed 2020-02-03 by Alwiz79