MiczFlor/RPi-Jukebox-RFID #768: :bug: RFID Reader - ID wird durch Phoniebox nicht eingelesen

Thr33LittleBirds, 2020-02-01:
## Hardware
**PI:**
Raspberry Pi Model B Rev 2p
**RFID Reader**
KKmoon HID 413d:2107
https://www.amazon.de/gp/product/B011XI2DE8/ref=as_li_qf_sp_asin_il_tl?ie=UTF8&camp=1638&creative=6742&creativeASIN=B011XI2DE8&linkCode=as2&tag=diebarnfeld-21

## Bug

### What I did
Ich habe den Raspberry aufgesetzt und das folgende Script verwendet
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

### What happened
Das Aufsetzen hat funktioniert und auch der RFIDReader schreibt eine ID ins Terminal (abgeschlossen durch ein Enter). Aber die ID erscheint nicht im WebFrontend und auch nicht in den Textdateien.

### Further information that might help
Der RFID Reader funktioniert. Die Karten werden eingelesen und ins Terminal geschrieben
Ich kann ihn unter RegisterDevice.py auswählen und er steht auch in der Textdatei
Per Hand kann ich die _./rfid_trigger_play.sh -i=1234565_ ausführen und der Wert wird erfolgreich auch im WebFrontend angezeigt.

Ich könnt in die Tischkante beißen... 
Ich hoffe ihr könnt mir helfen.

s-martin, 2020-02-01:
Ist vermutlich der gleiche Fehler wie in #747 und #761.

Wenn PR #767 gemergt ist, löst das hoffentlich ein paar Probleme.

s-martin, 2020-02-01:
Fix in #767 was merged. 

Could everyone still having this issue, check if a installation described in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch works now again?

Thr33LittleBirds, 2020-02-01:
Läuft! Ich danke Euch!

closed 2020-02-01 by Thr33LittleBirds

jochmann, 2020-03-22:
Checked, unfortunately an installation from scratch did not fix the issue for me. Also with a reader HID 413d:2107

s-martin, 2020-03-22:
What's the output of `tail -n 500 /var/log/syslog`?

jochmann, 2020-06-14:
Sorry it took me so long. Have been quite distracted by world events.

The output of `tail -n 500 /var/log/syslog` is quite long.

Moreover, I have a growing suspicion that the setup process is not working for me and important libraries are being skipped or somehow missed. I wonder if that happens due to packet loss via SSH-headless setup? In any event, having the install process fail without a warning might be an issueson its own. Perhaps not return a "installation complete" if there are missing libraries?

--

```
Jun 14 19:02:03 raspberrypi kernel: [    0.233366] CacheFiles: Loaded
Jun 14 19:02:03 raspberrypi kernel: [    0.247275] NET: Registered protocol family 2
Jun 14 19:02:03 raspberrypi kernel: [    0.248521] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes)
Jun 14 19:02:03 raspberrypi kernel: [    0.249031] TCP established hash table entries: 8192 (order: 3, 32768 bytes)
Jun 14 19:02:03 raspberrypi kernel: [    0.249191] TCP bind hash table entries: 8192 (order: 4, 65536 bytes)
Jun 14 19:02:03 raspberrypi kernel: [    0.249405] TCP: Hash tables configured (established 8192 bind 8192)
Jun 14 19:02:03 raspberrypi kernel: [    0.249643] UDP hash table entries: 512 (order: 2, 16384 bytes)
Jun 14 19:02:03 raspberrypi kernel: [    0.249742] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
Jun 14 19:02:03 raspberrypi kernel: [    0.250126] NET: Registered protocol family 1
Jun 14 19:02:03 raspberrypi kernel: [    0.251110] RPC: Registered named UNIX socket transport module.
Jun 14 19:02:03 raspberrypi kernel: [    0.251153] RPC: Registered udp transport module.
Jun 14 19:02:03 raspberrypi kernel: [    0.251179] RPC: Registered tcp transport module.
Jun 14 19:02:03 raspberrypi kernel: [    0.251204] RPC: Registered tcp NFSv4.1 backchannel transport module.
Jun 14 19:02:03 raspberrypi kernel: [    0.253484] hw perfevents: enabled with armv7_cortex_a7 PMU driver, 5 counters available
Jun 14 19:02:03 raspberrypi kernel: [    0.257711] Initialise system trusted keyrings
Jun 14 19:02:03 raspberrypi kernel: [    0.258109] workingset: timestamp_bits=14 max_order=18 bucket_order=4
Jun 14 19:02:03 raspberrypi kernel: [    0.271456] FS-Cache: Netfs 'nfs' registered for caching
Jun 14 19:02:03 raspberrypi kernel: [    0.272350] NFS: Registering the id_resolver key type
Jun 14 19:02:03 raspberrypi kernel: [    0.272419] Key type id_resolver registered
Jun 14 19:02:03 raspberrypi kernel: [    0.272446] Key type id_legacy registered
Jun 14 19:02:03 raspberrypi kernel: [    0.272485] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
Jun 14 19:02:03 raspberrypi kernel: [    0.279688] Key type asymmetric registered
Jun 14 19:02:03 raspberrypi kernel: [    0.279741] Asymmetric key parser 'x509' registered
Jun 14 19:02:03 raspberrypi kernel: [    0.279852] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 250)
Jun 14 19:02:03 raspberrypi kernel: [    0.280125] io scheduler noop registered
Jun 14 19:02:03 raspberrypi kernel: [    0.280161] io scheduler deadline registered (default)
Jun 14 19:02:03 raspberrypi kernel: [    0.280481] io scheduler cfq registered
Jun 14 19:02:03 raspberrypi kernel: [    0.280515] io scheduler mq-deadline registered (default)
Jun 14 19:02:03 raspberrypi kernel: [    0.280544] io scheduler kyber registered
Jun 14 19:02:03 raspberrypi systemd[1]: Starting Remount Root and Kernel File Systems...
Jun 14 19:02:03 raspberrypi kernel: [    0.283097] bcm2708_fb soc:fb: FB found 1 display(s)
Jun 14 19:02:03 raspberrypi kernel: [    0.293752] Console: switching to colour frame buffer device 82x26
Jun 14 19:02:03 raspberrypi kernel: [    0.301611] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 656x416
Jun 14 19:02:03 raspberrypi kernel: [    0.309742] bcm2835-rng 3f104000.rng: hwrng registered
Jun 14 19:02:03 raspberrypi kernel: [    0.312822] vc-mem: phys_addr:0x00000000 mem_base=0x3ec00000 mem_size:0x40000000(1024 MiB)
Jun 14 19:02:03 raspberrypi systemd[1]: Started udev Coldplug all Devices.
Jun 14 19:02:03 raspberrypi kernel: [    0.319101] vc-sm: Videocore shared memory driver
Jun 14 19:02:03 raspberrypi kernel: [    0.322337] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
Jun 14 19:02:03 raspberrypi kernel: [    0.343935] brd: module loaded
Jun 14 19:02:03 raspberrypi kernel: [    0.361133] loop: module loaded
Jun 14 19:02:03 raspberrypi systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Jun 14 19:02:03 raspberrypi kernel: [    0.364732] Loading iSCSI transport class v2.0-870.
Jun 14 19:02:03 raspberrypi kernel: [    0.368683] libphy: Fixed MDIO Bus: probed
Jun 14 19:02:03 raspberrypi kernel: [    0.371578] usbcore: registered new interface driver lan78xx
Jun 14 19:02:03 raspberrypi kernel: [    0.374392] usbcore: registered new interface driver smsc95xx
Jun 14 19:02:03 raspberrypi systemd[1]: Started Set the console keyboard layout.
Jun 14 19:02:03 raspberrypi kernel: [    0.377066] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Jun 14 19:02:03 raspberrypi kernel: [    0.407773] dwc_otg 3f980000.usb: base=(ptrval)
Jun 14 19:02:03 raspberrypi systemd[1]: Started Helper to synchronize boot up for ifupdown.
Jun 14 19:02:03 raspberrypi kernel: [    0.610891] Core Release: 2.80a
Jun 14 19:02:03 raspberrypi kernel: [    0.613597] Setting default values for core params
Jun 14 19:02:03 raspberrypi kernel: [    0.616375] Finished setting default values for core params
[... cut ...]
Jun 14 19:05:59 raspberrypi mpd[976]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Jun 14 19:05:59 raspberrypi systemd[964]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jun 14 19:05:59 raspberrypi systemd[964]: mpd.service: Failed with result 'exit-code'.
Jun 14 19:05:59 raspberrypi systemd[964]: Failed to start Music Player Daemon.
Jun 14 19:05:59 raspberrypi systemd[964]: Reached target Default.
Jun 14 19:05:59 raspberrypi systemd[964]: Startup finished in 2.262s.
```

s-martin, 2020-06-14:
I think you need to use the correct Reader file.

Please try
`cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.kkmoonRFIDreader home/pi/RPi-Jukebox-RFID/scripts/Reader.py`
and 
`sudo systemctl restart  phoniebox-rfid-reader`

jochmann, 2020-06-14:
```
cp: cannot create regular file 'home/pi/RPi-Jukebox-RFID/scripts/Reader.py': No such file or directory
```

The warning "no such file or directory" also appears during install several times.

MiczFlor, 2020-06-14:
I think the copy location is missing the initial slash
~~~
cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.kkmoonRFIDreader /home/pi/RPi-Jukebox-RFID/scripts/Reader.py
~~~

s-martin, 2020-06-14:
> I think the copy location is missing the initial slash
> 
> ```
> cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.kkmoonRFIDreader /home/pi/RPi-Jukebox-RFID/scripts/Reader.py
> ```

Oops, sorry.
Thanks for catching that.

jochmann, 2020-06-14:
with the leading slash the command went through (no confirmation printed, but at least no error message).

Still no change to the functionality, unfortunately. The rfid reader beeps, but phoniebox GUI does not show an ID.

s-martin, 2020-06-14:
What's the output of 
`sudo tail -n 200 /var/log/lighttpd/error.log`

and
` tail -n 200 /var/log/syslog`?

jochmann, 2020-06-14:
2020-06-14 21:07:53: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:53: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:55: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:55: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:55: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:57: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:58: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:58: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:58: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:07:59: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:07:59: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:07:59: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:08:00: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:08:00: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:08:00: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8
2020-06-14 21:08:01: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:08:01: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
[... cut ...]
2020-06-14 21:18:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(../settings/Latest_Folder_Played): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 4
2020-06-14 21:18:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  file_get_contents(/home/pi/RPi-Jukebox-RFID/shared/latestID.txt): failed to open stream: No such file or directory in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 5
2020-06-14 21:18:54: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/ajax.refresh_id.php on line 8

jochmann, 2020-06-14:
Jun 14 21:07:20 raspberrypi systemd[1]: Reached target Login Prompts.
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 1.
Jun 14 21:07:20 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 14 21:07:20 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 14 21:07:20 raspberrypi systemd[493]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 14 21:07:20 raspberrypi systemd[493]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 2.
Jun 14 21:07:20 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 14 21:07:20 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 14 21:07:20 raspberrypi systemd[518]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 14 21:07:20 raspberrypi systemd[518]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 14 21:07:20 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 14 21:07:20 raspberrypi systemd[1]: Started OpenBSD Secure Shell server.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 3.
Jun 14 21:07:21 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 14 21:07:21 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 14 21:07:21 raspberrypi systemd[588]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 14 21:07:21 raspberrypi systemd[588]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 4.
Jun 14 21:07:21 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 14 21:07:21 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 14 21:07:21 raspberrypi systemd[688]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 14 21:07:21 raspberrypi systemd[688]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 5.
Jun 14 21:07:21 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 14 21:07:21 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Start request repeated too quickly.
Jun 14 21:07:22 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 14 21:07:22 raspberrypi systemd[1]: Failed to start Phoniebox Rotary Encoder Service.
Jun 14 21:07:22 raspberrypi systemd[1]: Started Samba NMB Daemon.
Jun 14 21:07:22 raspberrypi systemd[1]: Starting Samba SMB Daemon...
Jun 14 21:07:22 raspberrypi systemd[1]: Started Lighttpd Daemon.
Jun 14 21:07:22 raspberrypi exim4[473]: Starting MTA: exim4.
Jun 14 21:07:22 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
Jun 14 21:07:22 raspberrypi daemon_rfid_reader.py[487]: 2020-06-14 21:07:22,688 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Jun 14 21:07:22 raspberrypi daemon_rfid_reader.py[487]: Traceback (most recent call last):
Jun 14 21:07:22 raspberrypi daemon_rfid_reader.py[487]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 38, in <module>
Jun 14 21:07:22 raspberrypi daemon_rfid_reader.py[487]:     cardid = reader.reader.readCard()
[... cut ...]
Jun 14 22:30:08 raspberrypi mpd[1684]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Jun 14 22:30:08 raspberrypi systemd[1671]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jun 14 22:30:08 raspberrypi systemd[1671]: mpd.service: Failed with result 'exit-code'.
Jun 14 22:30:08 raspberrypi systemd[1671]: Failed to start Music Player Daemon.
Jun 14 22:30:08 raspberrypi systemd[1671]: Reached target Default.
Jun 14 22:30:08 raspberrypi systemd[1671]: Startup finished in 2.250s.

s-martin, 2020-06-15:
In this line
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/c94448c6313cdc3e61aaea4b34e9a1de5a1eccdb/scripts/daemon_rfid_reader.py#L38

you have to remove one occurrance of `reader` so that the line is only `reader.readCard()`.

Afterwards please use `sudo systemctl restart phoniebox-rfid-reader`

jochmann, 2020-06-15:
can confirm that the line now reads `cardid = reader.readCard()` but after the sudo command and also after subsequent reboot, no change in bahavior.

s-martin, 2020-06-15:
Ok, can you post the syslog results after a card swipe again?

jochmann, 2020-06-15:
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 3.
Jun 15 09:07:09 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 15 09:07:09 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 15 09:07:09 raspberrypi systemd[580]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 15 09:07:09 raspberrypi systemd[580]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 15 09:07:09 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 4.
Jun 15 09:07:10 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 15 09:07:10 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun 15 09:07:10 raspberrypi systemd[642]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Jun 15 09:07:10 raspberrypi systemd[642]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 5.
Jun 15 09:07:10 raspberrypi systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Start request repeated too quickly.
Jun 15 09:07:10 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jun 15 09:07:10 raspberrypi systemd[1]: Failed to start Phoniebox Rotary Encoder Service.
Jun 15 09:07:11 raspberrypi systemd[1]: Started Samba NMB Daemon.
Jun 15 09:07:11 raspberrypi systemd[1]: Starting Samba SMB Daemon...
Jun 15 09:07:11 raspberrypi systemd[1]: Started Lighttpd Daemon.
Jun 15 09:07:11 raspberrypi exim4[476]: Starting MTA: exim4.
Jun 15 09:07:11 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
Jun 15 09:07:11 raspberrypi daemon_rfid_reader.py[492]: 2020-06-15 09:07:11,472 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Jun 15 09:07:12 raspberrypi systemd[1]: Started Samba SMB Daemon.
Jun 15 09:07:13 raspberrypi systemd[1]: Started Music Player Daemon.
Jun 15 09:07:13 raspberrypi systemd[1]: Starting Phoniebox Startup...
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
Jun 15 09:07:13 raspberrypi bash[850]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
Jun 15 09:07:13 raspberrypi bash[850]: SECONDSWIPE="RESTART"
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOIFACENAME="PCM"
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOVOLCHANGESTEP="3"
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOVOLMAXLIMIT="100"
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOVOLMINLIMIT="1"
Jun 15 09:07:13 raspberrypi bash[850]: AUDIOVOLSTARTUP="0"
Jun 15 09:07:13 raspberrypi bash[850]: VOLCHANGEIDLE="TRUE"
Jun 15 09:07:13 raspberrypi bash[850]: IDLETIMESHUTDOWN="0"
Jun 15 09:07:13 raspberrypi bash[850]: SHOWCOVER="ON"
Jun 15 09:07:13 raspberrypi bash[850]: READWLANIPYN="OFF"
Jun 15 09:07:13 raspberrypi bash[850]: EDITION="classic"
Jun 15 09:07:13 raspberrypi bash[850]: LANG="en-UK"
Jun 15 09:07:13 raspberrypi bash[850]: VERSION="2.0 - ed888da - master"
Jun 15 09:07:13 raspberrypi bash[850]: 0
Jun 15 09:07:15 raspberrypi bash[850]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Jun 15 09:07:15 raspberrypi bash[850]: #011version 1.25.10; written and copyright by Michael Hipp and others
Jun 15 09:07:15 raspberrypi bash[850]: #011free software (LGPL) without any warranty but with best wishes
Jun 15 09:07:16 raspberrypi bash[850]: Directory: /home/pi/RPi-Jukebox-RFID/shared/
Jun 15 09:07:16 raspberrypi bash[850]: Playing MPEG stream 1 of 1: startupsound.mp3 ...
Jun 15 09:07:16 raspberrypi bash[850]: MPEG 1.0 L III cbr64 44100 j-s
Jun 15 09:07:16 raspberrypi bash[850]: Artist:  Phoniebox
Jun 15 09:07:16 raspberrypi dhcpcd[474]: wlan0: fe80::1 is reachable again
Jun 15 09:07:16 raspberrypi dhcpcd[474]: wlan0: fe80::1 is reachable again
Jun 15 09:07:20 raspberrypi bash[850]: [0:04] Decoding of startupsound.mp3 finished.
Jun 15 09:07:20 raspberrypi systemd[1]: Started Phoniebox Startup.
Jun 15 09:07:29 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
[... cut ...]
Jun 15 09:15:32 raspberrypi mpd[1251]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Jun 15 09:15:32 raspberrypi systemd[1239]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jun 15 09:15:32 raspberrypi systemd[1239]: mpd.service: Failed with result 'exit-code'.
Jun 15 09:15:32 raspberrypi systemd[1239]: Failed to start Music Player Daemon.
Jun 15 09:15:32 raspberrypi systemd[1239]: Reached target Default.
Jun 15 09:15:32 raspberrypi systemd[1239]: Startup finished in 2.094s.

s-martin, 2020-06-15:
Did y swipe a card before you read the log? 

What does the lighttpd log say?

MiczFlor, 2020-06-15:
hi @jochmann 
you had a Phoniebox running before. was that the same hardware?

BStramke, 2020-07-31:
I have the same issue, i cannot see the CardID in the WebUI, but it will be sent to the Console Input directly (just like typing with a keyboard).

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $  python3 RegisterDevice.py
Choose the reader from list
0 C-Media Electronics Inc.       USB PnP Sound Device
1 HID 413d:2107
Device Number: 1

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ls -la /dev/input/by-id/
total 0
drwxr-xr-x 2 root root  80 Jul 31 13:40 .
drwxr-xr-x 4 root root 140 Jul 31 13:40 ..
lrwxrwxrwx 1 root root   9 Jul 31 13:40 usb-413d_2107-event-kbd -> ../event0
lrwxrwxrwx 1 root root   9 Jul 31 13:40 usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-event-if03 -> ../event1

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ lsusb
Bus 001 Device 006: ID 8086:0808 Intel Corp.
Bus 001 Device 007: ID 0781:5571 SanDisk Corp. Cruzer Fit
Bus 001 Device 005: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 004: ID 413d:2107
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jochmann, 2021-02-07:
@MiczFlor just to tie a bow on this, after I had to take some time off trouble shooting (was too frustrating to continue and then 2020 really got in the way, sorry):

Yes, I was running the same hardware before, which is why I never could understand or reproduce the original state. After reading #1046 did I finally get it: I must have swiped the working phoniebox with the configuration card, thus setting the reader into an unusable state. Running the fix described in the issue finally worked. Thank you so much for updating the docs so I could stumble across it!

I still think that the installation script could be a problem: Repeatedly I've experienced that a library was missing in the installation process due to connection issues with the relevant IPs online. The process continues to run, though, instead of abandoning right away and telling me about it. And only if you have at least passing experience with *nix can you tell and fix after the fact with some apt-get update or rerun of the installation routine. I don't know if this is an easy fix or even a problem for others, though.

Sascha4546, 2022-07-13:
Hello, i noticed that Neuftech changed the pcb of the reader ... i have the old and the new one here... the old version works fine and the new one seems to make problems

s-martin, 2022-07-13:
> Hello, i noticed that Neuftech changed the pcb of the reader ... i have the old and the new one here... the old version works fine and the new one seems to make problems
> 
> https://bilderupload.org/bild/404233483-old-works-fine https://bilderupload.org/bild/d0f533530-new-causing-problems

Could you open a new issue (and include the pictures in the issue itself)?

Comments in already closed issues are usually overlooked, because they are considered done and nobody opens them.