MiczFlor/RPi-Jukebox-RFID #773: NAS Musik Phoniebox howto?

Chairon22, 2020-02-02:
Hi Everybody,
does anybody have an idea howto get NAS (openmediavault) working with Phoniebox?
I have no idea (newbie) howto get the shared folders to shine up in Webapp...
Thx in Advance :-)
Great Projekt and im thankful for every the helping steps i already recieved!

ZyanKLee, 2020-02-02:
Phoniebox currently does not provide any means to do this in the web ui. (Afaik, at least.)

If I remember correctly there are several options implemented in openmediavault to connect to it, aren't there?
Samba, nfs and open-iscsi should be the most prominent.
You can search the web for those methods and how to use them on raspbian or Debian, Ubuntu and mount them on `RPi-Jukebox-RFID/shared/` and `RPi-Jukebox-RFID/playlists`. At least that is what I would try to start with. :wink:

Chairon22, 2020-02-03:
Ein Stuttgarter :-) Da brauch ich mir ja keinen Abmühen in Englisch...

Richtig - ich hatte auch schon mal halbherzig einen Versuch in dieser Richtung unternommen - OMV bietet natürlich alle möglichen Freigaben, Samba, Cifs, NFS und so... Hatte eine Anleitung gefunden - aber ich glaube - jetzt wo ich schreiben - fällt mir ein dass ich in OMV noch kein Nutzer "Pi" eingegeben hatte - vl. klappts deshalb nciht. Naja weiter testen - ich zumindest froh, dass ich es hinbekommen haben, dass das Ding nun die Tags ausliest... Weiterbasteln.

Danke Dir für die Info!

ZyanKLee, 2020-02-03:
Gerne - und weiterhin viel Erfolg. Falls dieses Problem damit behoben ist, schließe doch bitte das Ticket.

Chairon22, 2020-02-03:
Mache ich - vielen Dank - ich denke ich krieg das hi, daher schließe ich nun.

closed 2020-02-03 by Chairon22