MiczFlor/RPi-Jukebox-RFID #778: Mopidy works, but doesn't when started as a service

hoodvisions, 2020-02-06:
Hi there, I have to apologize right away for being a complete noob. When something doesn't work I re-install buster using etcher and start again. Now, I am running out of time and ideas...

Buster One Line Install +Spotify after you guys fixed the issues with the RFID Reader lately.
When opening the Spotify Interface it gives me an ERR_CONNECTION_REFUSED.

So after researching etc. I did sudo apt remove mopidy-spotify and sudo apt remove mopidy and reinstalled mopidy as described on mopidy.com.

Now, when I start mopidy manually Spotify works!!!
If I use sudo systemctl start mopidy the Spotify interface shows me a 404.

I have no idea what to do now as all I need is the mopidy stuff to run even after I power the raspberry off. Otherwise it's kind of a nuisance.
Any ideas how to get there? Thanks a lot!!!

hoodvisions, 2020-02-07:
I realize this is not a bug, but really an issue I probably created. However, can anyone point me the right direction where to find help and delete this thread afterwards? Sorry!

s-martin, 2020-02-07:
Right here is the place to get help :-)

Squelsh, 2020-02-09:
I had the same problem. For me it was a running `mpd` server which occupied the `mpd` port.
You should be able to see that problem in the logs ( "Unable to bind to port" or similar) . Mopidy somehow ignores this error when you start it manually.

I solved it by simply uninstalling mpd completely. It is not required for the Spotify edition.

Squelsh, 2020-02-09:
@MiczFlor 
Perhaps the installer script should only install mpd in the classic version.

s-martin, 2020-02-09:
> @MiczFlor
> Perhaps the installer script should only install mpd in the classic version.

It should behave exactly like that.

See line 655 and 656 in `scripts/installscripts/buster-install-default.sh`:

```
sudo systemctl disable mpd
sudo systemctl enable mopidy
```

Squelsh, 2020-02-09:
Oh, OK. Then please excuse me. I had a pretty screwed up installation whilst fighting with various conflicting software versions.

@hoodvisions What does the system's logfiles tell you?

hoodvisions, 2020-02-10:
Hi there, thanks a lot. I think my installation is also kinda weird. I have posted the dmesg log here: https://pastebin.com/REFaMKP1

I think I will also uninstall mpd now and see what happens. It seems I have to make another attempt at installing everything from scratch as so far I tend to break more things than fixing them :)

hoodvisions, 2020-02-10:
Oh but I got something interesting here....maybe? This is the output of `sudo systemctl status mopidy`

> ● mopidy.service - Mopidy music server
   Loaded: loaded (/lib/systemd/system/mopidy.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-02-10 13:20:43 CET; 35s ago
  Process: 417 ExecStartPre=/bin/mkdir -p /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 442 ExecStartPre=/bin/chown mopidy:audio /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 448 ExecStart=/usr/bin/mopidy --config /usr/share/mopidy/conf.d:/home/pi/.config/mopidy/mopidy.conf (code=exited, status=1/FAILURE)
 Main PID: 448 (code=exited, status=1/FAILURE)

Feb 10 13:20:43 raspberrypi mopidy[448]:     raw_config = _load(files, defaults, keyring.fetch() + (overrides or []))
Feb 10 13:20:43 raspberrypi mopidy[448]:   File "/usr/lib/python3/dist-packages/mopidy/config/__init__.py", line 159, in _load
Feb 10 13:20:43 raspberrypi mopidy[448]:     if f.is_dir():
Feb 10 13:20:43 raspberrypi mopidy[448]:   File "/usr/lib/python3.7/pathlib.py", line 1351, in is_dir
Feb 10 13:20:43 raspberrypi mopidy[448]:     return S_ISDIR(self.stat().st_mode)
Feb 10 13:20:43 raspberrypi mopidy[448]:   File "/usr/lib/python3.7/pathlib.py", line 1161, in stat
Feb 10 13:20:43 raspberrypi mopidy[448]:     return self._accessor.stat(self)
Feb 10 13:20:43 raspberrypi mopidy[448]: PermissionError: [Errno 13] Permission denied: '/home/pi/.config/mopidy/mopidy.conf'
Feb 10 13:20:43 raspberrypi systemd[1]: mopidy.service: Main process exited, code=exited, status=1/FAILURE
Feb 10 13:20:43 raspberrypi systemd[1]: mopidy.service: Failed with result 'exit-code'.

s-martin, 2020-02-10:
It's definitely an issue, but I'm not sure what the file permissions should be.

Maybe someone experienced with mopidy can help, @ZyanKLee ?

ZyanKLee, 2020-02-10:
Nah, I'm not really experienced with mopidy.

hoodvisions, 2020-02-10:
Ok I fixed the above message simply by copying the mopidy.conf file into a folder where mopidy has reading access to. The mopidy service is running now, but as long as it does myphonieboxIP:6680/iris/ will show a 404. It's so weird. I have to stop the mopidy service. Then run it manually via 'mopidy' and only then it works...

I will definitely start from scratch now, one last time, and if it doesn't work I'll bite the bullet and ask for someone to sell me a working phoniebox.

ZyanKLee, 2020-02-10:
As what user is mopidy running at your system? And why does it not have any
access to the config file.

hoodvisions, 2020-02-10:
Mopidy runs as user mopidy for some reason. I edited the service to use the conf file that came with the one line installation and is sitting in the home/pi folder, but this didn't work due to the permissions. However, when the service runs I get a 404 error. When I stop it the spotify interface simply doesn't respond and when I run mopidy manually everything is fine. It seems there's some stuff mixed up, from what I can tell and I don't want you guys to waste your time on this. I'll be re installing everything one more time starting with etcher and if this fails I'll drop a line here and close the issue. It must be either faulty hardware or me.

hoodvisions, 2020-02-10:
Ok I give up.

Raspberry Pi Model B Rev 2 with Raspbian Buster Image 2019-09-26-raspbian-buster-lite.img
USB Soundcard, Connection SSH via Wifi
Burned Image with Etcher. 

'sudo apt-get update
sudo apt-get upgrade

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh'

Spotify yes. 
Reboot.

Phoniebox Interface shows up.
Spotify interface ( ip:6680/iris) shows 'ERR_CONNECTION_REFUSED' in Chrome.

From here, when I install mopidy manually with their documentation spotify will work, but only when started manually via 'mopidy' as when it's started as a service 'sudo systemctl enable mopidy' and 'sudo systemctl start mopidy' it will show a 404 instead.

I suspect the installation as it is right now is simply not working 100% right. 
Is there a platform known where people sell their phoniebox maybe? I made a promise to show up with one, unfortunately :)

Many thanks for your help, so far, guys!

hoodvisions, 2020-02-10:
I am 99% convinced this is the exact same issue: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/784

Squelsh, 2020-02-10:
Hmm, I also had a various problems with mopidy (but I did a mix of install script and manual attempts).
I tried to run mopidy as mopidy user but this always failed when I tried to play local files.
No errors in the log, just no files in the cue. I could only fix this by running mopidy as the pi user.

What does mopidy tell you if you start it as standalone on the command line? Is mopidy listed as plugin?

hoodvisions, 2020-02-11:
When I start mopidy on the command line it works flawless. I can play Spotify playlists etc. If I run mopidy by starting the service or autostart on reboot it won't work. How do I list mopidy as a plugin and what does this mean?

s-martin, 2020-02-11:
As of now Phoniebox has no dedicated plugin concept.

MondMann2019, 2020-02-12:
I‘m experiencing the same issue. After stopping mopidy with `sudo systemctl stop mopidy` And starting ist via `mopidy` I can at least open the Spotify page. 
But there nothing shows up an the status of Spotify shows: read only. (Of course of have a premium account an entered all needed information). 

I had the Box running some time ago, but unfortunately tried something else, which didn’t work. Now local media plays, but Spotify is dead. I made a clean install.

s-martin, 2020-02-12:
I tried to fix something else on my Phoniebox yesterday, but experienced that `mpd` will just restart even when I used `sudo systemctl disable mpd`.

So maybe enabling mopidy and disabling mpd in the installer script https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/778#issuecomment-583855706 doesn`'t work (anymore)?
@MiczFlor I think we should look into that.

MiczFlor, 2020-02-12:
Are we looking at the master or develop install?

hoodvisions, 2020-02-12:
I have been using the A. default one line install script via `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

s-martin, 2020-02-12:
Probably related to #783

s-martin, 2020-02-17:
We have merged PR #789 into `master`.

Could you check, if you do a new installation, if this fixes your issue?

**ATTENTION**: Removing the directory `/home/pi/RPi-jukebox-RFID` deletes **all** your settings and also your **music files**!

hoodvisions, 2020-02-18:
Unfortunately no changes for me.

On a fresh SD card with 2019-09-26-raspbian-buster-full "burned" with Etcher.
Setup WiFi and static IP (dhcpcd.conf) as well as SSH Access and Boot into text terminal.
No System Updates.

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

Setup MPD when prompted, setup spotify and RFID when prompted.
Reboot, startup sound can be heard.

Open Phoniebox interface, upload a demo MP3 to a folder, press Play: no Sound :(
Click on Spotify interface (Iris) returns a Error 404 not found.

s-martin, 2020-02-18:
That's too bad :-(

What's the result of `sudo systemctl status mopidy` and `tail -n 500 /var/log/syslog`?

s-martin, 2020-02-18:
See also #783

hoodvisions, 2020-02-18:
`sudo systemctl status mopidy` returns

● mopidy.service - Mopidy music server
   Loaded: loaded (/lib/systemd/system/mopidy.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-18 15:57:04 CET; 3min 54s ago
  Process: 441 ExecStartPre=/bin/mkdir -p /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 453 ExecStartPre=/bin/chown mopidy:audio /var/cache/mopidy (code=exited, status=0/SUCCESS)
 Main PID: 454 (mopidy)
    Tasks: 17 (limit: 2200)
   Memory: 99.8M
   CGroup: /system.slice/mopidy.service
           └─454 /usr/bin/python2 /usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf

Feb 18 15:57:11 raspberrypi mopidy[454]: INFO     Audio output set to "alsasink"
Feb 18 15:57:12 raspberrypi mopidy[454]: ERROR    Spotify login error: <ErrorType.OTHER_PERMANENT: 10>
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     Logged into Spotify Web API as hoodvisions
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     Started spotify backend
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     Starting Mopidy core
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     Starting Mopidy frontends: MpdFrontend, HttpFrontend
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680
Feb 18 15:57:13 raspberrypi mopidy[454]: INFO     Starting GLib mainloop
Feb 18 16:00:50 raspberrypi mopidy[454]: INFO     Loaded 375 playlists

And the taillog can be found here;
https://pastebin.com/N2bMD4Wk

Thanks a lot for even looking into this! :)

s-martin, 2020-02-18:
So it looks like mopidy is running, which means PR #789 helped at least a little bit :-)

But I don't know enough about the Spotify integration why it's still a 404 error.

s-martin, 2020-02-18:
It seems that mopidy-iris dropped support for python 2 and mopidy2 after version 3.43.0. PR #793 should fix this.

JoeyRamone, 2020-02-18:
@s-martin I've checked your commit, ran "sudo python2.7 -m pip install Mopidy-Iris==3.43.0" and the 404 is gone. 
I am going to re-install the pi now and will drop a comment again. First check looks good...

JoeyRamone, 2020-02-18:
@s-martin @hoodvisions 
With one-line
'cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/Mopidy-iris-py2/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=Mopidy-iris-py2 bash ./buster-install-default.sh'

the installation worked fine and spotify is working again.

s-martin, 2020-02-18:
PR #793 was merged to `master` and should fix this issue.

Please test.

hoodvisions, 2020-02-19:
All right, cool guys! Spotify now works out of the box. Now it's only the phoniebox not wanting to play audio files. However, I will close this issue now as it's been opened in regards to the mopidy service and there's a bunch of threads about the audio issue opened already. Again, thanks so much guys!

closed 2020-02-19 by hoodvisions

MondMann2019, 2020-02-19:
Unfortunately the one line JoeyRamone mentions is not working for me. I always get an Error 404 : Not Found. 
Could you please check it? ;)

s-martin, 2020-02-19:
> Unfortunately the one line JoeyRamone mentions is not working for me. I always get an Error 404 : Not Found.
> Could you please check it? ;)

The command from @JoeyRamone was only an interim solution, but the PR #793 is now merged.

Please use the standard installation command:
ˋcd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.shˋ

**ATTENTION**: Removing the directory /home/pi/RPi-jukebox-RFID deletes **all your settings** and also your **music files**.

s-martin, 2020-02-19:
Please follow up for any further questions in #783 as this issue has been closed and is a duplicate anyway.

JoeyRamone, 2020-02-19:
As written in the pull request, I've tested with Raspi 4, Spotify only, standard Audio Line out (no hifi berry).
If problem still exists, I'll try tomorrow again with the master branch.

JoeyRamone, 2020-02-19:
@hoodvisions Did you try a "sudo python2.7 -m pip install Mopidy-Iris==3.43.0" and restart the pi, when you still got the 404?

MondMann2019, 2020-02-20:
I used the merged line and now everything is working perfectly!!! Thank you!!!