MiczFlor/RPi-Jukebox-RFID #779: MQTT Service startet nicht

nukleuz80, 2020-02-07:
Hallo zusammen,

ich habe die nötigen Schritte zum Einrichten des MQTT-Service laut Anleitung durchgeführt:
[Anleitung MQTT Service](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Smart-Home-remote-control-with-MQTT)

In der "daemon_mqtt_client.py" Datei wurden die Zugangsdaten für den ioBroker (MQTT) eingetragen. MQTT Server läuft auch, denn mehrere Clients im Netzwerk können darauf zugreifen.

Leider erhalte ich beim Service-Status anfordern (sudo systemctl status phoniebox-mqtt-client) folgende Fehlermeldung. 

` sudo systemctl status phoniebox-mqtt-client
● phoniebox-mqtt-client.service - Phoniebox MQTT Client Service
   Loaded: loaded (/etc/systemd/system/phoniebox-mqtt-client.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-02-07 09:59:53 CET; 10min ago
  Process: 1702 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py (code=exited, status=1/FAILURE)
 Main PID: 1702 (code=exited, status=1/FAILURE)

Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: phoniebox-mqtt-client.service: Service RestartSec=100ms expired, scheduling restart.
Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 5.
Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: Stopped Phoniebox MQTT Client Service.
Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: phoniebox-mqtt-client.service: Start request repeated too quickly.
Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Feb 07 09:59:53 rpi3b-phoniebox systemd[1]: Failed to start Phoniebox MQTT Client Service.
`

Ich benutze ein RPI3B+ mit Raspbian Buster (2019-09-26-raspbian-buster-full.img) in der PhonieBox Classic Version 2.0 rc7 (ohne Spotify).
Installiert wurde mit dem One-Line-Skript für Buster.

Die PhonieBox läuft mit allen Features wie RFID Reader und auch den GPIO-Buttons.

Danke für Unterstützung vorab.

nukleuz

nukleuz80, 2020-02-07:
Das Problem konnte selbst gelöst werden.

Die Zeilen in der **daemon_mqtt_client.py** mussten wie folgt geändert werden:

**mqttCA=""
mqttCert=""
mqttKey=""**

#mqttCA="/home/pi/MQTT/mqtt-ca.crt"
#mqttCert="/home/pi/MQTT/mqtt-client-phoniebox.crt"
#mqttKey="/home/pi/MQTT/mqtt-client-phoniebox.key"	

nukleuz

MiczFlor, 2020-02-07:
Hi @nukleuz80 
can you send the snippet that worked for you here in the thread? Not the entire file, but related to this part, which is taken from `daemon_mqtt_client.py` as it is now (in develop branch):
~~~
# ----------------------------------------------------------
#  SETTINGS
# ----------------------------------------------------------
DEBUG = False
mqttBaseTopic = "phoniebox"             # MQTT base topic
mqttClientId = "phoniebox"              # MQTT client ID
mqttHostname = "openHAB"                # MQTT server hostname
mqttPort = 8883                         # MQTT server port (typically 1883 for unencrypted, 8883 for encrypted)
mqttUsername = ""                       # username for user/pass based authentication
mqttPassword = ""                       # password for user/pass based authentication
mqttCA = "/home/pi/MQTT/mqtt-ca.crt"    # path to server certificate for certificate-based authentication
mqttCert = "/home/pi/MQTT/mqtt-client-phoniebox.crt"    # path to client certificate for certificate-based authentication
mqttKey = "/home/pi/MQTT/mqtt-client-phoniebox.key"     # path to client keyfile for certificate-based authentication
mqttConnectionTimeout = 60              # in seconds; timeout for MQTT connection
refreshIntervalPlaying = 5              # in seconds; how often should the status be sent to MQTT (while playing)
refreshIntervalIdle = 30                # in seconds; how often should the status be sent to MQTT (when NOT playing)
~~~
what does your script look like? Please delete your user and password :)

nukleuz80, 2020-02-07:
Hi MiczFlor,

here is the snippet from **daemon_mqtt_client.py** which worked for me.

`**SETTINGS**

DEBUG=True
mqttBaseTopic="phoniebox"				            # MQTT base topic
mqttClientId="phoniebox"				            # MQTT client ID
mqttHostname="192.168.0.56"			                # MQTT server hostname
mqttPort=1888						                # MQTT server port (typically 1883 for unencrypted, 8883 for encrypted)
mqttUsername=""						                # username for user/pass based authentication
mqttPassword=""						                # password for user/pass based authentication
mqttCA=""			                                # path to server certificate for certificate-based authentication
mqttCert=""	                                        # path to client certificate for certificate-based authentication
mqttKey=""	                                        # path to client keyfile for certificate-based authentication
mqttConnectionTimeout=60				            # in seconds; timeout for MQTT connection
refreshIntervalPlaying=5				            # in seconds; how often should the status be sent to MQTT (while playing)
refreshIntervalIdle=30					            # in seconds; how often should the status be sent to MQTT (when NOT playing)`

BTW: Before i changed the settings, i run the script directly "/usr/bin/python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py "(without the service command) there i get more issues from python debugger. So i get the info that the cert variables are wrong.

I hope it will help you or others.

nukleuz80, 2020-02-07:
Here is a screenshot from iobroker mqtt objects with the phoniebox data...

[image]

nikolaus-online, 2020-02-08:
I still have issues with the MQTT-service. I installed a fresh developer version on my pi (debian buster, one-line-install), and I had to install paho manually.

When I run the script manually it works like charm, but when I create it as a service as described, it fails. 

Here's what drives me nuts: Whenever I have run the script manually, the daemon runs without complaints. But when I reboot, it fails and keeps failing whenever restarted with "sudo systemctl restart phoniebox-mqtt-client".

`pi@raspberrypi:~ $ sudo systemctl status phoniebox-mqtt-client
● phoniebox-mqtt-client.service - Phoniebox MQTT Client Service
   Loaded: loaded (/etc/systemd/system/phoniebox-mqtt-client.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2020-02-08 20:03:22 CET; 47s ago
  Process: 919 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon-mqtt-client.py (code=exited, status=1/FAILURE)
 Main PID: 919 (code=exited, status=1/FAILURE)

Feb 08 20:03:22 raspberrypi systemd[1]: phoniebox-mqtt-client.service: Service RestartSec=100ms expired, scheduling r
Feb 08 20:03:22 raspberrypi systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 5
Feb 08 20:03:22 raspberrypi systemd[1]: Stopped Phoniebox MQTT Client Service.
Feb 08 20:03:22 raspberrypi systemd[1]: phoniebox-mqtt-client.service: Start request repeated too quickly.
Feb 08 20:03:22 raspberrypi systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Feb 08 20:03:22 raspberrypi systemd[1]: Failed to start Phoniebox MQTT Client Service.
`

Any ideas? Thanks in advance!

Nikolaus\

nukleuz80, 2020-02-08:
Hi @nikolaus-online,

in my case these settings had to be empty.

mqttCA=""	# path to server certificate for certificate-based authentication
mqttCert=""	# path to client certificate for certificate-based authentication
mqttKey=""	# path to client keyfile for certificate-based authentication

What are your settings?

nukleuz

nikolaus-online, 2020-02-09:
@nukleuz80 I changed that already.

> DEBUG = True
> mqttBaseTopic = "phoniebox"             # MQTT base topic
> mqttClientId = "phoniebox"              # MQTT client ID
> mqttHostname = "192.168.20.127"         # MQTT server hostname
> mqttPort = 1883                         # MQTT server port (typically 1883 for unencrypted, 8883 for encrypted)
> mqttUsername = "mqttklaus"                       # username for user/pass based authentication
> mqttPassword = "DidMQTTSvN"                       # password for user/pass based authentication
> mqttCA = ""    # path to server certificate for certificate-based authentication
> mqttCert = ""    # path to client certificate for certificate-based authentication
> mqttKey = ""     # path to client keyfile for certificate-based authentication
> mqttConnectionTimeout = 30              # in seconds; timeout for MQTT connection
> refreshIntervalPlaying = 5              # in seconds; how often should the status be sent to MQTT (while playing)
> refreshIntervalIdle = 30                # in seconds; how often should the status be sent to MQTT (when NOT playing)

Whenever I restart the daemon from the terminal, it works. Changed the rights of the .py-script with chmod -x . I'm running out of ideas. :-D

Thanks!

Nikolaus\

nukleuz80, 2020-02-09:
The only difference i see is the developer version you installed. My Version is 2.0 rc7.

I found something interesting in the ducumentation here:

> /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh 

Just copy this line and paste it into your terminal on the pi.

If you find something that doesn't work. And you tried and tried again, but it still doesn't work, please report your issue in the "issues" section. Make sure to include information about the system and hardware you are using, like:

Raspberry ZERO, OS Jessie, Card reader lists as (insert here) when running scripts/RegisterDevice.py, installed Phoniebox version 0.9.3 (or: using latest master branch).

I think a developer have to help you 😤

nukleuz

nikolaus-online, 2020-02-09:
Hey Nukleuz,

thanks for your help! I think I have found an ugly workaround:
I have added a RestartSec=10 to **phoniebox-mqtt-client.service**

Looks like this:

```
[Service]
User=pi
Group=pi
Restart=always
RestartSec=10
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon-mqtt-client.py
```

Now it works. 
[image]

Thanks again!

nukleuz80, 2020-02-11:
Hi Nikolaus,

i could reproduce the same issue with the mqtt service. After reboot the service didn´t start.
Your workaround is working! :)

Thank you! And i think the developer(s) will fix it.

nukleuz

MiczFlor, 2020-02-12:
Hi @nukleuz80 @nikolaus-online 
nice addition, thanks. I committed this to the develop branch:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/aa433a87074ab5c618eb1a1c2a207a114705b3b0
Will close the issue for now.

closed 2020-02-12 by MiczFlor

s-martin, 2020-02-12:
Should we add this change also to master?

MiczFlor, 2020-02-12:
@s-martin true. And done:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/sampleconfigs/phoniebox-mqtt-client.service.stretch-default.sample

commit aa433a87 (2020-02-12): MQTT service fix #779
commit bba7357d (2020-02-12): mqtt service restart fix #779