MiczFlor/RPi-Jukebox-RFID #783: Can't play anything / Spotify also not Working

DarthChoco, 2020-02-08:
After a clean installation there is no way to play any file. When clicking on any button there is an error message like this "An error occorured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup Output: RC: .1"

A click on the spotify link opens a new site with the information "Unable to connect" while trying to connect to the host with port 6680 and subdir "iris".

Is there any way to install an older stable release while there a such errors?

s-martin, 2020-02-08:
Are you using Stretch or Buster?

Did you install the regular stable installation from master or the one from develop?

s-martin, 2020-02-08:
And what's the result of `cat /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py`?

s-martin, 2020-02-08:
Please also press a button and post the result of `tail -n 500 /var/log/syslog`.

DarthChoco, 2020-02-09:
The installation was done on buster.

Output for "cat /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py"

`#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5                                                                                        102A
# due to the assignment of GPIO 19 and 21 to a buttons

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s                                                                                         interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for ad                                                                                        ditional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-f                                                                                        uer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please cr                                                                                        eate pull requests or contact me.

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

def def_recordstart():
    check_call("./scripts/playout_controls.sh -c=recordstart", shell=True)

def def_recordstop():
    check_call("./scripts/playout_controls.sh -c=recordstop", shell=True)

def def_recordplaylatest():
    check_call("./scripts/playout_controls.sh -c=recordplaylatest", shell=True)

shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(26,pull_up=True)
prev = Button(20,pull_up=True)
halt = Button(21,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds                                                                                         he will call a ra$
volU.when_held = def_volU
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 second                                                                                        s he will lower t$
volD.when_held = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt
#reco.when_pressed = def_recordstart
#reco.when_released = def_recordstop
#play.when_pressed = def_recordplaylatest

pause()`

Output for "tail -n 500 /var/log/syslog"

`Feb  9 09:01:10 raspberrypi systemd[1]: Created slice User Slice of UID 1000.
Feb  9 09:01:10 raspberrypi systemd[1]: Starting User Runtime Directory /run/user/1000...
Feb  9 09:01:10 raspberrypi systemd[1]: Started User Runtime Directory /run/user/1000.
Feb  9 09:01:10 raspberrypi systemd[1]: Starting User Manager for UID 1000...
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Paths.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Timers.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG network certificate management daemon.
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Sockets.
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Basic System.
Feb  9 09:01:11 raspberrypi systemd[6029]: Starting Music Player Daemon...
Feb  9 09:01:11 raspberrypi systemd[1]: Started User Manager for UID 1000.
Feb  9 09:01:11 raspberrypi systemd[1]: Started Session c3 of user pi.
Feb  9 09:01:12 raspberrypi mpd[6042]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Feb  9 09:01:12 raspberrypi systemd[6029]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Feb  9 09:01:12 raspberrypi systemd[6029]: mpd.service: Failed with result 'exit-code'.
Feb  9 09:01:12 raspberrypi systemd[6029]: Failed to start Music Player Daemon.
Feb  9 09:01:12 raspberrypi systemd[6029]: Reached target Default.
Feb  9 09:01:12 raspberrypi systemd[6029]: Startup finished in 1.252s.
Feb  9 09:04:14 raspberrypi lighttpd[820]: echo: write error: Broken pipe
Feb  9 09:04:18 raspberrypi lighttpd[820]: mpd error: Connection refused
Feb  9 09:04:18 raspberrypi lighttpd[820]: mpd error: Connection refused
Feb  9 09:04:19 raspberrypi lighttpd[820]: echo: write error: Broken pipe`

DarthChoco, 2020-02-09:
Output for "tail -n 500 /var/log/syslog"

`Feb  9 09:01:10 raspberrypi systemd[1]: Created slice User Slice of UID 1000.
Feb  9 09:01:10 raspberrypi systemd[1]: Starting User Runtime Directory /run/user/1000...
Feb  9 09:01:10 raspberrypi systemd[1]: Started User Runtime Directory /run/user/1000.
Feb  9 09:01:10 raspberrypi systemd[1]: Starting User Manager for UID 1000...
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Paths.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Timers.
Feb  9 09:01:11 raspberrypi systemd[6029]: Listening on GnuPG network certificate management daemon.
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Sockets.
Feb  9 09:01:11 raspberrypi systemd[6029]: Reached target Basic System.
Feb  9 09:01:11 raspberrypi systemd[6029]: Starting Music Player Daemon...
Feb  9 09:01:11 raspberrypi systemd[1]: Started User Manager for UID 1000.
Feb  9 09:01:11 raspberrypi systemd[1]: Started Session c3 of user pi.
Feb  9 09:01:12 raspberrypi mpd[6042]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Feb  9 09:01:12 raspberrypi systemd[6029]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Feb  9 09:01:12 raspberrypi systemd[6029]: mpd.service: Failed with result 'exit-code'.
Feb  9 09:01:12 raspberrypi systemd[6029]: Failed to start Music Player Daemon.
Feb  9 09:01:12 raspberrypi systemd[6029]: Reached target Default.
Feb  9 09:01:12 raspberrypi systemd[6029]: Startup finished in 1.252s.
Feb  9 09:04:14 raspberrypi lighttpd[820]: echo: write error: Broken pipe
Feb  9 09:04:18 raspberrypi lighttpd[820]: mpd error: Connection refused
Feb  9 09:04:18 raspberrypi lighttpd[820]: mpd error: Connection refused
Feb  9 09:04:19 raspberrypi lighttpd[820]: echo: write error: Broken pipe`

bastiitsab, 2020-02-09:
I have the exact same problem! I am trying to get phonibox running since last week. 
Can't play any songs in the web-site and can't connect to spotify (Connection refused) 

During the installation I saw the following error: 
```
Collecting Mopidy-Iris
  Using cached https://files.pythonhosted.org/packages/85/74/52465ef66d859720b7797a117af6be10264901b11116b63a8abbb845916d/Mopidy-Iris-3.44.0.tar.gz
  Installing build dependencies ... done
**Mopidy-Iris requires Python '>=3.7' but the running Python is 2.7.16**
Cloning into 'RPi-Jukebox-RFID'...
```

I am using a RPi 4 b

hoodvisions, 2020-02-10:
Same issue here. No sound playback via interface, no spotify (iris) after a clean install on buster. taillog after pressing play on a demo sound file after a clean reboot shows this https://pastebin.com/1wCVTGBY

bastiitsab, 2020-02-13:
@s-martin this change makes no difference for me. Can't play anything or use spotify like before :( 

I tried it with the Install of MPD during installation and without. Both not wirking.

s-martin, 2020-02-13:
Spotify edition works only with mopidy.

I have created PR #789 which may fix this, but this needs to be reviewed before it can be merged.

bastiitsab, 2020-02-13:
I have made the changes from the PR locally to test it

bastiitsab, 2020-02-14:
@s-martin I've done a clean installation now with the developer version. 
I choosed to not install MPD during installation.

The Startup Sound is working (Using Hifiberry MiniAMP) but playing local sounds doesn't do anything. Spotify gets a 404: Not Found.

_tail -n 500 /var/log/syslog_

> pi@phoniebox:~ $ tail -n 500 /var/log/syslog
Feb 14 21:33:00 phoniebox kernel: [    0.333973] bcmgenet fd580000.genet: failed to get enet clock
Feb 14 21:33:00 phoniebox systemd[1]: Started Daily apt download activities.
Feb 14 21:33:00 phoniebox kernel: [    0.334003] bcmgenet fd580000.genet: GENET 5.0 EPHY: 0x0000
Feb 14 21:33:00 phoniebox kernel: [    0.334034] bcmgenet fd580000.genet: failed to get enet-wol clock
Feb 14 21:33:00 phoniebox systemd[1]: Started Daily apt upgrade and clean activities.
Feb 14 21:33:00 phoniebox kernel: [    0.334063] bcmgenet fd580000.genet: failed to get enet-eee clock
Feb 14 21:33:00 phoniebox kernel: [    0.334098] bcmgenet: Skipping UMAC reset
Feb 14 21:33:00 phoniebox kernel: [    0.334341] unimac-mdio unimac-mdio.-19: DMA mask not set
Feb 14 21:33:00 phoniebox kernel: [    0.358142] libphy: bcmgenet MII bus: probed
Feb 14 21:33:00 phoniebox systemd[1]: Reached target Timers.
Feb 14 21:33:00 phoniebox kernel: [    0.398655] unimac-mdio unimac-mdio.-19: Broadcom UniMAC MDIO bus at 0x(ptrval)
Feb 14 21:33:00 phoniebox kernel: [    0.399556] usbcore: registered new interface driver r8152
Feb 14 21:33:00 phoniebox kernel: [    0.399636] usbcore: registered new interface driver lan78xx
Feb 14 21:33:00 phoniebox kernel: [    0.399703] usbcore: registered new interface driver smsc95xx
Feb 14 21:33:00 phoniebox systemd[1]: Listening on D-Bus System Message Bus Socket.
Feb 14 21:33:00 phoniebox kernel: [    0.400008] xhci_hcd 0000:01:00.0: xHCI Host Controller
Feb 14 21:33:00 phoniebox systemd[1]: Reached target Sockets.
Feb 14 21:33:00 phoniebox kernel: [    0.400052] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
Feb 14 21:33:00 phoniebox kernel: [    0.402229] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0000001000000890
Feb 14 21:33:00 phoniebox kernel: [    0.402563] genirq: irq_chip Brcm_MSI did not update eff. affinity mask of irq 55
Feb 14 21:33:00 phoniebox kernel: [    0.403428] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
Feb 14 21:33:00 phoniebox kernel: [    0.403462] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Feb 14 21:33:00 phoniebox kernel: [    0.403491] usb usb1: Product: xHCI Host Controller
Feb 14 21:33:00 phoniebox kernel: [    0.403514] usb usb1: Manufacturer: Linux 4.19.97-v7l+ xhci-hcd
Feb 14 21:33:00 phoniebox systemd[1]: Reached target Basic System.
Feb 14 21:33:00 phoniebox kernel: [    0.403538] usb usb1: SerialNumber: 0000:01:00.0
Feb 14 21:33:00 phoniebox systemd[1]: Starting Deferred execution scheduler...
Feb 14 21:33:00 phoniebox kernel: [    0.404051] hub 1-0:1.0: USB hub found
Feb 14 21:33:00 phoniebox kernel: [    0.404144] hub 1-0:1.0: 1 port detected
Feb 14 21:33:00 phoniebox kernel: [    0.404617] xhci_hcd 0000:01:00.0: xHCI Host Controller
Feb 14 21:33:00 phoniebox kernel: [    0.404654] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
Feb 14 21:33:00 phoniebox kernel: [    0.404690] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
Feb 14 21:33:00 phoniebox systemd[1]: Starting LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Feb 14 21:33:00 phoniebox kernel: [    0.405066] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 4.19
Feb 14 21:33:00 phoniebox kernel: [    0.405099] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Feb 14 21:33:00 phoniebox systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Feb 14 21:33:00 phoniebox kernel: [    0.405128] usb usb2: Product: xHCI Host Controller
Feb 14 21:33:00 phoniebox kernel: [    0.405151] usb usb2: Manufacturer: Linux 4.19.97-v7l+ xhci-hcd
Feb 14 21:33:00 phoniebox systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Feb 14 21:33:00 phoniebox kernel: [    0.405174] usb usb2: SerialNumber: 0000:01:00.0
Feb 14 21:33:00 phoniebox kernel: [    0.405687] hub 2-0:1.0: USB hub found
Feb 14 21:33:00 phoniebox kernel: [    0.405767] hub 2-0:1.0: 4 ports detected
Feb 14 21:33:00 phoniebox kernel: [    0.406974] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Feb 14 21:33:00 phoniebox systemd[1]: Started Regular background program processing daemon.
Feb 14 21:33:00 phoniebox kernel: [    0.407193] dwc_otg: FIQ enabled
Feb 14 21:33:00 phoniebox kernel: [    0.407202] dwc_otg: NAK holdoff enabled
Feb 14 21:33:00 phoniebox systemd[1]: Starting rng-tools.service...
Feb 14 21:33:00 phoniebox kernel: [    0.407211] dwc_otg: FIQ split-transaction FSM enabled
Feb 14 21:33:00 phoniebox kernel: [    0.407224] Module dwc_common_port init
Feb 14 21:33:00 phoniebox cron[352]: (CRON) INFO (pidfile fd = 3)
Feb 14 21:33:00 phoniebox kernel: [    0.407421] usbcore: registered new interface driver uas
Feb 14 21:33:00 phoniebox kernel: [    0.407531] usbcore: registered new interface driver usb-storage
Feb 14 21:33:00 phoniebox kernel: [    0.407715] mousedev: PS/2 mouse device common for all mice
Feb 14 21:33:00 phoniebox kernel: [    0.408969] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[... cut ...]
Feb 14 21:34:25 phoniebox mopidy[573]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 14 21:34:25 phoniebox mopidy[573]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 14 21:34:25 phoniebox mopidy[573]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 14 21:34:26 phoniebox lighttpd[828]: No LSB modules are available.
Feb 14 21:34:26 phoniebox lighttpd[828]: tail: cannot open '../logs/debug.log' for reading: No such file or directory
Feb 14 21:34:27 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
pi@phoniebox:~ $

s-martin, 2020-02-17:
We have merged PR #789 into `master`.

Could you check, if you do a new installation, if this fixes your issue?

**ATTENTION**: Removing the directory `/home/pi/RPi-jukebox-RFID` deletes **all** your settings and also your **music files**!

s-martin, 2020-02-17:
> @s-martin I've done a clean installation now with the developer version.
> I choosed to not install MPD during installation.
> 
> The Startup Sound is working (Using Hifiberry MiniAMP) but playing local sounds doesn't do anything. Spotify gets a 404: Not Found.

`develop` is still heavily work in progress, so I don't expect that this works at the moment.

But we have merged  PR #789 to `master` yesterday, which should fix at least some Spotify issues.

So you could try master again, see also https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/783#issuecomment-586860974

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

bastiitsab, 2020-02-18:
thanks @s-martin will try this today! Should I select to install MPD or not during Installation?

s-martin, 2020-02-18:
That's too bad :-(

What's the result of `sudo systemctl status mopidy` and `tail -n 500 /var/log/syslog`?

s-martin, 2020-02-18:
It seems that mopidy-iris dropped support for python 2 and mopidy2 after version 3.43.0. PR #793 should fix this.

s-martin, 2020-02-18:
PR #793 was merged to `master` and should fix this issue.

Please test.

bastiitsab, 2020-02-19:
> PR #793 was merged to `master` and should fix this issue.
> 
> Please test.

Can confirm, Spotify is working with this fix!
Big thanks for this! 

But local files can't be played anyways. Do you know where the problem is with local files playing? I could try to help or at least provide log-files!

s-martin, 2020-02-19:
> > PR #793 was merged to `master` and should fix this issue.
> > Please test.
> 
> Can confirm, Spotify is working with this fix!

Great to hear!

> But local files can't be played anyways. Do you know where the problem is with local files playing? I could try to help or at least provide log-files!

You could try to play a local file and then post the results of `tail -n 500 /var/log/syslog`

bastiitsab, 2020-02-19:
> Great to hear!
>
>
> You could try to play a local file and then post the results of `tail -n 500 /var/log/syslog`

see my previous post. 

Here I tried one of the test sample files: 

> Feb 19 10:32:06 phoniebox mopidy[567]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
> Feb 19 10:32:10 phoniebox lighttpd[824]: No LSB modules are available.
> Feb 19 10:32:10 phoniebox lighttpd[824]: tail: cannot open '../logs/debug.log' for reading: No such file or directory
> Feb 19 10:33:13 phoniebox mopidy[567]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
> Feb 19 10:33:13 phoniebox mopidy[567]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
> Feb 19 10:33:17 phoniebox lighttpd[824]: /home/pi/RPi-Jukebox-RFID/scripts/inc.writeFolderConfig.sh: line 106: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ZZZ/ABC/folder.conf: No such file or directory
> Feb 19 10:33:17 phoniebox lighttpd[824]: rm: cannot remove '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/ZZZ/ABC/folder.conf': No such file or directory
> Feb 19 10:33:18 phoniebox mopidy[567]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
>

hoodvisions, 2020-02-19:
Now that Spotify (iris) is working audio is still not. I have posted my taillog here after a clean install + reboot, after pressing the play button once for an uploaded test-mp3:
https://pastebin.com/Qd3cgmbQ
Hope this helps in any way.

s-martin, 2020-02-19:
> Update2: I just tried to "Scan Music Library". Afterwards I was able to play local files!? Have to test frther

That would make sense as you got the error 

>  Feb 19 10:33:17 phoniebox lighttpd[824]: /home/pi/RPi-Jukebox-RFID/scripts/inc.writeFolderConfig.sh: line 106: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ZZZ/ABC/folder.conf: No such file or directory

before.

Not sure, why the files weren’t indexed in the first place, though. @MiczFlor, do you have any ideas?

s-martin, 2020-02-19:
> Now that Spotify (iris) is working audio is still not. I have posted my taillog here after a clean install + reboot, after pressing the play button once for an uploaded test-mp3:
> https://pastebin.com/Qd3cgmbQ
> Hope this helps in any way.

I have seen a „warning“ that your local files aren’t indexed

> Feb 19 14:45:00 raspberrypi mopidy[457]: INFO     No local library metadata cache found at /var/lib/mopidy/local/library.json.gz. Please run `mopidy local scan` to index your local music library. If you do not have a local music collection, you can disable the local backend to hide this message

Did you try  `mopidy local scan`?

In addition I didn’t see anywhere that you would play a mp3 file.

hoodvisions, 2020-02-19:
You are right!! I have run the scan and now the file apparently plays, but I have no audio. So this might be an issue I have to solve on my own. Will test and come back asap!

bastiitsab, 2020-02-20:
It seems that the local files sometimes are not correct indexed. After manual indexing over the website the local files are working. 

Spotify is working like charm! Thanks again @s-martin ! 

But with spotify Playlists I have one Bug. The given name is not displayed on the home Screen. 
I've made a Screenshot from this. The Top Spotify List is marked with the yellow box from me. The second spotify is without that mark. 
[image]

s-martin, 2020-02-20:
Good to hear that it is now working.

Could you open a new issue for the display problem as it's not really related to this issue? Thanks.

hoodvisions, 2020-02-20:
I have run a fresh install and for some reason I have no sound neither in spotify (iris) nor the phoniebox. The Startup sound plays fine. Should I open a separate issue on this or is it likely it's the same issue?

s-martin, 2020-02-20:
Could you show the log output?

hoodvisions, 2020-02-21:
Yes, here it is on [pastebin](https://pastebin.com/6x3Q3LM8) :)

s-martin, 2020-02-22:
Good to hear, it was fixed for you.

I would close the ticket, if there are new problems, please open a new ticket.

closed 2020-02-22 by s-martin