MiczFlor/RPi-Jukebox-RFID #784: No Sound over Webinterface, no IRIS(404)

Thr33LittleBirds, 2020-02-09:
I've installed the Phoniebox very often in the past days.
Put I was not able to play Songs. These are my last Attempts.

### Attempt 1
Raspberry Pi Model B Rev 2
Raspbian Buster Image: 2019-09-26-raspbian-buster-lite.img
USB Soundcard

Connection SSH; Wifi

### Procedure
At 12:00 today

```
sudo apt-get update
sudo apt-get upgrade

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
```
Installation
- with Spotify-Option

-> reboot

### Result
No Mopidy Server

------------------------------------------------------------------------------------

### Attempt 2
Raspberry Pi Model B Rev 2
Raspbian Buster Image: 2019-09-26-raspbian-buster-lite.img
USB Soundcard

Connection SSH; Wifi

### Procedure
At 13:30 today

```
sudo apt-get update
sudo apt-get upgrade

develop branch

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh

[....]

sudo reboot
```

`sudo nano /usr/share/alsa/alsa.conf `

Set:
defaults.ctl.card 1
defaults.pcm.card 1

`cd /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/; ./CreateSampleAudiofoldersStreams.sh`

### Result
**Mopidy Server Status**
active (running) since Sun 2020-02-09 14:13:54 GMT; 2min 18s ago

**http://IP/iris**
404: Not Found

**Test**
speaker-test -t wav -c 6 -> Works
Play alphabet-abc.mp3 over Webinterface -> **Nothing is happening**

hlgrbmbm, 2020-02-10:
I have exactly the same problem.
The RFID card reader now works very well, but the linked music is not played (as well as over the webinterface)

best regards
axel

hoodvisions, 2020-02-10:
Having the exact same problem here. RFID service seems to be working now, but neither playback via interface nor Iris works (`ERR_CONNECTION_REFUSED` in Chrome).

s-martin, 2020-02-12:
Probably related to #783

s-martin, 2020-02-17:
We have merged PR #789 into `master`.

Could you check, if you do a new installation, if this fixes your issue?

**ATTENTION**: Removing the directory `/home/pi/RPi-jukebox-RFID` deletes **all** your settings and also your **music files**!

Thr33LittleBirds, 2020-02-18:
Hello. 
Here's my last try

**Raspbian Buster Image: 2019-09-26-raspbian-buster-lite.img
Connection SSH; Wifi, USB Soundcard**

**It's still not possible to play a file over the webinterface**

### Procedure
```
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
```
with Spotify

**Error while installation Mopidy-Iris**
Mopidy-Iris requires Python '>=3.7' but the running Python is 2.7.16

```
reboot
sudo nano /usr/share/alsa/alsa.conf 

Set:
defaults.ctl.card 1
defaults.pcm.card 1

cd /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/; ./CreateSampleAudiofoldersStreams.sh

reboot
```

### Result
Mopidy Server Status
active (running)

http://IP/iris
404: Not Found -> Error while installation Mopidy-Iris

### Test
speaker-test -t wav -c 6 -> Works
Press play on any Folder over Webinterface -> Nothing is happening

### tail -n 500 /var/log/syslog After press on folder to play
```
Feb 18 18:10:31 raspberrypi systemd[1]: Stopped Phoniebox GPIO Buttons Service.
Feb 18 18:10:31 raspberrypi systemd[1]: Started Phoniebox GPIO Buttons Service.
Feb 18 18:10:32 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:10:36 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:10:37 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]: /usr/lib/python3/dist-packages/gpiozero/pins/pi.py:270: PinNonPhysical: no physical pins exist for GPIO13
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   'no physical pins exist for %s' % repr(self)))
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]: /usr/lib/python3/dist-packages/gpiozero/pins/pi.py:270: PinNonPhysical: no physical pins exist for GPIO16
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   'no physical pins exist for %s' % repr(self)))
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]: Traceback (most recent call last):
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 54, in <module>
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:     volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 124, in __call__
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:     self = super(GPIOMeta, cls).__call__(*args, **kwargs)
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 432, in __init__
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:     bounce_time=bounce_time, pin_factory=pin_factory)
Feb 18 18:10:37 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 383, in __init__
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:     super(HoldMixin, self).__init__(*args, **kwargs)
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 186, in __init__
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:     self.pin.when_changed = self._pin_changed
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 450, in <lambda>
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:     lambda self, value: self._set_when_changed(value),
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 314, in _set_when_changed
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:     self._enable_event_detect()
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 250, in _enable_event_detect
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]:     bouncetime=self._bounce)
Feb 18 18:10:38 raspberrypi gpio-buttons.py[1518]: RuntimeError: Failed to add edge detection
[... 7 lines cut ...]
Feb 18 18:10:42 raspberrypi lighttpd[719]: rm: cannot remove '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/ZZZ/AudioFormatsTest/folder.conf': No such file or directory
```
### sudo systemctl status mopidy
```
● mopidy.service - Mopidy music server
   Loaded: loaded (/lib/systemd/system/mopidy.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-18 18:08:15 GMT; 42min ago
  Process: 427 ExecStartPre=/bin/mkdir -p /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 434 ExecStartPre=/bin/chown mopidy:audio /var/cache/mopidy (code=exited, status=0/SUCCESS)
 Main PID: 437 (mopidy)
   Memory: 237.2M
   CGroup: /system.slice/mopidy.service
           └─437 /usr/bin/python2 /usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf

Feb 18 18:50:02 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:08 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:14 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:20 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:27 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:33 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:39 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:45 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:51 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
Feb 18 18:50:57 raspberrypi mopidy[437]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
```

s-martin, 2020-02-18:
It seems that mopidy-iris dropped support for python 2 and mopidy2 after version 3.43.0. PR #793 should fix this.

s-martin, 2020-02-18:
PR #793 was merged to `master` and should fix this issue.

Please test.

s-martin, 2020-02-19:
#778 was fixed by PR #793 so I guess this issue should also be fixed.
But we'll wait for confirmation.

Thr33LittleBirds, 2020-02-19:
Okay I made a test a few minutes ago

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
+Spotify

- Mopidy-Iris is starting 
-> if i click on a song at the Iris-Frontend the player starts and **i can hear a sound just for a second.**
- Phoniebox Frontend
-> If i click on a Stream i can listen to the stream
-> If i click on a file, **i can hear the sound just for a second!**

Also it takes a few seconds to start a stream or a file.

s-martin, 2020-02-19:
What Raspberry Pi are you using?

s-martin, 2020-02-19:
Please post the output of `tail -n 500 /var/log/syslog` after trying to play a file

JuliH29, 2020-02-22:
Hi everybody,

I'll join the conversation as I am experience the issue that I can't play anything (neither Spotify nor local) as well.
I just installed Phoniebox via the masterbranch on buster (Version 2.0 rc7 [Installation protocol](https://pastebin.com/Rq9NVftq)).

The webapp does open, but I experience the following issues:

- The volume level is not indicated in % (even if I try to change it with the dropdown menu)
- Album art of local files does show up (with some delay) but the file never plays
- there is no play button
- If I press any of the other buttons I get error messages like `An error occorured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext Output: RC: .1`
- under "Info" I see `Mopidy Server Status
failed (Result: exit-code) since Sat 2020-02-22 13:48:19 CET; 9min ago
`

[Here](https://pastebin.com/J5PAscjL) is the output of `tail -n 500 /var/log/syslog
`.

Any idea what I could do? Should I try an install via the develope branch?
Your help would be very much appreciated!

s-martin, 2020-02-22:
You have the following error during installation: `pkg_resources.DistributionNotFound: The 'Mopidy==2.2.3' distribution was not found and is required by the application`

Looks like the current workaround doesn't work anymore.

So we need to fix #803 

@JuliH29 in the meantime you could try the mopidy 3 installation script, but you should expect problems.

JuliH29, 2020-02-22:
Thank you @s-martin four your quick reply.
I will wait for #803 to get fixed.

s-martin, 2020-02-22:
Try the following command to install with the Mopidy 3 script from `develop`:

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

But you should expect lots of problems as #803 is not implemented yet.

JuliH29, 2020-02-22:
OK, I changed my mind and just ran your script (version 2.0rc8 [installation protocol](https://pastebin.com/juxsDrMt)).

- The play button as well as the error messages for the other buttons are gone
- Mopidy Server Status: active (running) since Sat 2020-02-22 14:34:14 CET; 3min 25s ago
- If I open the Spotify tab I see the error message "Mopidy: Call to an invalid object. Check you are calling a valid Mopidy object."
- same tab, under settings: Status disconnted
- I still can't play any files (neither local nor Spotify)

What else can I do to help move along?

JuliH29, 2020-02-22:
**UPDATE:** Local files do play, spotify doesn't

s-martin, 2020-02-22:
You could try
`sudo python3 -m pip install mopidy-mpd` and
`sudo python3 -m pip install mopidy-local`

More details what need to be done could be find at #803 and https://mopidy.com/blog/2019/12/27/mopidy-3-faq

JuliH29, 2020-02-22:
After 
```
sudo python3 -m pip install mopidy-mpd 
sudo python3 -m pip install mopidy-local
```
I can play Spotify. So that's great!
But this error message remains:

> If I open the Spotify tab I see the error message "Mopidy: Call to an invalid object. Check you are calling a valid Mopidy object."

So I don't trust the whole thing yet...

s-martin, 2020-02-22:
Maybe you also need
`sudo python3 -m pip install Mopidy-Iris`

JuliH29, 2020-02-22:
Oh! We're getting there. For the first time in quite a while I saw "Connected" in the Spotify tab.
It takes about 20 seconds to connect but still shows 

> "Mopidy: Call to an invalid object. Check you are calling a valid Mopidy object."

s-martin, 2020-02-23:
@JuliH29 so does it work besides the issue #805?

JuliH29, 2020-02-23:
Yes, it does. Better than ever actually!

s-martin, 2020-02-24:
@Thr33LittleBirds do you still have issues for local files or has this been resolved for you, too?

s-martin, 2020-02-24:
PR #804 has been merged. Please try Mopidy 3 by using
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

JuliH29, 2020-02-29:
@s-martin do you have access to a Spotify premium account? If it would be helpful for further improving phoniebox spotify please send me an email, maybe I can help you out:
arm_9f7b6v@0box.eu
(That's an address created with trashmail, that's why it's so funny looking)

JuliH29, 2020-03-01:
> @JuliH29 so does it work besides the issue #805?

I have to get back to that unfortunately because my positive feedback seemed to be a bit premature.
I didn't change anything with my setup/installation and now - all of a sudden - spotify does not work anymore. The spotify tab looks quite empty, searches don't yield any results beside local files but the status is "connected" and I don't see any error messages.

Unfortunately I'm don't now anything about coding otherwise I would try to dig into this. Please let me know if there is anything else I can do to help.

s-martin, 2020-03-01:
Did you check `tail -n 500 /var/log/syslog` and `journalctl -u mopidy`?

JuliH29, 2020-03-02:
After a reinstall spotify runs without problems again (plays music, shows search results).
`tail -n 500 /var/log/syslog` and `journalctl -u mopidy` doesn't display any error messages (except for the issue discussed in #805).

closed 2020-03-03 by s-martin