MiczFlor/RPi-Jukebox-RFID #786: 🚀 | Kopfhörerausgang (variabel)

deheingit, 2020-02-10:
Hallo zusammen,

ich bastel gerade eine Phoniebox und hätte eine Frage/Vorschlag:

- Ich würde die Box mit einem miniamp bauen und dort direkt die Lausprecher anschließen
--> Was apssiert wenn ich dann einen Kopfhörer in die Buchste stecke? ich vermute nichts, da das ALSA Device fest eingetragen ist, oder?

--> Wäre es möglich die Phoniebox so zu verwenden, dass bei eingestecktem Kopfhörer (3,5mm) auf diesen umgeschaltet wird (wie man es auch von anderen Geräten kenn) und beim abziehen wieder zurück

VG Dennis

Yordan1976, 2020-02-10:
Ich habe diese Kopfhörerbuchsen verwendet: https://de.aliexpress.com/item/32847629394.html?spm=a2g0s.9042311.0.0.27424c4dLXIJ5o
Die unterbrechen dann das Signal zu den Lautsprechern mechanisch, solange was eingesteckt ist. Aber auch hier geht es wohl ohne MiniAmp einfacher.

Yordan1976, 2020-02-10:
So sieht das Konstrukt aus:
[image]

deheingit, 2020-02-11:
Hi @Yordan1976: Danke. 
Das heißt bei dir hat das Umschalten eigentlich nichts mit dem rpi zu tun sondern ist einfach von der USV Soundkarte gesteuert...

Yordan1976, 2020-02-11:
Nein, die Soundkarte steuert da gar nichts, die gibt nur das Signal aus. Das Umschalten passiert ganz alleine durch das mechanische Öffnen/Schließen der Kontakte in der Buchse.

ZyanKLee, 2020-02-16:
solche Buchsen habe ich auch genommen für diesen Zweck. Funktioniert auch sehr gut.

domu83, 2020-02-16:
@Yordan1976 ... hab mir die Seinerzeit auch bestellt, aber noch nicht verbaut. Hast Du evtl die Pinbelegung zur Hand? Ist mir nicht ganz ersichtlich, aber evtl bin ich auch einfach doof, nicht auszuschließen 🙄

Yordan1976, 2020-02-16:
Da habe ich mir doch tatsächlich gute Notizen zu gemacht. Bitte sehr!
[image]

domu83, 2020-02-16:
@Yordan1976  Der Hammer, Du bist ja cool! fecht herzlichen Dank!!

mimi-ac, 2020-03-05:
Hallo,
kann es sein, dass oben beschriebene Anschluss nicht in Kombination mit dem Hifiberry Miniamp funktioniert?

deheingit, 2020-03-06:
Hi @mimi-ac: Ich habe es auch so verstanden, dass der Stecker nur mit aktiven lautsprechern funktioniert. Der Minimp ist ja ein DAC, an den dann passive Lautsprecher angschlossen werden. z.B. normale Kopfhörer oder Aktivboxen kann man meines wissens nicht an den miniamp anschließen

domu83, 2020-03-27:
@Yordan1976 ich hatte mich am WE mal wieder an dem Thema versucht, allerdings bin ich gescheitert... Offensichtlich bin ich zu doof selbst mit Deiner Verkabelungsinfo das hinzubekommen.
Welche Lautsprecher hast Du denn genommen?

Yordan1976, 2020-03-27:
Diese Speedlink Twoxo, weil die auch kein störendes Rauschen haben:
https://smile.amazon.de/gp/product/B01HDR5EIK/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1
Aber im Prinzip hast du ja einfach drei Leitungen (Signal L + R und GND), von denen die Buchse die Signalleitungen beim Einstecken mechanisch trennt.

domu83, 2020-03-27:
danke Dir.... irgendwie hab ich in meiner Konstellation mit den Logitechboxen und mit dem Kabel scheinbar einen kurzschluss erzeugt. Next try...

manajoe, 2021-01-05:
@Yordan1976 Ich hab deine Anleitung mit den Trust-Boxen ([https://www.amazon.de/Trust-Lautsprecher-USB-Stromversorgung-Smartphone-schwarz/dp/B00JRW0M32/]) umgesetzt und noch eine Frage, wie du das mit der Lautstärke-Regelung gelöst hast.
Ich nutze eine USB-Soundkarte und habe bisher am Drehregler der Lautsprecher die Lautstärke runtergeregelt und im Gehäuse versteckt. Lauter/Leiser mache ich dann über die Phoniebox VolumeControl, von 0 bis 100%.
Wenn ich jetzt einen Kopfhörer an die zwischengeschaltete Buchse anstecke, ist der Ton unfassbar laut, der Drehregler der Lautsprecher greift hier ja nicht. Um eine erträgliche Lautstärke zu bekommen, müsste ich softwareseitig MAXVOL auf viell. 30% einstellen, dafür dann den Drehregler der Lautsprecher hochdrehen, damit ich dort wieder was höre.
Oder hast du das anders gelöst?

Danke und Grüße
Manajoe

Yordan1976, 2021-01-05:
Ich muss zugeben, dass der Kopfhörerausgang bisher noch nicht genutzt wird und mir das Problem daher nicht aufgefallen ist. Ich gehe aber davon aus, dass es nicht anders sein wird. Aber dein Weg ist prinzipiell nicht falsch. Die Boxen habe ich auch so eingestellt, dass sich bei 100% nicht die Nachbarn beschweren. Es wäre schön, wenn man softwareseitig feiner regeln könnte. Ich glaube erst ab 3% wird überhaupt etwas ausgegeben und wenn man die Boxen am Poti lauter dreht, ist das schon nicht so leise, wie ich es in ruhiger Umgebung gerne hätte.

vanMacG, 2023-05-07:
Auch wenn das Thema schon etwas älter ist wollte ich kurz meine Erfahrungen teilen.
Die kleine Kopfhörerbuchse gibt es auch bei Reichelt: [EBS 35](https://www.reichelt.de/klinkeneinbaubuchse-3-5-mm-stereo-ebs-35-p7301.html)
Was bei der Pin-Belegung von @Yordan1976 noch zu erwähnen ist: An Pin 1 = Masse muss das Massesignal von der Soundkarte (Audio-Quelle) und der Lautsprecher angelötet werden. D.h. es sind dort dann zwei Litzen angelötet, an den anderen Pins nur eine Litze.
Bzgl. Lautstärke: Ich verwende diese günstigen Lautsprecher von [nedis](https://www.reichelt.de/lautsprecher-pc-laptop-stereo-3-5-mm-buchse-n-cspr10020bk-p242512.html). Als Audio-Quelle verwende ich direkt den Klinkenausgang vom Raspi. Wenn ich den Lautstärkeregler der Lautsprecher auf max drehe (ist am Ende in der Box versteckt), ist die Lautstärke relativ gut mit dem Kopfhörer abgeglichen. Die Gesamtlautstärke regle ich dann über die Software. Die Abstimmung war bei mir direkt ziemlich gut imho.