MiczFlor/RPi-Jukebox-RFID #787: 🚀 | Aufnahmefunktion

deheingit, 2020-02-10:
Hallo zusammen

- Hat zufällig jmd. etwas implementiert womit man eigene Aufnahmen anfertigen kann? ich dachte in etwa so:

- RFID Karte die für aufnahmen festgelegt wurde an das Gerät halten

- Eine Taste startet die aufnahme

- Aufnahmen über angeschlossenes Mikrofon

- Eine Taste beendet die Aufnahme

- Beim nächsten mal wird ie Aufnahme abgespielt

- Drückt man die Aufnahmetaste wird die Aufnahme überspielt (nur bei vorher definierten karten-ids)

VG
 Dennis

Yordan1976, 2020-02-10:
Es gibt dazu zumindest bereits ein paar Funktionen: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#cardcontrol
Ich habe meine Box mit Line-In gebaut und auch mal getestet mit einer Karte die Aufnahme zu starten und beim erneuten Dranhalten wieder zu stoppen. Das hat auch funktioniert, aber ich habe es nicht weiterverfolgt.

deheingit, 2020-02-10:
Interessant. Wie kann man denn Line-In nutzen? Ginge das z.B. über den 3,5mm Anschluss des rpi wenn der Hifiberrry DAC eh die Ausgabe übernimmt?

Yordan1976, 2020-02-10:
Das weiß ich leider nicht. Bei meiner USB-Soundkarte hat es Aus- und Eingang.

s-martin, 2020-02-11:
> Interessant. Wie kann man denn Line-In nutzen? Ginge das z.B. über den 3,5mm Anschluss des rpi wenn der Hifiberrry DAC eh die Ausgabe übernimmt?

I don't think so. The 3.5mm Jack is just a "4-pole stereo **output** and composite video port", see https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/

deheingit, 2020-02-11:
ok, ich könnte aber theoretisch zusätzlich zum hifiberry noch ein USB Dac nutzen nur für Line-In, oder?
Damit macht die Software zwar noch nicht das richtige ;) - aber erstmal theoretisch

ZyanKLee, 2020-03-07:
yes in theory that should work.