MiczFlor/RPi-Jukebox-RFID #791: Spike: checkout yocto as optional solution to create phoniebox RPi images:

ZyanKLee, 2020-02-16:
https://www.yoctoproject.org/

lj2000, 2020-02-17:
Thank you for opening this spike! To my mind yocto is perfect for creating our specific Linux distribution / Board support package for the Phoniebox. 
In another (private) tinker project I set up an yocto environment for my RPI with the help of the following site:
https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Yocto.html

Of course we need some special packages for the phoniebox, too
But I think this page helps us to perform the first steps for getting the RPI up and running using the yocto build tooling.

BTW: There is another reason for yocto: At the moment the Phoniebox is based on the Raspberry Pi. 
Using yocto we "just" need to change the meta-layer of the hardware and so we can employ the Phoniebox on other hardware platforms, too.

ZyanKLee, 2020-02-18:
Thanks a lot for this link. I was kind of lost with this whole endeavor
when I struggled to get the build succeeding. Perhaps this will help me
with it.

Would you be willing to assist with a PoC or answer questions when I hit
the next wall?

lj2000, 2020-02-22:
I am very happy to assist you with a PoC and/or ( try :) ) to answer questions! I already have a running yocto environment for my (other) raspberry project which I want to update to the actual yocto revision the next days and weeks.

saeugetier, 2020-02-24:
I may help you with the yocto integration as well. It might be an easy deployment solution. I have done some projects with raspberry pi and yocto in the past.

ZyanKLee, 2020-02-24:
Yesterday I found few minutes to tinker with it. Followed the manual linked by @lj2000 and tried to create my own meta layer with a recipe. That failed sure to obvious reasons (user pi enjoy being available). Then I went to bed.

Thanks for offering your help. I will definitely come back to you once I understood the basics enough to not make a fool of myself by asking too easy questions :wink:

ThomasRausU, 2021-12-02:
Hi, I'm currently working on this https://github.com/ThomasRausU/yocto-rfid-jukebox
May someone likes to help with the recipes and learn about yocto.

pabera, 2021-12-02:
@ThomasRausU Nice, This looks very interesting. You template seems to target 1 single Jukebox setup though which is, looking at it historically, something unique. Every single Phoniebox has a different set up it seems.
This is why we build an installer to mitigate certain aspects like special RFID readers or custom sound cards.

We are currently working on a [V3 of the Phoniebox](https://github.com/MiczFlor/RPi-Jukebox-RFID#-version-3-is-coming-) software ([dev branch](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/develop)) and we had thoughts about building Pi images using buildroot. They would not allow us to bypass custom installations for RFID or sound, but we could avoid the Phoniebox builders to install the actual software or webapp. 

If you like to help with that, why don't you join our developer chat on [Matrix.org](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/791#issuecomment-984337661) (via Element.io). We discuss all sorts of things there and we love to have more members on board.

ThomasRausU, 2021-12-02:
> 
Hi, the https://github.com/ThomasRausU/yocto-rfid-jukebox is an example what you can do with the https://github.com/ThomasRausU/meta-jukebox yocto layer - there is also a future3/develop branch which is referenced in the yocto-rfid-jukebox example. Current dev status is that backend is running, frontend is build and running. Soundcard and rfid reader is working.
As you said there are a lot of different setups - I personally don't know what people are expecting what phoniebox should be. Should it be a image that is working out of the box or a application to install beside other applications. For my part I like to have a phonybox for my child and build it with yocto, because I know and love yocto. As I don't like php I started to continue this winter with V3. My plan is to have a working box by xmas ;)
I will join the chat.

saeugetier, 2021-12-02:
> Hi, I'm currently working on this https://github.com/ThomasRausU/yocto-rfid-jukebox May someone likes to help with the recipes and learn about yocto.

Nice work. I will look into your repository. And maybe i find some time to contribute. I have some experiences with Yocto. I was building a photobooth application in Yocto: https://github.com/saeugetier/photobooth

pabera, 2021-12-02:
Same [invitation](https://matrix.to/#/!pOwSabINahqdqVzbdA:matrix.org?via=matrix.org) goes to you @saeugetier 😉 .. We could use help in this domain!