MiczFlor/RPi-Jukebox-RFID #798: Playlist issue with subfolders only with RFID

ganafets82, 2020-02-20:
Hi there.
Good news fist: I got my phoniebox running well (I2S sound with amplifier, OLED-Display, RC522-RFID, GPIO Buttons, and on/off shim).
That was a long way. Rotary encoder and line out will follow.
That was my first Rasperry Pi project and my fist contact with ssh (runs headless). 

I use Rasbian buster 10 and phoniebix 2.0 rc7 classic on a Raspi 4

But there is a problem described following:

What works:
-playing single folders without subfolders (livestream, podcast, normal playlist...all works) via browser and RFID
-playing folders that contain subfolders via browser

Not working:
-playing the same folder that contain subfolders via RFID

I did not expect that starting a playlist behaves different with RFID and Browser.

Folder structure is this:

* audiofolders
  * mainfolder
    * subfolder 1 
      * livestream.txt (contains valid streamin-URL)
    * subfolder 2
      * livestream.txt (contains valid streamin-URL)
    *subfolder 3
    * ...

Playing "mainfolder" via browser adds all streams from the subfolders to the file ".../playlists/mainfolder-&RCRSV%.m3u" and starts playing as it should.

After swiping a RFID-card connected to "mainfolder", ".../playlists/mainfolder.m3u" is created, but without content (0 Bytes). If something is playing, this stops. The display shows that phoniebox expects a RFID input.

If I rename mainfolder-&RCRSV%.m3u to mainfolder.m3u I can start the playlist via RFID and I can play and navigate the playlist as expected.

Emptying the playlists dir, trying with a new card, deleting the content of latestID.txt, Latest_Folder_Played, Latest_Playlist_Played, Latest_RFID, all without success.

My workaround: adding several stream-URLs to a livestream.txt -> works, but so I cant assing a certain station to a RFID-card.

Also this shows the songtitles in the playlist instead of the radiostation name (this happens also with livestreams in subfolders and is actually not related to the main issue).

My first attempt was putting each stream-URLs in a different .m3u file (all in one directory), to handle the streams like normal files.
I thought so I can use the shuffle option with webradio.
This did not work at all (creates empty playlist both with RFID and browser).

Beste Grüße
Stefan

GeneralMidi, 2022-03-04:
Hi, quick question. How did you get the rfid reader _and_ the oled display working? I'd like to do that too, but not sure how to connect multiple i2c devices at the same time. Cheers.

chiefbrain, 2026-02-22:
I also like to play folders recursively.
In the frontend, one button plays with recursive; false (mainfolder.m3u), the other plays with recursive: true (mainfolder-&RCRSV%.m3u),

For RFID cards, the daemon `daemon_rfid_reader.py` only calls `rfid_trigger_play.sh` with the cardid, not other options like recursive. So currently, there is no option to play a folder recursively.

I hacked it by changing line 434 in `/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh`:
`if [ "$VALUE" == "recursive" ]; then`
->
`if [[ "$VALUE" == "recursive" || -f "${PATHDATA}/../shared/shortcutsRecursive/${CARDID}" ]]; then`

If a folder `/home/pi/RPi-Jukebox-RFID/shared/shortcutsRecursive` exists containing a file named like the cardid (e.g. a copy of the cardid file from shortcuts), it uses the &RCRSV% playlists.