MiczFlor/RPi-Jukebox-RFID #800: Kein Neustart der Playlist bei aufliegendem RFID-Chip

davidfri, 2020-02-20:
## Feature Description
Kein Neustart der Playlist bei aufliegendem RFID-Chip
### What functionality would you like to see in your phoniebox?

Bei unserer Phoniebox werden die Chips aufgelegt. Aus einem mir noch nicht nachvollziehbaren Grund wird die ID nach "same_id_delay" erneut verarbeitet. Spricht etwas dagegen diesen Teil zu entfernen, sodass nur bei einem Chip-Wechsel die ID ausgewertet wird?

### How do you envision the feature to work from a users perspective?

Der Chip (oder die Figur) kann auf der Box stehen bleiben. ;)

### Further information that might help

[remove_same_id_delay.patch.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4232308/remove_same_id_delay.patch.zip)

MondMann2019, 2020-02-21:
Gibt’s dafür nicht den Punkt „nur Systembefehle mehrfach ausführen“ in den Einstellungen unter „  Erneute Aktivierung“?

davidfri, 2020-02-21:
Mist.... Sorry, den Punkt habe ich echt nicht gesehen. Tut genau, was es soll ;)

closed 2020-02-21 by davidfri