MiczFlor/RPi-Jukebox-RFID #802: GPIO plug and play

MarkIpswich, 2020-02-21:
As to what this says

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons

I am looking for the buttons that require no soldering, just plug and play but when doing a search on Amazon etc I am  getting the ones I think that need soldering.

Where do I buy these from please?

s-martin, 2020-02-22:
The connectors on the raspberry are called Dupont connectors. 

Most arcade buttons you could buy e.g. on eBay require soldering, but maybe you could use blade connectors without soldering. 

But the soldering of two cables to an arcade button is very easy, maybe you have a friend who can help you. Maybe there’s a hack space in your area, where people could solder the buttons for you.

MarkIpswich, 2020-02-22:
Can you give me a link to the blade connectors that will work and I'll have a look.

s-martin, 2020-02-23:
This would be a cable I had in mind:
https://www.aliexpress.com/item/32820998411.html

But you probably need to measure on your arcade buttons which blade connector would fit.

Nevertheless I suggest to find a friend to solder cables with dupont connector to the arcade buttons, because the connection is more reliable.

MarkIpswich, 2020-02-23:
Is the dupont connector them things sticking up on the Raspberry Pi?

s-martin, 2020-02-23:
> Is the dupont connector them things sticking up on the Raspberry Pi?

Yes

MarkIpswich, 2020-02-23:
Through so.

So they need soldering onto the arcade buttons (which I don't have any yet) and the same for the dupont connector?

The blank ends of the wire to the connector?

s-martin, 2020-02-24:
> So they need soldering onto the arcade buttons (which I don't have any yet) and the same for the dupont connector?

This cable has already Dupont connectors (the black end).
You find plenty of cables with Dupont connectors on Amazon, eBay, etc. To connect to the raspberry pi you need Dupont female connectors.

The other end is the blade connector, which could fit your arcade button, but I recommend soldering the cable to the arcade button.

> Then once that is done will they work or do I need to install something as I enabled them when installing the phoniebox.

You need to configure GPIO buttons, see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons (but that description is a little dated, you don't need any pull-up resistors anymore).

closed 2020-12-18 by s-martin