MiczFlor/RPi-Jukebox-RFID #803: Upgrade Mopidy to version 3

s-martin, 2020-02-22:
As Python 2 is dropped by more and more parts it makes sense to upgrade Mopidy for Spotify integration to version 3.
Mopidy provides an extensive 
[FAQ](https://mopidy.com/blog/2019/12/27/mopidy-3-faq/) how to upgrade.

Important things to consider:

- Mopidy 3 works not on stretch, so version 2 should just be kept as is. New installations should be encouraged to use buster
- [Mopidy-MPD](https://mopidy.com/ext/mpd) and [mopidy-local](https://mopidy.com/ext/local) are now own extensions and need to be installed separately 

[mopidy-iris](https://mopidy.com/ext/iris/) has a bug https://github.com/jaedb/Iris/issues/474 in the last version for python 2, see #759. So Python 3 and Mopidy 3 would work around this.

[mopidy-gmusic](https://mopidy.com/ext/gmusic/) Integration will also only work with python 3, see #741.

@ZyanKLee providee already a install script for Mopidy 3 https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/installscripts/buster-install-mopidy3.sh but mostly untested.

I would work on this issue, but I don’t have a Spotify account, so other people would need to test this.

s-martin, 2020-02-24:
PR #804 has been merged. Please try Mopidy 3 by using
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

s-martin, 2020-02-24:
This issue should be closed, when mopidy3 script is merged into default script

Groovylein, 2020-02-26:
I've just tried the installation and mentioned the following error:
`rm: cannot remove '/home/pi/RPi-Jukebox-RFID/settings/global.conf': No such file or directory`
Maybe check if exists and then remove the file.

Currently, I'm not able to test sound, Spotify, ... since this is an RPi for testing.
Hope that helps :)

s-martin, 2020-02-27:
Yes, you're right that should be fixed.

s-martin, 2020-02-29:
We need to add pip upgrade commands, in case the user already has an older version installed, see https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/759#issuecomment-592781545

ZyanKLee, 2020-02-29:
I'd prefer requirements.txt files to pin versions and install those we know
exactly to work.

s-martin, 2020-02-29:
Good idea. 

GitHub would also create automatic pull requests, if any security issues appear for package versions used in requirements.txt.

Should we just put all possible packages in one requirements.txt or differentiate between classic, Spotify and maybe other components and e.g. Gmusic?

ZyanKLee, 2020-02-29:
I'd use multiple specific ones, during install they can be all added to one
install command. `pip -f req-base.txt -f req-spotify.txt` etc

Groovylein, 2020-02-29:
Hi @s-martin,
Why did you close the pull request? I think it's a good idea using requirements.txt

s-martin, 2020-02-29:
Yes, but some local changes were accidentally mixed in. I'll make a new one.

s-martin, 2020-03-03:
PR #804 at least solved #784

s-martin, 2020-03-07:
PR #804 fixed also #810

Hamsert, 2020-03-07:
So to solve #607 I just have to do this at my Phoniebox currently used, right?
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh

ZyanKLee, 2020-03-07:
this would re-install the whole stuff on your phoniebox and erase all media in the process. We would be glad if you could test it with a different sd-card abd a fresh buster installation. (just to be sure you can easily go back to your "working" setup afterwards)

s-martin, 2020-03-07:
Yes, testing this would really help.

`Attention`: please make Backups as this could delete your settings, RFID cards and music!

Hamsert, 2020-03-07:
Ok so fresh buster install via the automatic script and then run the above command, correct? Could try this in a few days.

ZyanKLee, 2020-03-07:
the above line IS the spotify installer script but with mopidy3 instead of mopidy2

Hamsert, 2020-03-07:
> @Hamsert the above line IS the spotify installer script but with mopidy3 instead of mopidy2
> thanks

Okay, so I install buster and then run above script and I should be good?

s-martin, 2020-03-07:
Yes

closed 2020-03-07 by ZyanKLee

Hamsert, 2020-03-08:
When I try it get get this result:
HTTP request sent, awaiting response... 404 Not Found
2020-03-08 10:49:42 ERROR 404: Not Found.
chmod: cannot access 'buster-install-mopidy3.sh': No such file or directory
bash: ./buster-install-mopidy3.sh: No such file or directory

s-martin, 2020-03-08:
> When I try it I get this result:
> HTTP request sent, awaiting response... 404 Not Found
> 2020-03-08 10:49:42 ERROR 404: Not Found.
> chmod: cannot access 'buster-install-mopidy3.sh': No such file or directory
> bash: ./buster-install-mopidy3.sh: No such file or directory

Sorry, we merged the mopidy script last night into the regular installer script.

Please use:
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh`

hlgrbmbm, 2020-03-08:
I am going to test the script - I really want the phoniebox to run :)

hlgrbmbm, 2020-03-08:
When I open Spotify from the Phoniebox website, I get the following errormessage:
"Mopidy: Call to an invalid object. Check you are calling a valid Mopidy object"

s-martin, 2020-03-08:
> When I open Spotify from the Phoniebox website, I get the following errormessage:
> "Mopidy: Call to an invalid object. Check you are calling a valid Mopidy object"

Can you provide a screenshot or logs?
`journalctl -u mopidy`

hlgrbmbm, 2020-03-08:
Sure, the log output is massive:

pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Sun 2020-03-08 12:50:33 CET
Mar 08 12:47:29 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 08 12:47:29 raspberrypi systemd[1]: Started Mopidy music server.
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ S
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.internal.l
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ E
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ D
Mar 08 12:47:44 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.__main__ E
Mar 08 12:47:44 raspberrypi mopidy[455]: Traceback (most recent call last):
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     extension.setup(registry)
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     from .library import ImageLibrary
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     class ImageLibrary(local.Library):
Mar 08 12:47:44 raspberrypi mopidy[455]: AttributeError: module 'mopidy.local' h
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands M
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
lines 2-25
Mar 08 12:47:29 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 08 12:47:29 raspberrypi systemd[1]: Started Mopidy music server.
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ S
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.internal.l
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ E
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ D
Mar 08 12:47:44 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.__main__ E
Mar 08 12:47:44 raspberrypi mopidy[455]: Traceback (most recent call last):
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     extension.setup(registry)
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     from .library import ImageLibrary
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     class ImageLibrary(local.Library):
Mar 08 12:47:44 raspberrypi mopidy[455]: AttributeError: module 'mopidy.local' h
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands M
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
Mar 08 12:47:45 raspberrypi mopidy[455]: INFO     [Audio-2] mopidy.audio.actor A
lines 2-26
Mar 08 12:47:29 raspberrypi systemd[1]: Starting Mopidy music server...
Mar 08 12:47:29 raspberrypi systemd[1]: Started Mopidy music server.
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ S
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.config Loa
Mar 08 12:47:42 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.internal.l
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ E
Mar 08 12:47:43 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.__main__ D
Mar 08 12:47:44 raspberrypi mopidy[455]: ERROR    [MainThread] mopidy.__main__ E
Mar 08 12:47:44 raspberrypi mopidy[455]: Traceback (most recent call last):
Mar 08 12:47:44 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:47:44 raspberrypi mopidy[455]:     extension.setup(registry)
[... 987 lines cut ...]
Mar 08 12:47:52 raspberrypi mopidy[455]: INFO     [MainThread] mopidy_mpd.actor 
Mar 08 12:47:52 raspberrypi mopidy[455]: INFO     [MainThread] mopidy.commands S
[... cut ...]
Mar 08 12:48:23 raspberrypi mopidy[455]:     raise value
Mar 08 12:48:23 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:48:23 raspberrypi mopidy[455]:     response = self._handle_receive(env
Mar 08 12:48:23 raspberrypi mopidy[455]:   File "/usr/local/lib/python3.7/dist-p
Mar 08 12:48:23 raspberrypi mopidy[455]:     return callee(*message.args, **mess
Mar 08 12:48:23 raspberrypi mopidy[455]:   File "/usr/lib/python3/dist-packages/
Mar 08 12:48:23 raspberrypi mopidy[455]:     tracks = schema.search_tracks(c, q,
lines 2-57

s-martin, 2020-03-08:
Could you try again with
`journalctl -f -u mopidy`

so the output doesn’t get truncated at the end of the lines?

hlgrbmbm, 2020-03-08:
pi@raspberrypi:~ $ journalctl -f -u mopidy
-- Logs begin at Sun 2020-03-08 13:09:50 CET. --
Mar 08 13:10:48 raspberrypi mopidy[1345]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Mar 08 13:10:48 raspberrypi mopidy[1345]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.1
Mar 08 13:10:48 raspberrypi mopidy[1345]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Mar 08 13:10:49 raspberrypi mopidy[1345]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Mar 08 13:10:49 raspberrypi mopidy[1345]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59986
Mar 08 13:10:49 raspberrypi mopidy[1345]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59990
Mar 08 13:10:54 raspberrypi mopidy[1345]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59998
Mar 08 13:10:59 raspberrypi mopidy[1345]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60002
Mar 08 13:11:05 raspberrypi mopidy[1345]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60006
Mar 08 13:11:10 raspberrypi mopidy[1345]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60010
Mar 08 13:11:16 raspberrypi mopidy[1345]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60014
Mar 08 13:11:22 raspberrypi mopidy[1345]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60018
Mar 08 13:11:27 raspberrypi mopidy[1345]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60026
Mar 08 13:11:33 raspberrypi mopidy[1345]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60032
Mar 08 13:11:39 raspberrypi mopidy[1345]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60036
Mar 08 13:11:45 raspberrypi mopidy[1345]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60040
Mar 08 13:11:50 raspberrypi mopidy[1345]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60044
Mar 08 13:11:56 raspberrypi mopidy[1345]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60048

[mopidy_logs.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4303174/mopidy_logs.log)

Hamsert, 2020-03-08:
I installed buster
Used this: cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh
Made the configurations: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch
Now, when I click on Spotify in the Webinterface I get: 404: Not Found
Do you need any logs, if so, how can I provide them?

ZyanKLee, 2020-03-08:
@Hamsert there is a bug in the script I will fix shortly. please run `cd /home/pi/RPi-Jukebox-RFID/ && sudo python3 -m pip install -r requirements-spotify.txt`

Hamsert, 2020-03-08:
Sweet, it works, thx! The webinterface feels way faster and also the connection to iris feels way better than the stretch version/mopidy&python 2.

ZyanKLee, 2020-03-08:
fixed in #847

Hamsert, 2020-03-08:
So if I would like to install it again to a new sd card, how would the full fixed command look like?

s-martin, 2020-03-08:
When #843 and #847 are merged the usual install command should work again.

ZyanKLee, 2020-03-08:
just merged those two PRs.

hlgrbmbm, 2020-03-08:
Even after reinstalling everything (Dev. Script) I recieve the same Error Message.
Just question in general, do I need to create a new Spotify clientID and secret?

ZyanKLee, 2020-03-08:
that may be the case, yes. Don't know why that might be, but its definitely a possibility.

JuliH29, 2020-03-08:
> Just question in general, do I need to create a new Spotify clientID and secret?

I upgraded to Mopidy 3 recently with the same client ID and secret and didn't have any issues.

htpzc, 2020-04-28:
Hi, i've tested the phoniebox on my pi zero with mopidy and got the spotify connect problem. 
So i switched to the v3 as described here.

First everything went fine. Fast start, Local Playlist plays sound, spotify plays sound, NICE!.
But after that i installed the shim onoff script and scanned the library
The scanning produces a lot of error messages because of not stopping mpd and not starting mopidy (i have that in my mind and hope that i am right, because, sorry i didnt save the logs).

Afterwards i rebooted, in hope everything would start fine again, but now i have the following problem with starting of mpd, and of this nothing is working
$ tailf -f /var/log/syslog 
---------------------------------------------------------------------------------------------------
...
Apr 28 19:48:01 raspberrypi systemd[839]: Starting Music Player Daemon...
Apr 28 19:48:01 raspberrypi systemd[1]: Started Session c1 of user pi.
Apr 28 19:48:01 raspberrypi systemd[1]: Startup finished in 2.283s (kernel) + 56.644s (userspace) = 58.927s.
Apr 28 19:48:06 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Apr 28 19:48:11 raspberrypi lighttpd[827]: echo: write error: Broken pipe
Apr 28 19:48:16 raspberrypi mpd[873]: Apr 28 19:48 : exception: Failed to set group 0: Operation not permitted
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mo
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mo
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/m
Apr 28 19:48:16 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.config Loading config from command line options
Apr 28 19:48:16 raspberrypi mopidy[491]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopid
Apr 28 19:48:16 raspberrypi systemd[839]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Apr 28 19:48:16 raspberrypi systemd[839]: mpd.service: Failed with result 'exit-code'.
Apr 28 19:48:16 raspberrypi systemd[839]: Failed to start Music Player Daemon.
Apr 28 19:48:16 raspberrypi systemd[839]: Reached target Default.
Apr 28 19:48:16 raspberrypi systemd[839]: Startup finished in 18.258s.
Apr 28 19:48:17 raspberrypi lighttpd[827]: echo: write error: Broken pipe
Apr 28 19:48:19 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.__main__ Enabled extensions: http, stream, iris
Apr 28 19:48:19 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Apr 28 19:48:22 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Apr 28 19:48:22 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Apr 28 19:48:22 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Apr 28 19:48:22 raspberrypi mopidy[491]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend,
Apr 28 19:48:22 raspberrypi mopidy[491]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Apr 28 19:48:23 raspberrypi lighttpd[827]: echo: write error: Broken pipe
Apr 28 19:48:25 raspberrypi mopidy[491]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in o
Apr 28 19:48:25 raspberrypi mopidy[491]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API a
Apr 28 19:48:29 raspberrypi lighttpd[827]: echo: write error: Broken pipe
Apr 28 19:48:35 raspberrypi lighttpd[827]: echo: write error: Broken pipe
---------------------------------------------------------------------------------------------------
So two things are not ok: 
- lighttpd[827]: echo: write error: Broken pipe --> this will popupabout 20 times and then will not arrive anymore. The Frontend is working, so i think this is not the main problem.

- mpd[873]: Apr 28 19:48 : exception: Failed to set group 0: Operation not permitted

not sure what todo
- tested to switch the mpd start to run with mpd user and not with root -> other error messages (permission problems etc.)
- added root to the root group (thought the script is checking this in /etc/groups maybe) -> didnt help
- when starting with $ sudo /etc/init.d/mpd start -> mpd is starting but it is not working with mopidy (i expect). No Sound output and in the frontend it will not play.

Any ideas?

Best regards Marcus