MiczFlor/RPi-Jukebox-RFID #806: Will not scan RFID tags

bustercody3, 2020-02-22:
I am using a raspberry pi 4 model b running the **latest** version of raspbian **(2020-02-13)**. I followed this tutorial on how to use the rc522 RFID scanner with a raspberry pi: https://pimylifeup.com/raspberry-pi-rfid-rc522/ / https://www.youtube.com/watch?v=evRuZRxvPFI&t=172s . After I had ran **Write.py** it would not recognize the RFID tag. I’m a beginner so I don’t understand most of this. The tags I were using **came with the scanner**. All my wires are **set up correctly and are soldered properly**. When I place the tag on it it will not write **“Written”** in the terminal. Any solutions? Here is a photo of my setup:
[image]

s-martin, 2020-02-23:
Did you check  https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special and https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader

s-martin, 2020-12-18:
I’m closing this now, feel free to reopen it, if the problem persists.

closed 2020-12-18 by s-martin