MiczFlor/RPi-Jukebox-RFID #807: Sound starts stuttering after approx 10 minutes of local playback (mp3)

hoodvisions, 2020-02-23:
Every 5 minutes a new issue... now that the box works again + spotify I have a new issue that only occurs when an MP3 is playing for longer than ~10 minutes, sometimes 11, sometimes 13, sometimes 9 minutes. The sound suddenly starts stuttering/hickups. I can "fix" this temporarily by pausing the audio for a second, but it will reoccur after a few minutes. This is annoying as my kid cannot listen to audiobooks alone.

Here is what I tried:
Added: sdhci-bcm2708.missing_status=0 sdhci-bcm2708.sync_after_dma=0
to: /boot/cmdline.txt

Changed the position of the box.
Re-uploaded and played different MP3 files that have over 10 minutes playtime.

Used htop to see if the cpu has trouble, but no. No changed when the stuttering begins (20% CPU).

User mpg123 to play the file.
THere is NO STUTTERING when using mpg123.

So I assume it is not about the kernel or hardware, but mpd or so, but I have no idea at all how to fix this. Does anyone have a good idea?
Many thanks in advance!

ZyanKLee, 2020-02-23:
just to clarify: this does not occur when you play 3 mp3s with each 5 minutes in length in a row?

ZyanKLee, 2020-02-23:
I can't think why this would have anything to do with the client that's controlling the mpd/mopidy. But I deem it very likely to be an issue with mopidy.

s-martin, 2020-02-23:
@hoodvisions do you currently use mopidy 3?
If yes, please check, if you used all the changes from #804.

hoodvisions, 2020-02-24:
@ZyanKLee exactly, playing multiple files in a row is fine.

@s-martin I have run the installation on friday, so it's likely it is not the v3. I am very hesitant to run `sudo apt upgrade` though as it took me days and days of work to finally get it up and running :)
Is there a way to make a "snapshot" of my current setup? Could I make an image of the SD card so I could get back to this point whenever I want?

What I haven't tested yet is what happens if I play a spotify track that lasts for longer than 12 minutes. Will do this today. 
Many thanks to you and your help, guys!!!

ZyanKLee, 2020-02-24:
sure, you can just copy the filesystem from the SD card to an image file on your computer. On linux for example you could run:
```
sudo dd if=/dev/mmcblk0 > Phoniebox-Backup.img
```

ZyanKLee, 2020-02-24:
Can you check if memory consumption is increasing while playing the large file?
I suspect that the file is slowly loaded into RAM by mopidy and kept there for some strange reasons.
If you could verify that suspicion, we should see if that issue has been solved in newer versions of the mopidy server (read: mopidy3 perhaps?).

hoodvisions, 2020-02-25:
Okay so memory consumption is pretty much stable at 20-30%. I am using a Raspberry 3B, by the way. I haven't made a backup yet - thanks a lot for the hint at how to do it.

I have also tested it playing "large" files from spotify (Tool, Metallica songs etc.) to see if local/spotify makes a difference. So yeah, it works when playing anything from spotify.

Okay, so now upgraded to mopidy 3 and now spotify doesn't work anymore, as the mopidy service stays dead for some reason. Damn! I need this box to run so I'd rather have crackling sound than nothing. I will resume (hopefully it works) to the image I just made and live with a faulty box I guess :)

s-martin, 2020-03-04:
Could you post `tail -n /var/log/syslog` and `journalctl -u mopidy`?

hoodvisions, 2020-03-04:
Sure, journalctl -u mopidy returns this: https://pastebin.com/8ADb7mzJ
And the other one returns tail: invalid number of lines: ‘/var/log/syslog’
When I add a 500 to it the output looks pretty much the same as the other one. Hundreds of lines with
`Mar  4 16:15:07 raspberrypi mopidy[457]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600`

Hm. Is this supposed to be filled to the brim with these lines?

ZyanKLee, 2020-03-04:
Well, that seems like some mpd-client polling the mpd all the time for the current status (the phoniebox webinterface perhaps?).
It's ugly, but should not be a problem per sé.

ZyanKLee, 2020-03-04:
There are messages in the pastebin about the Spotify account data not being readable, though?

s-martin, 2020-03-04:
You are still using Mopidy 2.2

> Mar 02 20:17:06 raspberrypi systemd[1]: Started Mopidy music server.
> Mar 02 20:17:10 raspberrypi mopidy[457]: /usr/lib/python2.7/dist-packages/mopidy/ext.py:202: PkgResourcesDeprecationWarn
> Mar 02 20:17:10 raspberrypi mopidy[457]:   extension_class = entry_point.load(require=False)
> Mar 02 20:17:12 raspberrypi mopidy[457]: INFO     Starting Mopidy 2.2.2

If you are on Buster, you could try to use Mopidy 3. See PR #804 

Please use 

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh`

hoodvisions, 2020-03-06:
I am using it *again* :)
When I update to mopidy3 nothing works anymore, neither local playback nor can I reach the spotify (IRIS) interface, so I reverted back to the prior backup point. A friend of mine purchased the exact same hardware as I have and will try it this or next weekend.

s-martin, 2020-03-06:
So if you have any results and log files on your new machine that would be great.

JuliH29, 2020-03-06:
> I am using it _again_ :)
> When I update to mopidy3 nothing works anymore, neither local playback nor can I reach the spotify (IRIS) interface, so I reverted back to the prior backup point. A friend of mine purchased the exact same hardware as I have and will try it this or next weekend.

Did you try [this](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/784#issuecomment-589958561) and [this](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/784#issuecomment-589971913)?

s-martin, 2020-05-30:
We released 2.0 a couple of days ago with many improvements, so you might want to check, if this fixes your issue.

s-martin, 2020-12-18:
I’m closing this now, feel free to reopen the issue, if the problem still persists in 2.2

closed 2020-12-18 by s-martin