MiczFlor/RPi-Jukebox-RFID #810: Resume Spotify Streams not working

kuesimon, 2020-02-24:
Hello,
I want to use the resume:on option for Spotify Streams. The option can be chosen but if I scan the rfid on the box again the whole systems seems to be not working anymore. Restart is also not working. I have to uncheck the resume option and restart to get the box running again. 

Can somebody help me with this issue? 
Thanks and best regards 
Simon

System: rpi3, USB sound card, gpio buttons, software master branch as of 22.03.2020

ZyanKLee, 2020-02-24:
I can at least confirm this issue. Noticed it myself while testing other stuff.

kuesimon, 2020-02-25:
Today I had the issue again but not only with resume but also with shuffle mode of a stream.

quottax, 2020-02-27:
Its the same for me. Resuming a Spotify-Stream is not possible.
I read on another page that the problem is the float-Number in elapsed time and i modified the script so that there is a convertation from float to integer. But then there are new Problems with stopping the stream.
Is there a solution for this?

s-martin, 2020-02-28:
@quottax: could you post here what you changed and what errors this has caused?

quottax, 2020-02-28:
I added this line in line 93 in resume_play.sh:
`ELAPSED=${ELAPSED/\.*}`

like this
`CURRENTFILENAME=$CURRENTFILENAME`
`ELAPSED=$ELAPSED`
`ELAPSED=${ELAPSED/\.*}`
`PLAYSTATUS="Stopped"`

after that it resumes spotify too, but its not stable. So the Stop over RFID doesnt work at any time for example.

kuesimon, 2020-03-07:
@s-martin after installing your suggested script in #700 the resuming/shuffling now is working. So the upgrade to Mopidy 3 will fix this issue in the master branch.

> We are upgrading to Mopidy 3 #803, which should fix some issues. PR #804 has been merged. Please try Mopidy 3 by using
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-mopidy3.sh; chmod +x buster-install-mopidy3.sh; GIT_BRANCH=develop bash ./buster-install-mopidy3.sh

I will close the issue for now.

closed 2020-03-07 by kuesimon