MiczFlor/RPi-Jukebox-RFID #811: Please help me get the rotary knob to work

hoodvisions, 2020-02-25:
Okay, so I gave up on the sound issues and a few other things and I installed this box around a 50 times now, threw >150 EUR at things like soldering iron, cables etc. I want to conclude this section now and just live with a semi-working box where I can at least change the volume properly.

I bought exactly the hardware specified here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume

I used female to female cables to connect the pins as in the ASCII image.

I installed the script(s) as instructed, The service phoniebox-rotary-encoder is running. When typing `sudo systemctl status phoniebox-rotary-encoder` it returns

Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK MPD 0.21.4
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: expr: syntax error: unexpected argument '10'
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: ./scripts/playout_controls.sh: Zeile 214: [: -le: Einstelliger (unär
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK MPD 0.21.4
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: expr: syntax error: unexpected argument '10'
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: ./scripts/playout_controls.sh: Zeile 214: [: -le: Einstelliger (unär
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK MPD 0.21.4
Feb 25 18:11:31 raspberrypi rotary-encoder.py[448]: OK

I can turn the knob like hell, but it won't change the volume at all...........
I bought a second knob. Same result.

I triple checked all the cables.
CLK -> PIN29
DT -> PIN31
GND -> PIN30
+ -> PIN17

For the love of god, please help me end this torture. The phoniebox turned out to be far more expensive and a pain in the ... than simply buying a toniebox. I just want this to work and then never touch it again :)

s-martin, 2020-02-26:
@veloxidSchweiz worked on the rotary knobs and maybe can help.

s-martin, 2020-02-26:
I can see that line 214 in playout_controls.sh has changes in `develop` to `master`, but need to investigate further, what's the exact issue. I don't have any experience with rotary knobs.

veloxidSchweiz, 2020-02-27:
Hi @hoodvisions ,
You could try my branch to check  at least the functionality.
 I am still working on the code but yesterday I tried it and it was working perfect.
https://github.com/veloxidSchweiz/RPi-Jukebox-RFID/blob/feature/configureableGPIOButtons/components/gpio_control/GPIODevices/rotary_encoder.py
Executing this file (with python3) will allow you test the knob. It will ask you for the pins and then when turning the knob you see if increase/decrease are recognized.

You LL be a

hoodvisions, 2020-02-27:
@veloxidSchweiz I tried this, thanks. Now, it asks me for the PINS and then says "running". When turning the knob nothing happens. 
I have used these cables here: https://www.amazon.de/gp/product/B01EV70C78/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 Is it possible these don't work?

hoodvisions, 2020-02-27:
Okay so when I HOLD the rotary knob in one of these tiny "edges" (you know, that little resistance when turning it) I get a signal. So a friend wrote me a script that spills out all GPIO signals each second and holding or turning the knob like hell leads to signals sometimes being recognized.

So I think the issue here is the Phoniebox simply cannot read these signals due to python being too slow maybe? Right now the knob https://www.amazon.de/gp/product/B07TKK4QQD/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1 is not working out of the box. :(

veloxidSchweiz, 2020-03-06:
Hi @hoodvisions,
I do have the same knobs and do not have any problems at all...
The cables look fine as well. At the moment my box is closed and i cannot check to which supply voltage i connected the knob (5V vs 3.3V).

hoodvisions, 2020-03-06:
This is weird... I have reinstalled everything from scratch again, but same results. I will assume that the knobs are simply both nonfunctional and purchase another one from a different vendor maybe.

hoodvisions, 2020-03-07:
In the meantime a friend of me bought the exact same hardware. Run the setup etc but the rotary simply doesn't work. And he found out that it is about the external soundcard. So the rotary only changes volume when using the internal soundcard, but not an external USB soundcard.
Is there any setting somewhere to be found to tell the script to use the correct soundcard instead?

s-martin, 2020-03-07:
Please check here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working how to change the `alsa.conf`

hoodvisions, 2020-03-07:
Many thanks!!
Okay, so here is how it works when using "my" USB external Soundcard with the CM108 chipset by Sabrent:

`sudo nano /usr/share/alsa/alsa.conf`
Change `defaults.ctl.card 0` and `defaults.pcm.card 0` to `defaults.ctl.card 1` and `defaults.pcm.card 1` depending on your soundcard. ( https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/updating-alsa-config )

SO if you now end up like me where a hardware knob and software volume buttons don't do anything, you need to do this:

`sudo nano /etc/mpd.conf`
and look for this:

` # An example of an ALSA output:
  #       
    audio_output {
            type            "alsa"
            name            "My ALSA Device"
    #       device          "hw:0,0"        # optional
    #       mixer_type      "software"      # optional
    #       mixer_device    "default"       # optional
    #       mixer_control   "PCM"           # optional
    #       mixer_index     "0"             # optional
    }`

and uncomment (delete # in front of) mixer_type "software".
`sudo reboot`
Now not only the playback works but also the volume settings do as well as the rotary knob linked in the manual :)

Thanks a lot for your help, guys!

closed 2020-03-07 by hoodvisions